springfall2008/batpred #1978: Pre day core update weirdness

ProximusAl, 2025-02-07:
**Describe the bug**
Update to Predbat core updates and then continues to want to update to same version

**Expected behaviour**
For it not to

**Predbat version**

8.14

[image]

ProximusAl, 2025-02-07:
I notice installed version is *small*v8.14 versus available *large*V8.14

WyndStryke, 2025-02-07:
Same behaviour on my installation, but I notice that there was no description alongside the update.

I went back to 8.13.3(?), and now the 8.14.0 update has a description.  So I'm giving it another go as we speak.

gcoan, 2025-02-07:
> I was also getting errors when trying to run the new compare, but perhaps that's due to either an error in my apps.yaml, or not being an Octopus customer.

You shouldn't need to be an octopus customer as the comparison uses publicly available octopus rate api's

I have auto installed the new release and am also seeing a message that there is still an update available

WyndStryke, 2025-02-07:
> You shouldn't need to be an octopus customer as the comparison uses publicly available octopus rate api's

Ah that's handy :-)  In that case I can use it to compare between my tariff (E-on Next Drive) and the Octopus family of tariffs to see how they compare.  Might actually be interesting to set up multiple suppliers that way.

Tried re-doing the apps.yaml and the compare is working now (can't see what was wrong with the original, but looks happy with a fresh copy&paste).

lewiswatt, 2025-02-07:
> Ah that's handy :-) In that case I can use it to compare between my tariff (E-on Next Drive) and the Octopus family of tariffs to see how they compare. Might actually be interesting to set up [...]

Could you elaborate on anything you did differently, other than just re-copy/paste it into your apps.yaml? I can't seem to shake that exact error you had earlier.

WyndStryke, 2025-02-07:
Originally I copied & pasted the extract from the documentation, but then added a new entry for my existing tariff.

The second time around, I just copied & pasted it without changing anything.

So my guess is that there was a problem with the section I added.

This was the bad version, to fix it I just removed the 'cap_seg' section which I had modified to match E-on Next Drive.

```
# Tariff comparison feature
  # Adjust this list to the tariffs you want to compare, include your current tariff also
  compare_list:
    - id: 'current'
      name: 'Current Tariff'
    - id: 'cap_seg'
      name: 'Price cap import/Seg export'
      rates_import:
        -  start: "00:00:00"
           end: "07:00:00"
           rate: 6.7
        -  start: "07:00:00"
           end: "00:00:00"
           rate: 25.41
      rates_export:
         - rate 16.5
    - id: 'igo_fixed'
      name: 'Intelligent GO import/Fixed export'
      rates_import_octopus_url: 'https://api.octopus.energy/v1/products/INTELLI-BB-VAR-23-03-01/electricity-tariffs/E-1R-INTELLI-BB-VAR-23-03-01-A/standard-unit-rates/'
      rates_export_octopus_url: 'https://api.octopus.energy/v1/products/OUTGOING-VAR-BB-24-10-26/electricity-tariffs/E-1R-OUTGOING-VAR-BB-24-10-26-A/standard-unit-rates/'
    - id: 'igo_agile'
      name: 'Intelligent GO import/Agile export'
      rates_import_octopus_url: 'https://api.octopus.energy/v1/products/INTELLI-BB-VAR-23-03-01/electricity-tariffs/E-1R-INTELLI-BB-VAR-23-03-01-A/standard-unit-rates/'
      rates_export_octopus_url: 'https://api.octopus.energy/v1/products/AGILE-OUTGOING-BB-23-02-28/electricity-tariffs/E-1R-AGILE-OUTGOING-BB-23-02-28-A/standard-unit-rates/'
    - id: 'go_fixed'
      name: 'GO import/Fixed export'
      rates_import_octopus_url: 'https://api.octopus.energy/v1/products/GO-VAR-BB-23-02-07/electricity-tariffs/E-1R-GO-VAR-BB-23-02-07-A/standard-unit-rates/'
      rates_export_octopus_url: 'https://api.octopus.energy/v1/products/OUTGOING-VAR-BB-24-10-26/electricity-tariffs/E-1R-OUTGOING-VAR-BB-24-10-26-A/standard-unit-rates/'
    - id: 'go_agile'
      name: 'GO import/Agile export'
      rates_import_octopus_url: 'https://api.octopus.energy/v1/products/GO-VAR-BB-23-02-07/electricity-tariffs/E-1R-GO-VAR-BB-23-02-07-A/standard-unit-rates/'
      rates_export_octopus_url: 'https://api.octopus.energy/v1/products/AGILE-OUTGOING-BB-23-02-28/electricity-tariffs/E-1R-AGILE-OUTGOING-BB-23-02-28-A/standard-unit-rates/'
    - id: 'agile_fixed'
      name: 'Agile import/Fixed export'
      rates_import_octopus_url: 'https://api.octopus.energy/v1/products/AGILE-24-10-01/electricity-tariffs/E-1R-AGILE-24-10-01-A/standard-unit-rates/'
      rates_export_octopus_url: 'https://api.octopus.energy/v1/products/OUTGOING-VAR-BB-24-10-26/electricity-tariffs/E-1R-OUTGOING-VAR-BB-24-10-26-A/standard-unit-rates/'
    - id: 'agile_agile'
      name: 'Agile import/Agile export'
      rates_import_octopus_url: 'https://api.octopus.energy/v1/products/AGILE-24-10-01/electricity-tariffs/E-1R-AGILE-24-10-01-A/standard-unit-rates/'
      rates_export_octopus_url: 'https://api.octopus.energy/v1/products/AGILE-OUTGOING-BB-23-02-28/electricity-tariffs/E-1R-AGILE-OUTGOING-BB-23-02-28-A/standard-unit-rates/'
    - id: 'flux'
      name: 'Flux import/Export'
      rates_import_octopus_url: 'https://api.octopus.energy/v1/products/FLUX-IMPORT-23-02-14/electricity-tariffs/E-1R-FLUX-IMPORT-23-02-14-A/standard-unit-rates'
      rates_export_octopus_url: 'https://api.octopus.energy/v1/products/FLUX-EXPORT-BB-23-02-14/electricity-tariffs/E-1R-FLUX-EXPORT-BB-23-02-14-A/standard-unit-rates'
    - id: 'cosy_fixed'
      name: 'Cosy import/Fixed export'
      rates_import_octopus_url: 'https://api.octopus.energy/v1/products/COSY-22-12-08/electricity-tariffs/E-1R-COSY-22-12-08-A/standard-unit-rates'
      rates_export_octopus_url: 'https://api.octopus.energy/v1/products/OUTGOING-VAR-BB-24-10-26/electricity-tariffs/E-1R-OUTGOING-VAR-BB-24-10-26-A/standard-unit-rates/'
    - id: 'cosy_agile'
      name: 'Cosy import/Agile export'
      rates_import_octopus_url: 'https://api.octopus.energy/v1/products/COSY-22-12-08/electricity-tariffs/E-1R-COSY-22-12-08-A/standard-unit-rates'
      rates_export_octopus_url: 'https://api.octopus.energy/v1/products/AGILE-OUTGOING-BB-23-02-28/electricity-tariffs/E-1R-AGILE-OUTGOING-BB-23-02-28-A/standard-unit-rates/'
  
```

```
current | Current Tariff | 2025-02-07 23:15:49 | -127.24 | -155.63 | -120.69 | 36.1 | 50.65 | 0.6 | 0.0 | 0 | Best Existing
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
igo_fixed | Intelligent GO import/Fixed export | 2025-02-07 23:16:20 | -55.82 | -82.19 | -48.4 | 31.63 | 45.8 | 0.6 | 0.0 | 0 |  
igo_agile | Intelligent GO import/Agile export | 2025-02-07 23:17:37 | -60.72 | -89.23 | -41.17 | 34.13 | 48.36 | 0.6 | 0.0 | 0 |  
go_fixed | GO import/Fixed export | 2025-02-07 23:18:08 | 14.52 | 26.68 | 31.08 | 29.69 | 43.64 | 5.14 | 0.0 | 0 |  
go_agile | GO import/Agile export | 2025-02-07 23:19:16 | -27.48 | -61.78 | -5.57 | 23.32 | 36.17 | 0.6 | 0.0 | 0 |  
agile_fixed | Agile import/Fixed export | 2025-02-07 23:20:05 | 178.33 | 218.8 | 223.84 | 12.87 | 26.81 | 5.69 | 0.0 | 0 |  
agile_agile | Agile import/Agile export | 2025-02-07 23:22:13 | 178.33 | 218.8 | 223.84 | 12.87 | 26.81 | 5.69 | 0.0 | 0 |  
flux | Flux import/Export | 2025-02-07 23:22:42 | 61.61 | 6.42 | 63.02 | 23.85 | 39.36 | 0.6 | 0.0 | 0 |  
cosy_fixed | Cosy import/Fixed export | 2025-02-07 23:23:14 | 122.51 | 278.94 | 310.56 | 13.14 | 27.32 | 14.79 | 0.0 | 0 |  
cosy_agile | Cosy import/Agile export | 2025-02-07 23:24:36 | 54.55 | 8.71 | 45.89 | 23.78 | 39.64 | 0.6 | 0.0 | 0
```

lewiswatt, 2025-02-07:
Yeah that's done it. Removing the cap_seg section has completely fixed it! Cheers

gcoan, 2025-02-07:
I'd deleted the seg comparison myself as I wasn't interested in it, so hence didn't have this issue

I've been playing with the comparison tool and added  a feature request for improvements I suggest could be made #1980

Feel free to add any other thoughts/observations to it

WyndStryke, 2025-02-07:
Trefor mentioned in his video about how it could also be used to compare plans with different options, so maybe expanding on that in the docs a bit.

For example I seem to get a much more lucrative plan if it can export during charge slots, not sure why it makes such a big difference.   So it occurred to me that this could be used to answer the perennial question "why did it do that plan" - take the plan that the user is querying, adjust some of the non-default options back to default for each alternative, and re-run the plan, and then they can see the effect that each setting had on the outcome.  Some will have minimal effect, and others will have a significant effect.

WyndStryke, 2025-02-08:
Oh, nobody seems to have mentioned it yet, so I will - the eternal-update issue disappeared a while back, so maybe the release was tweaked?  Is it now OK for you, or are you still seeing it?

gcoan, 2025-02-08:
> Oh, nobody seems to have mentioned it yet, so I will - the eternal-update issue disappeared a while back, so maybe the release was tweaked?  Is it now OK for you, or are you still seeing it?

yep, its gone for me as well

closed 2025-02-08 by ProximusAl