springfall2008/batpred #1979: Upgrading inverter/battery advice

TheSargey, 2025-02-07:
Evening folks,

I'm in the process of upgrading from a Givenergy gen 1 inverter and gen 1 batteries to the Gen3 inverter and associated Gen3 batteries. I'll be increasing capacity of both inverter and battery storage.

With predbat, my assumption is it would be best to completely reinstall predbat and start afresh rather than  simply updating apps.yaml with the new config? However I'd lose my historical usage etc, just wanted some advice if there's a better way than just a complete reinstall.

Many thanks as always for any advice given.

Best,
Sarge

gcoan, 2025-02-07:
You shouldn't need to reinstall Predbat, simply restart GivTCP so it picks up the new inverter details and then Predbat so it picks up the details from GivTCP.

In terms of history, the new inverter and battery will have new serial numbers so if you have used things like givtcp_xxxx_total_battery_charge_kwh/discharge/pv_energy in the Energy dashboard in HA then you will need to add the new entity names and you'll have two sets of battery charge and discharge entities as the history doesn't migrate.

There are ways to migrate the history though, I'm working on writing up how to do this

Congratulations with your new toys !

TheSargey, 2025-02-08:
Thanks for the rapid response Geoff, don't you ever sleep?! 😂

OK, sounds great. In terms of history I was thinking of house load and that
sort of thing. However I do have a lot of custom dashboard bits that have
my specific givtcp_xxx entities so yes I'll need to rework these.

Good to know it should just be a case of restarting the two addons, looking
forward to having the new setup implemented and finally getting paid to
export! Doubt it'll ever cover the cost of multiple installs but... Ce la
vie.

Thanks again sir, all the best.

Sarge

springfall2008, 2025-02-09:
You might need to set 'geserial' manually so it doesn't pick the old one

TheSargey, 2025-02-10:
> You might need to set 'geserial' manually so it doesn't pick the old one

Thanks Trefor, the old inverter is being removed completely so hopefully it won't see it.  But I'll bear that in mind.

Thanks again to you both for your responses.

Best,

Sarge

closed 2025-02-10 by TheSargey