springfall2008/batpred #1980: FR: Enhancements to tariff compare feature

gcoan, 2025-02-07:
**Is your feature request related to a problem? Please describe.**
Just tried out the new tariff compare feature, a brilliant addition Trefor, thanks for adding this, I think it'll be really useful.

A few observations/comments on using it for the first time:

1. suggest the iboost and carbon columns be suppressed if they are not configured in Predbat, in my case they are not, and are both always zero

[image]

2. Similarly, suppress actual(car) if a car isn't setup in apps.yaml, as I don't have one, the data is always the same as actual

3. I'm not really sure what Actual is showing me, its not the current cost, nor is it a projection

4. When running a manual compare, at 23:30 in my case the total figures include all of today's consumption plus the projection for tomorrow.  This gives really large projected values that aren't really useful.
Wouldn't happen with the midnight automatic comparison, but suggest that if a manual comparison is done a today value of zero be taken otherwise its a day and a part day comparison.

[image]

5. The comparison appears to be of 'now' plus 24 hours, so running it at 23:30 it runs the comparison up to the same time tomorrow.
I suggest that the comparison needs to run for a longer time window (say mid-day the following day) even if its then truncated back to show just tomorrow's data.
The reason for this is illustrated on Cosy which I'm currently on.  There is a cheap rate charging period now from 22:00-00:00 but the predbat comparison doesn't do any charging in that period because there is no future load that will be consumed - so predbat (rightly) determines its not worth charging.
However of course in reality you would charge in that period to take advantage of the cheap rate period.

[image]

WyndStryke, 2025-02-08:
My suggestion is this:

* A magic button to answer the eternal question "Why did it generate the plan like this?"

If the user clicks it, the system will create alternatives which each differ by a single system configuration, one at a time reverting to the default (maybe ignore the harmless ones).  So they can see the effect that each configuration setting had on their plan, whether large or small.

springfall2008, 2025-02-08:
1. suggest the iboost and carbon columns be suppressed if they are not configured in Predbat, in my case they are not, and are both always zero

Done

2. Similarly, suppress actual(car) if a car isn't setup in apps.yaml, as I don't have one, the data is always the same as actual

Done

4. When running a manual compare, at 23:30 in my case the total figures include all of today's consumption plus the projection for tomorrow. This gives really large projected values that aren't really useful.
Wouldn't happen with the midnight automatic comparison, but suggest that if a manual comparison is done a today value of zero be taken otherwise its a day and a part day comparison.

I've changed the code to start a 0 for all comparisons now

5. The comparison appears to be of 'now' plus 24 hours, so running it at 23:30 it runs the comparison up to the same time tomorrow.
I suggest that the comparison needs to run for a longer time window (say mid-day the following day) even if its then truncated back to show just tomorrow's data.
The reason for this is illustrated on Cosy which I'm currently on. There is a cheap rate charging period now from 22:00-00:00 but the predbat comparison doesn't do any charging in that period because there is no future load that will be consumed - so predbat (rightly) determines its not worth charging.
However of course in reality you would charge in that period to take advantage of the cheap rate period.

Actually the comparison should run for 48 hours and then only count the next 24 hours to avoid odd issues like dumping the battery at the end of the period. It's possible something was broken here so I did make a change to ensure its always the case - please re-test.

6. Thanks for finding all the different tariff URL's, that's a pain!
Can you add tracker and Economy 7 please

I can find their economy 7 but its not really the same as you need a smart meter (snug)
Not sure what you mean by tracker, 3 monthly one?

7.Linked to point 4, the manual comparison (run on 7/2) appears alongside the automatic comparison (run on 8/2), but since the manual comparison is of 'today'+24 hours whereas the automatic is from midnight (at zero cost)+24 hours the comparison graph values are distorted:

The manual will only show at the end of the chart, it will be dropped on the next day in favour of the datapoint at midnight

springfall2008, 2025-02-08:
https://github.com/springfall2008/batpred/releases/tag/v8.14.2

gcoan, 2025-02-08:
> Thanks for finding all the different tariff URL's, that's a pain!
> Can you add tracker and Economy 7 please
> I can find their economy 7 but its not really the same as you need a smart meter (snug)
> Not sure what you mean by tracker, 3 monthly one?

Thanks for all the responses and bug fixes, I'll upgrade to the new release and let you know of any issues.

Economy 7 and Tracker don't appear to be on the list of product codes https://api.octopus.energy/v1/products/ 

By Tracker I meant the tariff is variable like Agile except it gives you a single daily price based upon wholesale rates. Its very similar to the SVR so TBH not worth including at the momment.
I see you added Snug, its and E7-like tariff but its for owners of specific electric storage heaters that Octopus can control remotely,  so similar to IOG for storage heaters.  Quite niche.

Here's the manual setup for Economy 7 with SEG:

```
- id: 'eco7_seg'
  name: 'Eco7 import/SEG export'
  rates_import:
    - start: "00:30:00"
      end: "07:30:00"
      rate: 13.52
    - start: "07:30:00"
      end: "00:30:00"
      rate: 32.24
  rates_export:
     - rate: 4.1
```

PianSom, 2025-02-08:
I am not entirely convinced that showing pricing to the nearest 0.01p is (a) necessary or (b) accurate :)

Perhaps costs could be displayed like £4.27 rather than 427.31 [p]?

gcoan, 2025-02-08:
> 2. Similarly, suppress actual(car) if a car isn't setup in apps.yaml, as I don't have one, the data is always the same as actual
> Done

Hi Trefor, am running 8.14.3.  Still seeing identical lines in the graph for actual and actual (no car):

[image]

> 8. In the automatic midnight comparison there is a discrepancy between the costs presented in the table and the html plans shown below.
e.g. the current plan in the table has cost £902 but the projected cost is £8.69. There is a similar difference on all of the other tariff comparisons between the top table and the bottom HTML charts:

I'm still seeing a difference between the figures in the summary table and the end cost in the plan:

[image]
[image]

And agree, can we have the table formatted as £ and pence (or whatever current configured in apps.yaml), don't need 2dp.

Brilliant addition to predbat

springfall2008, 2025-02-08:
The car line depends on 'car_charging_hold' being turned on or off rather than the number of cars, mostly as some people use that to remove data from other sources.

Can you show the differences, you only showed the table and not the end of plan?

gcoan, 2025-02-08:
> The car line depends on 'car_charging_hold' being turned on or off rather than the number of cars, mostly as some people use that to remove data from other sources.

Thanks, turned car charging hold off and the car line disappears. 

> Can you show the differences, you only showed the table and not the end of plan?

Here's the table and plan for current and Cosy (but there is a similar difference on all the tariff plans):

[image]
[image]
[image]
[image]
[image]
[image]
[image]

springfall2008, 2025-02-08:
Okay, that's because the pounds value in the HTML plan is at the start of the 30 minute slot and the cost in the compare table is at the end of the period.

springfall2008, 2025-02-08:
New release made changing to £'s and also including now re-planning the car charging if enabled

gcoan, 2025-02-09:
new release looks good thanks Trefor

closed 2025-02-15 by springfall2008