springfall2008/batpred #1992: Charging battery early in Octopus Go peak rate

nbullus, 2025-02-10:
**Describe the bug**

The battery started charging from the grid at 00:00 using peak rates instead of 00:30 when the off-peak rate starts. Checking the Predbat log shows an error at 00:00 which then seems to have enabled the charge schedule earlier than should have been. The GivTCP log confirms the charge schedule was set at 00:00 rather than 00:30.

**Expected behaviour**

Predbat should plan and use off-peak rate to charge battery.

**Predbat version**

8.14.4

**Environment details**

- GE AIO
- Standard HAOS installer

**Screenshots**

See log files

**Log file**

Predbat

[image]

GivTCP

[image]

**Predbat debug yaml file**

None

gcoan, 2025-02-10:
could you add a logfile please to aid investigation

nbullus, 2025-02-10:
> could you add a logfile please to aid investigation

[predbat_debug_01_20_00.yaml.txt](https://github.com/user-attachments/files/18727100/predbat_debug_01_20_00.yaml.txt)

nbullus, 2025-02-10:
This issue may be related to the new tariff compare. I noticed that my current tariff was incorrectly set to Intelligent Go rather than standard Go. Not sure if it assumed I was on intelligent and tried to get intelligent rates and failed which then had some knock on effect of overriding my normal Go rates.

nbullus, 2025-02-10:
Trefor. Checking the timings of IOG, the off-peak starts at 00:00 so this is just a misconfiguration in my apps.yaml which caused it to charge early. 

However, Predbat is then using this incorrect 'current tariff' to plan my charge/discharge times. Predbat should already know what my correct current tariff is from the plugin so not sure why you need to specify it explicitly in the comparison list. Could the comparison tool not pull your current tariff in automatically to avoid typing the wrong URL in?

nbullus, 2025-02-11:
I corrected the URLs but the compare is still failing as per below log. Only the current tariff is in the compare and all the other tariffs are empty with no data:

2025-02-11 00:00:38.512899: Error: Exception raised 'str' object has no attribute 'get'
2025-02-11 00:00:38.513639: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 1013, in run_time_loop
    self.update_pred(scheduled=True)
  File "/config/predbat.py", line 772, in update_pred
    self.comparison.run_all()
  File "/config/compare.py", line 437, in run_all
    result_data = self.run_single(tariff, rate_import_base, rate_export_base, end_record, debug=debug, fetch_sensor=fetch_sensor, car_charging_slots=save_car_charging_slots)
  File "/config/compare.py", line 225, in run_single
    existing_tariff = self.fetch_rates(tariff, rate_import_base, rate_export_base)
  File "/config/compare.py", line 75, in fetch_rates
    pb.rate_import = pb.basic_rates(tariff["rates_import"], "rates_import")
  File "/config/fetch.py", line 1355, in basic_rates
    start_str = this_rate.get("start", "00:00:00")
AttributeError: 'str' object has no attribute 'get'
2025-02-11 00:00:38.522628: Info: record_status Error: Exception raised 'str' object has no attribute 'get'
2025-02-11 00:00:38.522693: Error: 'str' object has no attribute 'get'
Traceback (most recent call last):
  File "/config/hass.py", line 184, in timer_tick
    item["callback"](None)
  File "/config/predbat.py", line 1018, in run_time_loop
    raise e
  File "/config/predbat.py", line 1013, in run_time_loop
    self.update_pred(scheduled=True)
  File "/config/predbat.py", line 772, in update_pred
    self.comparison.run_all()
  File "/config/compare.py", line 437, in run_all
    result_data = self.run_single(tariff, rate_import_base, rate_export_base, end_record, debug=debug, fetch_sensor=fetch_sensor, car_charging_slots=save_car_charging_slots)
  File "/config/compare.py", line 225, in run_single
    existing_tariff = self.fetch_rates(tariff, rate_import_base, rate_export_base)
  File "/config/compare.py", line 75, in fetch_rates
    pb.rate_import = pb.basic_rates(tariff["rates_import"], "rates_import")
  File "/config/fetch.py", line 1355, in basic_rates
    start_str = this_rate.get("start", "00:00:00")
AttributeError: 'str' object has no attribute 'get'

chalfontchubby, 2026-08-20:
Thanks for the report. This issue has been open for over a year with no recent activity - we're volunteers working through a large backlog and need to filter what's still relevant. If this is still a problem for you, please let us know and we'll keep it open. Otherwise, we'll close it in a week (around 2026-08-28).

nbullus, 2026-08-21:
I can’t recreate issue now as no longer on this tariff. Just close it down as it’s so old.

closed 2026-08-21 by gcoan