springfall2008/batpred #1995: Publish only error

tayhall, 2025-02-10:
**Describe the bug**

Seeing this as status
changed to Error: Exception raised 'Compare' object has no attribute 'publish_only'

**Expected behaviour**
I'm guessing I don't have some entities correctly mapped but the product is working great and has been for some time.

Screen grab shows more entity errors 
**Predbat version**

8.14.4

**Environment details**

- givenergy
- Standard Docker
- zappi
- intelligent octopus 

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

[image]

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

**Predbat debug yaml file**
Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to predbat_debug_XX_XX_XX.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

gcoan, 2025-02-10:
@tayhall there have been a number of bug fix releases to the compare function, it would be worth you upgrading to the latest release 8.14.4 to see if this clears the problem?

springfall2008, 2025-02-15:
Should be fix already

closed 2025-02-15 by springfall2008