springfall2008/batpred #1996: Compare Chart Showing Â£ for currency symbol in compare table

tmh88, 2025-02-10:
**Describe the bug**
Compare Chart Showing Â£ for currency symbol in compare table only affects table not plans below or actual plan but shows in apps.yaml html

**Expected behaviour**
£ symbol

**Predbat version**

V8-14-4

**Environment details**

- Sunsynk Inverter and Batteries
- Standard HAOS

**Screenshots**

[image]

[image]

**Predbat apps.yaml snip**

```pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # Timezone to work in
  timezone: Europe/London

  # If you are using Predbat outside of HA then set the HA URL and Key (long lived access token here)
  #ha_url: 'http://homeassistant.local:8123'
  #ha_key: 'xxx'

  # Currency, symbol for main currency second symbol for 1/100s e.g. $ c or £ p or e c
  currency_symbols:
    - '£'
    - 'p'

springfall2008, 2025-02-15:
Fix in v8.15.1

tmh88, 2025-02-15:
> Fix in v8.15.1

Thanks

closed 2025-02-15 by tmh88