springfall2008/batpred #1997: Plan does not set discharge freeze slots unless Metric Min Improvement Export is reduced below default

mpartington, 2025-02-10:
**Describe the bug**
I have noticed during the recent better weather that my system prefers to charge to 100% on solar and then export (even if best10 has plenty of spare capacity). 

I've forced the conditions by adding a PV scale factor of 10, but whatever I change (inc turning off exert mode), I can not get it to freeze export in favour of an export later in the plan unless I reduce Metric Min Improvement Export from 5 (the recommended default) to 0.1

**Expected behaviour**
Where the plan predicts significant spare capacity, set a freeze discharge to avoid round trip losses

**Predbat version**

8.14.4

**Environment details**

- Inverter and battery setup
- Standard HAOS installer or Docker
- Anything else?

**Screenshots**
The below example has a PV scale factor of 10 to force the conditions for this behaviour

[image]

[image]

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

**Predbat debug yaml file**

0.1 Metric Min Improvement Export
[predbat_debug (3).yaml.txt](https://github.com/user-attachments/files/18741298/predbat_debug.3.yaml.txt)

5 Metric Min Improvement Export

[predbat_debug (2).yaml.txt](https://github.com/user-attachments/files/18741296/predbat_debug.2.yaml.txt)

springfall2008, 2025-02-11:
I guess the metric min improvement export probably should apply to freeze export

springfall2008, 2025-02-11:
How does this look?

[image]

springfall2008, 2025-02-11:
Fix on main for testing.

mpartington, 2025-02-11:
Your plan looks good, shame the weather isn't going to be kind with testing. 

I have turned off expert mode (to force defaults) and bumped PV scaling to 10 and the plan looks perfect.

WyndStryke, 2025-02-11:
Nice one, I didn't even realise this was why I wasn't getting freeze in the plan, I thought it was my config ...