springfall2008/batpred #1999: Override rates being ignored

onetoomany, 2025-02-11:
v8.14.4

I've a powerup this afternoon, and have this in my yaml file:

  rates_import_override:
   -  date: '2025-02-11'
      start: '14:00:00'
      end: '16:30:00'
      rate: 0
      load_scaling: 1.0

However, the rates are still showing as standard import rates.
[image]

2025-02-11 07:10:02.010761: Adding rate rates_import_override: {'date': '2025-02-11', 'start': '14:00:00', 'end': '16:30:00', 'rate': 0, 'load_scaling': 1.0} => 02-11 14:00:00 to 02-11 16:30:00 @ 0.0 date 2025-02-11 00:00:00 day_of_week [] increment False

[predbat.log](https://github.com/user-attachments/files/18749394/predbat.log)

springfall2008, 2025-02-11:
I've found the bug, if you set without the date for now and then remove it tomorrow that should work

springfall2008, 2025-02-11:
Fix on 'main' also for testing

johnwb87, 2025-02-11:
I spotted this using the manual API to inject a rate override. Came to raise the bug report and it's already solved. Thank for being so on top of it Trefor.

closed 2025-02-13 by springfall2008