springfall2008/batpred #2004: Tariff comparison ignores input_number.predbat_set_reserve_min

TonyHoyle, 2025-02-11:
**Describe the bug**
Tariff comparison ignore input_number.predbat_set_reserve_min

Fox inverters can't be set to allow discharge below 10%, so I have overridden the default 4% in the the config.
This works for normal prediction but the tariff comparison doesn't take into account that value.

(this is minor, as the difference between tariffs are normally much greater than the error introduced by this).

**Expected behaviour**
Similar behaviour to the standard prediction algorithm.

**Predbat version**

8.15.0

**Environment details**

- Fox 6kw + 10kw battery
-HAOS installer

**Screenshots**

[image]

WyndStryke, 2025-03-19:
I turned off predbat's readonly mode for the first time yesterday, and immediately had a problem with it trying to discharge below the allowable limit (Fox batteries are 90% DoD so the bottom 10% of the SoC range should never be touched).   I'm not sure if any of these settings will help, I'll see tonight:

**apps.yaml** (this setting was commented out in Fox.yaml).  Note that there are two, a number which can be used to set it, and a read-only sensor to read the pre-existing setting.  I set it to 20%.
```
# Original was commented out .  reserve - sensor name for the reserve setting in %   
# This must never be set to less than 10% (to match 90% DoD), preferably a little higher to be safe
# Also see the predbat setting input_number.predbat_set_reserve_min which should be set to 10 minimum
  reserve:
    - number.min_soc_on_grid
```

Also the following might be relevant.  Fox.yaml did not contain an entry for battery_min_soc.   

```
# Original did not contain this line.  Needed since 90% DoD and we must never drop below 10%.
  battery_min_soc:
    - 10
```

chalfontchubby, 2026-08-20:
Thanks for the report. This issue has been open for over a year with no recent activity - we're volunteers working through a large backlog and need to filter what's still relevant. If this is still a problem for you, please let us know and we'll keep it open. Otherwise, we'll close it in a week (around 2026-08-28).

WyndStryke, 2026-08-20:
> Thanks for the report. This issue has been open for over a year with no recent activity - we're volunteers working through a large backlog and need to filter what's still relevant. If this is still [...]

Is there any way to make this config setting default to 10% for Fox modbus inverters?  They're 10% DoD so having it default to 4% causes the battery to run out early, unless the new user spots the setting and corrects it.

> Set Reserve Min (Inverter 1), input_number.predbat_set_reserve_min,  input_number | 11 % | 4 % |  

and similarly the following should be battery capacity * 10%

> * Best SoC Min | input_number.predbat_best_soc_min | input_number | 2.0 kWh | 0.0 kWh |  

Fox integration settings:

* Number.min_soc
* number.min_soc_on_grid (used for the reserve)

* number.min_soc should be >= 10% and <= 100%

* number.min_soc_on_grid should be >= number.min_soc and <= 100%

chalfontchubby, 2026-08-21:
Thanks for the detail @WyndStryke - keeping this open. The Fox-specific default suggestion (tying reserve/best_soc_min defaults to the inverter's own min_soc rather than a flat 4%) is worth its own look.