springfall2008/batpred #2005: Strange behaviour around saving session

Meatballs1, 2025-02-12:
When the saving session came in Predbat planned to charge to 93%, despite only an hour of export being worthwhile during the peak period. I wasn't back to catch that plan, but after the current saving session it is planning to charge (with a limit) at the Go peak rate, only to cover peak rate load, and then head into the 8.5p overnight. After that it thinks its worthwhile to continue to charge/not cover the load into the peak period.

**Predbat version**

v8.15.0

[image]

[predbat_12_02_2025.log](https://github.com/user-attachments/files/18772414/predbat_12_02_2025.log)

[predbat_plan_12_02_2025.html.txt](https://github.com/user-attachments/files/18772420/predbat_plan_12_02_2025.html.txt)

[predbat_debug_12_02_2025.yaml.txt](https://github.com/user-attachments/files/18772423/predbat_debug_12_02_2025.yaml.txt)

springfall2008, 2025-02-13:
I just had a quick look, you have best_soc_keep set to 5.0 with a battery size of 20.0 so you are using to stay about 25% battery level which could only be achieved by holding the battery after the session.

If I turn keep down to 1.0 the plan looks more what you would expect:

[image]

Meatballs1, 2025-02-13:
Had considered this but thought it was ignored in the next 4 hours? Unclear why it still wanted to charge/hold from 0530 the next morning, a standard Octopus Go plans never do this and only appeared with the saving session plan. 

That went away after a new plan was recalculated when I checked an hour or so later.

springfall2008, 2025-02-13:
Its ignored for the next 4 hours, meaning if it falls below the keep after 4 hours time it would be considered a bad plan.

Predbat however will never charge at the highest import rate unless your export rate exceeds this, meaning only when a saving session comes along is charging at peak rate allowed, hence why keep has no impact others.

And you are right, as it gets closer to the time thing change, e.g. the period of the 4 hour window or the set of rates being considered.

Meatballs1, 2025-02-13:
Hmm should the logic check if the highest export rate has already occurred?

closed 2025-02-14 by Meatballs1