springfall2008/batpred #2009: Receiving the following status error 	Error: Exception raised 0

robertmsolway, 2025-02-13:
Since updating to v8.15.0 and adding the YAML code for the comparison feature I am getting this message in the status "Error: Exception raised 0".

It started by saying "Inverter Unable to read REST data from http://homeassistant.loocal:6345/readData - Rest will be Disabled" but has now changed to the above error message. 

Have tried to create a predbat_debug.yaml from the dashboard but get the screen "500 Internal Server Error Server got itself in trouble".

gcoan, 2025-02-13:
without your logfile or apps.yaml there is very little to go on.  There may be a problem with the latest code or the edits you have made to apps.yaml.

Try in the first instance commenting the compare configuration out of apps.yaml.

The givtcp rest warning is likely an unrelated warning.

If its still happening please supply more info so it can be investigated

chalfontchubby, 2026-08-29:
Thanks for the report. This issue has been open for over 6 months with no recent activity - we're volunteers working through a large backlog and need to filter what's still relevant. If this is still a problem for you, please let us know and we'll keep it open. Otherwise, we'll close it in a week (around 2026-09-05).

chalfontchubby, 2026-09-09:
Closing as flagged - no response to the note above, so we're assuming this is no longer live.

If you're still hitting this, please reopen or open a fresh issue with a current version number and a `predbat_debug_*.yaml` captured while it's happening; we can replay that directly against current code.

_Posted by Claude on Rik's behalf._

closed 2026-09-09 by chalfontchubby