springfall2008/batpred #2010: Strange Predbat status error

SwiftRR, 2025-02-13:
**Describe the bug**
I received a notification on my iPhone stating that I had a critical Predbat status error.
The message questioned whether Predbat had stalled. After about 10 minutes, predbat restarted and is running fine.

**Expected behaviour**
No status error with no restart!
**Predbat version**

v8.15.0

**Environment details**

- Inverter and battery setup: Givenergy AC3.0 inverter with two 8.2 kWh batteries
- HAOS on Raspberry pi4 (on external SSD): 
- Core 2025.2.2; Supervisor 2025.02.1; Operating system 14.2
- Anything else?

**Screenshots**
None relevant as all show working Predbat.

**Log file**
Log file and predbat.log attached

[6adb4f0d_predbat_2025-02-13T17-00-30.768Z.log](https://github.com/user-attachments/files/18787578/6adb4f0d_predbat_2025-02-13T17-00-30.768Z.log)

[predbat-3.log](https://github.com/user-attachments/files/18787586/predbat-3.log)

**Predbat debug yaml file**
I have run debug file but this is after the event so I am unsure whether this will be useful.
Attached below.

[predbat_debug_17_00_00.yaml.txt](https://github.com/user-attachments/files/18787711/predbat_debug_17_00_00.yaml.txt)

gcoan, 2025-02-13:
It looks like the failure is a web socket connect error at 16:39

What then happened is predbat stalled and the predbat error checking automation (that's in the documentation) picked up the error and restarted predbat and it all then worked fine

The workaround to stop the web socket errors is to create a ha_key for predbat to use - see updated documentation https://github.com/gcoan/batpred/blob/main/docs/apps-yaml.md#home-assistant-connection

SwiftRR, 2025-02-13:
Thanks. 

I am running predbat standalone and the documentation suggests that ha_url would then not be needed.
I have set up ha_url anyway in apps.yaml, with a long-lived access code which I have generated using the instructions in your documentation. 
I will also look at the recent discussion in the GE community forums related to ha_url (which I have never used before – indeed it was not in my apps.yaml, perhaps a relic from appdaemon days).

Yesterday I couldn't access ha from my iPad, with HA cycling back and forth with nothing in the address bar. If I then started the url with 'h', the address bar was populated and ha loaded. This looks like it may be linked.

I do have a static IP address assigned to HA which works. 
Would I do better using the static IP address rather than homeassistant.local:8123 ?

This episode has been a good test for the predbat error monitor automation which I obviously had set up. The notification came up with the error messages in the automation and predbat restarted.
Without this automation, predbat for me would just have died!

I will now monitor to see if there is any reoccurrence of the issue.

Rob

gcoan, 2025-02-13:
> I do have a static IP address assigned to HA which works.
> Would I do better using the static IP address rather than homeassistant.local:8123 ?

yes

it reduces the DNS lookups from every call from Predbat to HA needing to find the IP address.  That's what I have setup.

There have been a few people (myself included) that have had the websocket error.  Since I added the ha_key I have not seen it happen since.

One side effect of setting ha_url is that the pathnames to the logfile, debug file, etc in the logfile are now wrong (the logic that finds the slug id breaks if you set ha_url).  Fix for this coming in my next PR but this is minor, only affects the logging

SwiftRR, 2025-02-13:
Thanks

In apps.yaml, 
for ha_url, should I replace homeassistant.local:8123 with the static IP address with :8123, 
i.e. 192.168.68.XX:8123 ?

Rob

gcoan, 2025-02-14:
> In apps.yaml,
> for ha_url, should I replace homeassistant.local:8123 with the static IP address with :8123,
> i.e. 192.168.68.XX:8123 ?

Yes

And you don't need to obfuscate your IP address.

192.168.x.x is a block of addresses reserved for private networks, they are not internet routable or accessible. Most people's home networks are on 192.168 address ranges, usually 192.168.0.x or 192.168.1.x.

Knowing what the individual address of your HA server is is of no use to anyone unless they are able to connect onto your home network, and in which case you have bigger security concerns to worry about !

SwiftRR, 2025-02-14:
I changed the ha_url line in my apps.yaml to:
`  ha_url: '192.168.68.74:8123'`

Some error again, I am afraid.
Predbat keeps trying to restart but hitting a new exception, each time.
I am letting it cycle through each exception.

Strangely, I have the home assistant app running on my iMac and predbat loads just fine.
The 'server' is set correctly to `192.168.68.74:8123`.
Predbat is also running just fine in the companion app on my iPhone and iPad.

I have also found that I can no longer access the webUI.

I have now changed the ha_url line in my apps.yaml back to :
`  ha_url: 'homeassistant.local:8123'`
Predbat has now started again properly 

I am attaching the logs.

[6adb4f0d_predbat_2025-02-14T09-19-21.904Z.log](https://github.com/user-attachments/files/18796512/6adb4f0d_predbat_2025-02-14T09-19-21.904Z.log)

[predbat-4.log](https://github.com/user-attachments/files/18796549/predbat-4.log)

Incidentally, I updated HA 2 days ago to v2015.2.2. Whether this has anything to do with my issue?

Thanks

Rob

gcoan, 2025-02-14:
> I changed the ha_url line in my apps.yaml to:
> ha_url: '192.168.68.74:8123'

That won't work

> I have now changed the ha_url line in my apps.yaml back to :
> ha_url: 'homeassistant.local:8123'

That's not valid either, am surprised it works. Predbat is probably ignoring it.

You need to set ha_url to a valid URL, i.e. 

ha_url: 'http://192.168.68.74:8123'

Am a bit confused about you saying predbat works fine in the companion app but not in other ways.  The companion app is just an embedded web browser to enable connection to Home Assistant, its not running predbat, its just a way to of accessing HA where Predbat is running as an addon

Fix ha_url to be a valid URL and it will work, promise !

SwiftRR, 2025-02-14:
Thanks.

I now have ha_url in apps.yaml set with http:// at the start.
I hope this will have sorted things and that your predbat error automation isn't triggered again. 

It is confusing that browsers (Safari and Chrome) don't show http:// before an IP address but they do show http:// before homeassistant.local and other sites (as long as they are not just IP addresses.) _But all my other urls are https rather than http_

I have been following the thread discussion about ha_url with @wavydavy and @piansom (and you).

I will report back after monitoring again, after which we should be able to close the issue.

Rob

closed 2026-08-28 by chalfontchubby