springfall2008/batpred #2012: iBoost overnight ready by time.

johnwb87, 2025-02-14:
**Is your feature request related to a problem? Please describe.**
I've just enabled iBoost for the first time. I am using the Smart Option and I am on IOG so long overnight rate. Predbat plans to heat water immediately in the cheap overnight period. This will increase the standing losses of the tank as hot water demand isn't until the morning.

**Describe the solution you'd like**
To have something like a ready by time when using the smart option to push iBoost as late as possible to optimise standing losses.

**Describe alternatives you've considered**
Currently just turn iBoost off, or change some settings so Predbat just sees the energy sensor.

**Additional context**
I'm trying to integrate as much as possible into predbat so that it has best visibility of what is going on in the house.

johnwb87, 2025-02-14:
I guess I can mostly achieve what I want by adding the energy sensor to the car_charging_energy and stick with a fixed timer.

closed 2025-02-14 by johnwb87

gcoan, 2025-02-14:
there's a custom sensor that will do this in the next PR that I'll push through:

https://github.com/gcoan/batpred/blob/main/docs/output-data.md#iboost-solar-diverter-data

johnwb87, 2025-02-14:
@gcoan Are you referring to iboost_best? If i understand that correctly that tells me when iBoost is expected but won't have any effect on the planning.

Thanks.

gcoan, 2025-02-14:
> @gcoan Are you referring to iboost_best? If i understand that correctly that tells me when iBoost is expected but won't have any effect on the planning.
> 
> Thanks.

yes you are correct, the template sensor will tell you when its scheduled, it doesn't affect the plan

sorry