springfall2008/batpred #2013: Lock-up due to web Socket issue

StratoCumulus450, 2025-02-14:
I have encountered a number of Predbat 'lock-ups' this week, which I have resolved by restarting Predbat. Last night my battery did not charge as Predbat stalled without me noticing.

I have found the following repeated in my Predbat log files, but unfortunately it means nothing to me;-

2025-02-13 06:54:28.517104: Warn: Web Socket closed, will try to reconnect in 5 seconds
2025-02-13 06:54:33.521546: Info: Start socket for url http://supervisor/core/api/websocket
2025-02-13 06:54:33.529198: Error: Web Socket exception in startup: 502, message='Invalid response status', url=URL('http://supervisor/core/api/websocket')
2025-02-13 06:54:33.529456: Error: Traceback (most recent call last):
  File "/config/ha.py", line 184, in socketLoop
    async with session.ws_connect(url) as websocket:
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 1167, in __aenter__
    self._resp = await self._coro
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 821, in _ws_connect
    raise WSServerHandshakeError(
aiohttp.client_exceptions.WSServerHandshakeError: 502, message='Invalid response status', url=URL('http://supervisor/core/api/websocket')

My configuration is as follows;

Home Assistant OS 14.2 (add-on)
Home Assistant Core 2025.2.3
Predbat 8.15.0
Compute platform RPi5 with 256Gb SSD
GivEnergy Hybrid Inverter and 9.5kW Gen 2 battery

gcoan, 2025-02-14:
try creating a long lived access token in HA and setting this and ha_url in apps.yaml

https://github.com/gcoan/batpred/blob/main/docs/apps-yaml.md#home-assistant-connection

StratoCumulus450, 2025-02-14:
Thank you. I have made the change and will see what happens over the coming days and report back.

StratoCumulus450, 2025-02-18:
All stable now after long lived access token change.