springfall2008/batpred #2014: Error: -1800

nblount, 2025-02-14:
**Describe the bug**
Predbat appears to be stuck, not generating any plan, logs keep repeating -1800 error.

**Expected behaviour**
Not giving error and generating plan

**Predbat version**
8.15.0

**Environment details**
- GivEnergy 5kW + 3 Batteries 9.5kW
- Standard HAOS installer

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
18095 | 2025-02-14 00:15:01.530908: Error: Traceback (most recent call last):
18094 | 2025-02-14 00:15:01.530650: Error: Exception raised -1800
18063 | 2025-02-14 00:10:01.425051: Error: -1800
18062 | 2025-02-14 00:10:01.425005: Info: record_status Error: Exception raised -1800
18060 | KeyError: -1800
18051 | 2025-02-14 00:10:01.417664: Error: Traceback (most recent call last):
18050 | 2025-02-14 00:10:01.417581: Error: Exception raised -1800
18019 | 2025-02-14 00:05:00.907847: Error: -1800
18018 | 2025-02-14 00:05:00.907818: Info: record_status Error: Exception raised -1800
18016 | KeyError: -1800
18007 | 2025-02-14 00:05:00.902518: Error: Traceback (most recent call last):
18006 | 2025-02-14 00:05:00.899860: Error: Exception raised -1800
2558 | 2025-02-13 18:51:50.204146: Warn: Web Socket closed, will try to reconnect in 5 seconds

Have rebooted HA, tried rebuilding the Predbat Addon, restarted Predbat. Issues appear to have started around midnight.

nblount, 2025-02-14:
More info from logs, looks like it may have something to do with Octopus Saving slots?:

2025-02-14 00:05:00.899838: Setting Octopus saving session in range 02-12 17:30:00 - 02-12 18:30:00 export False rate 12.0
2025-02-14 00:05:00.899860: Error: Exception raised -1800
2025-02-14 00:05:00.902518: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 1021, in run_time_loop
    self.update_pred(scheduled=True)
  File "/config/predbat.py", line 574, in update_pred
    self.fetch_sensor_data()
  File "/config/fetch.py", line 1067, in fetch_sensor_data
    self.load_saving_slot(self.octopus_saving_slots, export=False, rate_replicate=self.rate_import_replicated)
  File "/config/octopus.py", line 283, in load_saving_slot
    self.rate_import[minute] += rate
KeyError: -1800

nblount, 2025-02-14:
Disabled the following lines in the apps.yaml and we are back to working:
#octopus_saving_session: 're:(binary_sensor.octopus_energy([0-9a-z_]+|)_saving_session(s|))'
#octopus_saving_session_octopoints_per_penny: 8

springfall2008, 2025-02-15:
Fix in v8.15.1

closed 2025-02-15 by springfall2008