springfall2008/batpred #2015: FR: Direct web links to each Predbat web console page

gcoan, 2025-02-15:
**Is your feature request related to a problem? Please describe.**
I have on my dashboard a link to the Predbat web console, vis:

```
- type entities:
  entities:
    - type: weblink
      name: Predbat Web Console
      url: /hassio/ingress/6adb4f0d_predbat?Compare
```

I'd like the ability to be able to have links to specific dashboard pages.

By trial and error I have found that:

url: /hassio/ingress/6adb4f0d_predbat?1 goes to the Compare page (as does ?3 and ?4)

url: /hassio/ingress/6adb4f0d_predbat?Compare goes to the Plan page

url: /hassio/ingress/6adb4f0d_predbat?compare goes to the Config page

I'd ideally like these to be logical links that can be added to the documentation.

It would also be useful if the compare table could be embedded in a dashboard itself, e.g. as a html table, or a web clip

But just being able to know and safely use specific URL's would be a great start

**Describe the solution you'd like**
See above

**Describe alternatives you've considered**
Manually have to navigate to the right page

Use undocumented URLs that "seem to work"

springfall2008, 2025-02-15:
The list of pages is in the code

So its /plan, /log, /compare etc...

        app.router.add_get("/", self.html_index)
        app.router.add_get("/plan", self.html_plan)
        app.router.add_get("/log", self.html_log)
        app.router.add_get("/menu", self.html_menu)
        app.router.add_get("/apps", self.html_apps)
        app.router.add_get("/charts", self.html_charts)
        app.router.add_get("/config", self.html_config)
        app.router.add_get("/dash", self.html_dash)
        app.router.add_get("/debug_yaml", self.html_debug_yaml)
        app.router.add_get("/debug_log", self.html_debug_log)
        app.router.add_get("/debug_apps", self.html_debug_apps)
        app.router.add_get("/debug_plan", self.html_debug_plan)
        app.router.add_get("/compare", self.html_compare)

gcoan, 2025-02-16:
Thanks Trefor, I sort of thought they'd be logically structured, only I can't work out how to get them to work correctly!

url: /hassio/ingress/6adb4f0d_predbat/plan

gives an error message

[image]

I tried it with urls of /plan and /6adb4f0d_predbat/plan etc and they took me to the default dashboard each time

springfall2008, 2025-02-18:
My charts link looks like this:
[Charts](http://homeassistant.local:8123/api/hassio_ingress/iqUrZxmmSLs3rbXHF9OWgA6hpNRxPVnDL62FNi-7RHo/charts)

gcoan, 2025-02-18:
Hmm, OK, but I don't know how to create that URL!
It looks like it has some kind of internal id within it, I tried it on mine and it didn't work http://homeassistant.local:8123/api/hassio_ingress/iqUrZxmmSLs3rbXHF9OWgA6hpNRxPVnDL62FNi-7RHo/charts

503: Service Unavailable

If I go to addons/predbat/open web ui I am presented with the URL http://homeassistant.local:8123/hassio/ingress/6adb4f0d_predbat

If I then add /charts at the end I get the [hassio error above](https://github.com/springfall2008/batpred/issues/2015#issuecomment-2661168162)

gcoan, 2026-08-20:
I never did get this to work, but can live without it

closed 2026-08-20 by gcoan