springfall2008/batpred #2018: Exception: Battery size from REST is 0.0

SwiftRR, 2025-02-15:
**Describe the bug**
I had an exception in the early hours of today (15th Feb) which cleared itself.

**Expected behaviour**
No exception with no restart!

**Predbat version**
v8.15.0

**Environment details**
Inverter and battery setup: Givenergy AC3.0 inverter with two 8.2 kWh batteries
HAOS on Raspberry pi4 (on external SSD):
Core 2025.2.2; Supervisor 2025.02.1; Operating system 14.2

**Screenshots**
Add in log showing exception
[image]

**Log file**
_predbat.log attached._
The exception extract above is clear at 03:01 am 15th February 2025
[predbat.1.log](https://github.com/user-attachments/files/18810920/predbat.1.log)

**Predbat debug yaml file**
[predbat_debug_12_40_00-2.yaml.txt](https://github.com/user-attachments/files/18810924/predbat_debug_12_40_00-2.yaml.txt)

Another user has had the same problem, which I hope will follow my issue here

Rob

springfall2008, 2025-02-15:
I'm not sure how to address this, it appears GivTCP returned a bad value, the next run was okay. Any errors in the givtcp logs?

SwiftRR, 2025-02-15:
I have attached my givTCP log.

[533ea71a_givtcp_2025-02-15T14-32-53.048Z.log](https://github.com/user-attachments/files/18811053/533ea71a_givtcp_2025-02-15T14-32-53.048Z.log)

There is nothing showing at the time of my exception (although plenty of glitches over the last few days)!
Something doesn't quite seem right.
I am on givTCP 3.0.4 which seems to be the most stable v3 (v3 has had a lumpy start!)

I have copied below the predbat log extract from another user so that you are able to compare with mine.
The ERROR title was also: `Error: Reported battery size from REST is 0.0, but it must be >0`
```
2025-02-14 09:12:21.476556: Error: Exception raised
2025-02-14 09:12:21.480546: Error: Traceback (most recent call last):
File "/config/predbat.py", line 967, in update_time_loop
self.update_pred(scheduled=False)
File "/config/predbat.py", line 594, in update_pred
self.fetch_inverter_data()
File "/config/execute.py", line 636, in fetch_inverter_data
inverter = Inverter(self, id)
File "/config/inverter.py", line 356, in init
raise ValueError
ValueError
2025-02-14 09:12:21.515728: Info: record_status Error: Exception raised
2025-02-14 09:12:21.515957: Error:
Traceback (most recent call last):
File "/config/hass.py", line 184, in timer_tick
item["callback"](https://community.givenergy.cloud/d/4986-experiences-with-givtcp3-and-predbat/None)
File "/config/predbat.py", line 973, in update_time_loop
raise e
File "/config/predbat.py", line 967, in update_time_loop
self.update_pred(scheduled=False)
File "/config/predbat.py", line 594, in update_pred
self.fetch_inverter_data()
File "/config/execute.py", line 636, in fetch_inverter_data
inverter = Inverter(self, id)
File "/config/inverter.py", line 356, in init
raise ValueError
ValueError
```

The other clue is that I upgraded to HA 2025.2.2, two days ago. 
HA upgrades often seem to throw up issues.
I also upgraded to Predbat 8.15.0 when it was released 4 days ago.
I have noticed that you have just put up v8.15.1 and I will upgrade to that after sending you this comment. 

Rob

henry3rd1961, 2025-02-15:
Hi, I had the same issue/
2025-02-14 09:12:21.476449: Error: Reported battery size from REST is 0.0, but it must be >0
2025-02-14 09:12:21.476556: Error: Exception raised
2025-02-14 09:12:21.480546: Error: Traceback (most recent call last):
File "/config/predbat.py", line 967, in update_time_loop
self.update_pred(scheduled=False)
File "/config/predbat.py", line 594, in update_pred
self.fetch_inverter_data()
File "/config/execute.py", line 636, in fetch_inverter_data
inverter = Inverter(self, id)
File "/config/inverter.py", line 356, in init
raise ValueError
ValueError
2025-02-14 09:12:21.515728: Info: record_status Error: Exception raised
2025-02-14 09:12:21.515957: Error:
Traceback (most recent call last):
File "/config/hass.py", line 184, in timer_tick
item["callback"](https://community.givenergy.cloud/d/5411-second-year-live-on-predbat/None)
File "/config/predbat.py", line 973, in update_time_loop
raise e
File "/config/predbat.py", line 967, in update_time_loop
self.update_pred(scheduled=False)
File "/config/predbat.py", line 594, in update_pred
self.fetch_inverter_data()
File "/config/execute.py", line 636, in fetch_inverter_data
inverter = Inverter(self, id)
File "/config/inverter.py", line 356, in init
raise ValueError
ValueError

henry3rd1961, 2025-02-15:
Similarly there were no GIVTCP errors at the time.

henry3rd1961, 2025-02-15:
I have not upgraded to the last two versions, so we can eliminate the version perhaps?

springfall2008, 2025-02-15:
Does the issue go away after a small time?

SwiftRR, 2025-02-15:
I think we are in the position where we keep an eye on logs and monitor to see if the issue goes away!

Rob

henry3rd1961, 2025-02-15:
Agreed, I have not had the issue since.

closed 2025-02-18 by springfall2008