springfall2008/batpred #2021: FR: Use Ohme integration action to get IOG Slots

jon798, 2025-02-16:
Since the Ohme integration was moved into HA Core some of the sensors are no longer provided and I am seeing the Intelligent Dispatching sensor not returning any planned slots, only completed slots.

In the docs for the Ohme integration there’s now an action: ohme.list_charge_slots which seems to return slots still in progress, at least, and accurate to 15mins (matches the Ohme app, anyway).

This seems to return past, present and future slots as long as the car is plugged in - this might be a good way to bring info into Predbat?

Thanks for your ongoing efforts on this excellent add on!

springfall2008, 2025-02-18:
I've not had anyone else report this as an issue, can you share some more details on the list_charge_slots and also what other sensors you have?

murkle, 2025-02-19:
@springfall2008 There's a new version of the Ohme integration (with some things currently missing/work-in-progress). danr is going to look at a proper predbat integration, see https://github.com/dan-r/HomeAssistant-Ohme/issues/108#issuecomment-2663208220

BenCathro, 2025-03-16:
I'm also in this situation. Currently if I get an IOG slot outside the normal window my home battery dumps into the car and I can't figure out how to pull the planned sessions into Predbat now. Has anyone figured this out?

chris2172, 2025-03-16:
I believe this is dependent on the actual charger.
I have a Zappi, and when I get slots outside the off-peak period, this is reflected in the Octopus plugin.
This is not the case with Ohme, which my in-laws have.
In my situation, predbat will recalculate when it sees the additional 7p slots but won't for the in-laws.
What I would suggest is to set up an automation to:
When Power to the car > 100W
Set Predbat select.predbat_mode to "Monitor"
Start Home battery charging or set home battery to hold as current SOC
When Power to the car = 0W
Do the reverse.

springfall2008, 2025-03-16:
For Ohme you need to configure differently

https://springfall2008.github.io/batpred/devices/#new-ohme

BenCathro, 2025-03-16:
> I believe this is dependent on the actual charger.
> I have a Zappi, and when I get slots outside the off-peak period, this is reflected in the Octopus plugin.
> This is not the case with Ohme, which my [...]

Thanks for the suggestion. This would stop the battery exporting to the car but would require a different work around to get Predbat to charge the battery whenever an IOG slot is active.

It looks like @springfall2008 has figured out a solution. Thank you for dedicating time to this. I've implemented your changes and will see how things look for my next car charging session.

One quick question:

ohme_name: 're:time.ohme_(.+)_target_time'

This line is new. I'm not sure what its purpose is in relation to the incrementing energy sensor. Does it define when the sensor should reset?

closed 2026-08-28 by chalfontchubby