springfall2008/batpred #2025: Saving Session Join Error

JonathanLew1s, 2025-02-17:
**Describe the bug**
Prebat cannot join saving session.

**Expected behaviour**
Join the session automatically.

**Predbat version**

8.15.1

**Log file**
2025-02-17 12:40:13.660452: Warn: Service call octopus_energy/join_octoplus_saving_session_event data {'event_code': 'EVENT_6_170225', 'entity_id': 'event.octopus_energy_a_*******_octoplus_saving_session_events'} failed with response None

gcoan, 2025-02-17:
It worked ok for me

Have you signed up to Octopoints which is a pre-requisite for the saving sessions?

What happens if you manually sign up in the Octopus app/website?

JonathanLew1s, 2025-02-17:
Been doing this for the past 3 events, just retries every 5 minutes and never actually signs up. I will check the octoplus account in case anything changed since the saving sessions last winter.

Recreated apps.yaml based on latest template and nothing obviously wrong there, the event entity on octopus energy integration is active and appears working. Only thing I can think is the octopus energy join action is somehow borked on the bottlecapdave integration, will reinstall tomorrow and try signup on the web on the next event.

gcoan, 2025-02-17:
yes it does rather suggest a problem on the Octopus Integration or the Octopus API.

If it worked last year then you should be already signed up to Octopoints from last year.

I looked at the Octopus integration documentation and nothing obvious about things needing to be enabled or configured other than the Octopoints signup 

https://bottlecapdave.github.io/HomeAssistant-OctopusEnergy/services/#octoplus

closed 2025-02-18 by JonathanLew1s

gcoan, 2025-02-18:
glad you found the issue

enjoy today's massive 13p saving session !