springfall2008/batpred #2027: Error: Exception raised 'SOC_kWh'

thewookiewon, 2025-02-18:
**Describe the bug**
Error encountered in the log file immediately after a restart of predbat

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Predbat version**

1.2.4 addon

**Environment details**

- GivEnergy 3.6kw hybrid + 9.5 & 5.2 batteries
- Standard HAOS installer
- Raspberry Pi 5

**Screenshots**
NA

**Log file**
2025-02-18 18:00:36.524351: Error: Exception raised 'SOC_kWh'
2025-02-18 18:00:36.526248: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 970, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 574, in update_pred
    self.fetch_inverter_data()
  File "/config/execute.py", line 640, in fetch_inverter_data
    inverter.update_status(self.minutes_now)
  File "/config/inverter.py", line 847, in update_status
    self.soc_kw = dp3(self.rest_data["Power"]["Power"]["SOC_kWh"] * self.battery_scaling)
KeyError: 'SOC_kWh'
2025-02-18 18:00:36.541392: Info: record_status Error: Exception raised 'SOC_kWh'
2025-02-18 18:00:36.541550: Error: 'SOC_kWh'
Traceback (most recent call last):
  File "/config/hass.py", line 190, in timer_tick
    item["callback"](None)
  File "/config/predbat.py", line 976, in update_time_loop
    raise e
  File "/config/predbat.py", line 970, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 574, in update_pred
    self.fetch_inverter_data()
  File "/config/execute.py", line 640, in fetch_inverter_data
    inverter.update_status(self.minutes_now)
  File "/config/inverter.py", line 847, in update_status
    self.soc_kw = dp3(self.rest_data["Power"]["Power"]["SOC_kWh"] * self.battery_scaling)
KeyError: 'SOC_kWh'

**Predbat debug yaml file**
Will upload later

thewookiewon, 2025-02-18:
[predbat_debug.yaml.txt](https://github.com/user-attachments/files/18851480/predbat_debug.yaml.txt)

thewookiewon, 2025-02-18:
Looks like I had to manually restart the GivTCP addon too, wonder if your recent changes to self heal Predbat might need to toggle a restart of GivTCP too?

springfall2008, 2025-02-18:
I'll also add code to prevent this from causing a crash

springfall2008, 2025-02-18:
Fix on main