springfall2008/batpred #2033: Import/Export Compare values & not taking account of rate overrides

gcoan, 2025-02-21:
**Describe the bug**
Three different issues noticed with the compare function tonight:

1. The Compare shows current is much cheaper than Cosy Agile that I'm currently on.

On investigation I have identified that current is taking account of rate overrides in apps.yaml for a power up event today:
[image]

Whereas the other projections are not.  I'd suggest that if a rate override has been put in apps.yaml or via the manual_api then it should be honoured in all tariffs
[image]

2. Import/export values look impossibly high.

Noticed import projections of over 100kWh, and export projections that can't be achieved either.
[image]

e.g. Agile Fixed has 13.5kW of projected export.

In the plan there are 4 export slots.  I have 2 inverters at 2.4kW, so 4.8 per hour. At best 9.6 + solar - house load.  
2 of the slots are at night, two are in the day but when load>solar, so can't achieve this export figure.
[image]
[image]

import not checked but similarly way too high

3. Viewability on a mobile phone

The graphs are completely unreadable both portrait and landscape:
[image]
[image]

I'd question whether the title/menu bar needs to be so big and needs to be fixed on small screens like this as it uses quite a bit of real estate

**Expected behaviour**
1. Take account of rate import/export overrides in apps.yaml for all tariffs

2. Import and export figures that are correct !

3. Make the menu bar float on the top of the page when on small resolution screens and scale the graph so that it is readable on all devices

**Predbat version**
8.14.4

**Environment details**
2x Gen 1 hybrid inverters
HAOS Core 2025.2.4

**Log file**

gcoan, 2025-02-22:
r/e point 3, if the text 'Predbat' were made a bit smaller and the bat logo was aligned a bit higher then the whole menu bar could be halved in vertical size quite easily

still consideration about making it scroll off the page rather than be fixed at the top on small apperture screen sizes, but just reducing the vertical size would make quite an improvement

jrwm47, 2025-02-23:
Like Geoff, I'm getting some odd-looking numbers in my Comparison table. I'm on Agile In/Fixed Out with a GE AIO (13k5Wh) and about 3k5Wp panels, no EV, iboost currently off. Predbat is set to vanilla (default, non-expert yaml).

For the past few days, my Current has agreed with Agile In/Fixed Out but today it doesn't - see attached. Also the un-dimensioned In/Out Comparison numbers look high. It's windy and predbat was doing a lot of 30/30 arbitrage earlier today.

Despite these niggles, I love predbat and welcome the new tariff comparison feature. Happy to provide any debug data might help. - Julian

[image]

[image]

gcoan, 2025-02-27:
Sorry @springfall2008 three further observations/bugs/features:

4. Is it possible for the colours in the chart to be adjusted?  In particular I find that the 'current' and 'flux' always come out at the same shade of blue so its very hard to distinguish between them:
[image]

5. The compare entities all have identical friendly names in HA, can these be changed to include the tarrif name because at the moment they are indistinguishable other than the different entity id's:
[image]

6. And on the subject of the compare entities, if I look at the entity value I can see a snapshot of entity state over time, but with a spinning wheel in the Logbook area:
[image]

If I click 'show more' to see a graph of compare £ over time then it sits with a spinning wheel for ages and ages before eventually reporting 'no logbook entries'.
Don't have this problem with other Predbat entities, must be something to do with the way these compare entities are written to HA.

gcoan, 2025-09-18:
Have you had a chance to look at this @springfall2008 ?   

These are all still issues occurring with the compare function 

thanks

springfall2008, 2026-09-17:
Automated first-pass triage (a maintainer will review before any action is taken).

**Classification:** mixed — the headline request (point 1) is an **enhancement** (design decision on which user overrides apply to compare scenarios); point 2 may still be a bug but can't be verified from the attached data; points 3/4 are enhancements; point 6 is unclear. Priority not set while the headline is a feature request.

Investigated against current main (`002f6b86`, v9.0.3-35) — the attached log confirms this report is from **v8.14.4, Feb 2025**, so ~19 months of compare changes separate the report from main. I read the code paths for each sub-point; I did not run the test suite, as `test_compare.py` exercises the scenario plumbing and settles none of the six claims below.

**Per sub-point:**

1. **Rate overrides not honoured in other tariffs** — confirmed as current behaviour on main, and part of it is deliberate. `run_all()` snapshots the *live* rate tables (including any `rates_import_override`/`rates_export_override` entries in apps.yaml) as the base (`compare.py:598-599`), which is why "current" shows the Power Up override. Each scenario then resets to that base and, if the tariff defines its own `rates_import`/`rates_export`, replaces the table wholesale with no `prev` passed (`compare.py:100-104` and `:130-133`), dropping the apps.yaml overrides. Excluding *manual_api* overrides from scenarios is documented as intentional (`fetch.py:2129-2131`: "the live system's manual API overrides apply to the actual running plan and must not leak into those simulations"). Whether apps.yaml *static* overrides should propagate to every simulated tariff is a design decision for the maintainer. Workaround today: each tariff entry can carry its own `rates_import_override`/`rates_export_override` block (`compare.py:138-147`), so the Power Up window can be replicated per tariff.

2. **Import/export values look impossibly high** — I could not verify the specific numbers from the Feb 2025 log, but there is a systematic explanation worth checking before treating it as a bug: the Import/Export columns are accumulations over **every minute of the full 48-hour scenario horizon**, not sums over the plan's slots — `prediction.py:1340` and `:1353` add to them whenever the modelled grid flow imports/exports, including PV export outside any planned slot and arbitrage charging the scenario plan chooses under the simulated tariff. So "the plan has 4 export slots × 2 inverters" doesn't bound the Export column, and a scenario plan that arbitrages harder than the live plan will legitimately show more import/export. A "day slots can't export because load > solar" assumption also doesn't hold — a planned discharge slot exports battery power regardless of load vs solar. If the values still look physically impossible on the current version, I'd need fresh data to take it further (see request below).

3. **Mobile viewability** — still valid as an enhancement; the compare page has no small-screen layout handling.

4. **Chart colours (current vs Flux indistinguishable)** — still valid as an enhancement.

5. **Identical friendly names** — the code appends the tariff name: `"friendly_name": "Compare " + name` (`compare.py:474`, unchanged since v8.14.4). If entities still show identical names today, the `name:` values in the `compare_list` entries themselves are likely duplicated — that would be a configuration fix, not a code one. A current screenshot would settle it.

6. **Logbook spinning on compare entities** — compare entities are written through the same `dashboard_item()` → `set_state` path as all other Predbat entities (`compare.py:485`, `output.py:3828`), so there's nothing compare-specific in the write path that would explain it. This looks like an HA-side behaviour of the 2025.2-era UI; if it still occurs on a current HA version a fresh repro (HA version + which entity) would be needed.

**Information request:** for point 2 to progress as a bug it needs a current `predbat_debug.yaml` (none was attached, and the log is from Feb 2025 / v8.14.4) plus a screenshot of the comparison table on the current version — the same applies to point 5 if the names still collide.

gcoan, 2026-09-17:
Thanks for asking the BOT to look at this Rik.

On the points questioned above,

1. Other people have raised this, agree it is a design decision as to whether rate overrides are included or not and I would agree, if you have manually configured rate overrides in apps.yaml then these should only be used for the current tariff.
But if there is an Axle or a zero price override configured in select.predbat_manual_api then not taking it into account in the compare means the compare is invalid as these are clearly forthcoming rates that all tariffs should be considering.  Adding these kind of rate overrides to the compare rates isn't practical.  

2. The import/export figures still look impossibly high:

[image]

Taking current as an example, 23kWh exported, the plan shows no force export and only 15kWh of solar generation, and some of that goes to charging up the batteries and on house load.  I would expect the import/export figures to represent the 24 hours of the compare plan, not a 48 hour plan horizon that isn't shown anywhere on the compare screen.  This feels like a bug to me.

[image]
[image]

Logfile: 

[predbat.log](https://github.com/user-attachments/files/32356114/predbat.log)

3. and 4.  As above, valid enhancements 

5. Looks like the friendly names have been updated to include the tariff name at some point in the past, so this is done

[image]

6. Also looks to be resolved, can view the history of the compare entities with no issues now

[image]

springfall2008, 2026-10-07:
Automated triage failed to complete for this issue - see the triage bot's logs for details. Not retrying automatically; remove `BOT_FAILED` and re-add `BOT_REVIEW` to try again.

chalfontchubby, 2026-10-11:
Axle overrides would be valid across tariffs (since it's not your own provider) - but other things like octopus specific overrides are more interesting. Some assume you are octopus only - so would not effect compares against eon etc. Some might only apply to a specific octopus tariff. (can't think of an example - but knowing how messy octopus stuff is - who knows. Car sharging slots for IOG might be a case in point). Gets messy.

gcoan, 2026-10-11:
Agreed, it could get messy with rate overrides such as for Octopus Power Up/Down events and doing a compare against Eon tariffs.

For me that's not a scenario I would get to because all my compares are for Octopus tariffs.

Definitely agree Axle ought to be included in the compare for the reason you say.  To be honest I would rather it be documented as a limitation that any rate overrides are applied to all tariffs regardless of whether they are valid or not for that provider because I'd prefer to see what the plan is on say Flux with a power down vs my current tariff with a power down.  The current situation ends up with a different set of inconsistencies between the Current plan and any of the plan compares.