springfall2008/batpred #2034: Charge/export cycles being planned despite predicted net positive cost

tomkentpayne, 2025-02-21:
**Describe the bug**

Predbat never used to plan exports before today or yesterday (I've had it running ~1.5months), only doing so during saving sessions, but it has planned a couple of a export slots overnight that don't seem to make sense by its own predictions, which over the period from 23:00 - 01:30 show a net cost of 50p despite the net state of charge dropping by 5% and the predicted load totalling 1.02kWh, which at the average import price would only cost ~13.4p if running straight off the grid.

**Expected behaviour**

Predbat to not export randomly overnight, and instead only charge whatever amount is necessary to cover the next day's predicted solar defecit.

**Predbat version**

8.15.1

**Environment details**

- Inverter and battery setup
  - Growatt SPH inverter, 3.6kW, and Growatt Ark LV battery, ~ 7.5kWh.
- Standard HAOS installer or Docker
  - Standard HAOS, running under Proxmox
  - Predbat installed in standard way as HA add-on
- Anything else?
  - Non-default config items:
    - `predbat_best_soc_keep` set to 1.0kWh
    - `predbat_set_reserve_min` set to 15%
    - `predbat_octopus_intelligent_charging` set to False
    - `predbat_set_charge_low_power` set to True (But this doesn't work due to `charge_rate` missing from apps.yaml)
    - `predbat_set_export_freeze` set to False

**Screenshots**

Plan (annotated to show what I mean) - [image]

**Log file**

Log - [predbat.log](https://github.com/user-attachments/files/18907273/predbat.log)

**Predbat debug yaml file**

[predbat_debug_12_50_00.yaml.txt](https://github.com/user-attachments/files/18907286/predbat_debug_12_50_00.yaml.txt)

**apps.yaml file**

[apps.yaml.txt](https://github.com/user-attachments/files/18907353/apps.yaml.txt)

Many thanks in advance to anyone that can help me diagnose this problem, or can tweak things in Predbat to fix it :)

HeavyPXL, 2025-02-24:
I think you might be missing the standing charge in your cost calculation. Assuming a 65p daily rate included in the Sat 00:00 time slot it looks like you were predicted to be about 15p in profit over that period.

If you want to bias Predbat against export you can use the "Metric Min Improvement Export" setting in expert mode to raise the profit margin Predbat considers worth exporting for. Default is 5p.

gcoan, 2025-02-24:
> If you want to bias Predbat against export you can use the "Metric Min Improvement Export" setting in expert mode to raise the profit margin Predbat considers worth exporting for. Default is 5p.

Thanks.  And do note that this profit margin is measured over a 30 minute slot

tomkentpayne, 2025-02-24:
Ah OK, thanks @HeavyPXL I think you're right.

Although doing the same kind of comparison of what the load would've cost if the battery charge/discharge was left out of the equation for the two slots 20:30 and 21:00 raise a similar question, where the load would've cost ~9p, and the battery is being charged and discharged to roughly the same soc at a net cost of 9-10p.

I'll have a look at the metric_min_improvement settings and see if I can tweak it the right amount :)

closed 2025-02-24 by tomkentpayne