springfall2008/batpred #2035: freeze charge, doesn't!

gcoan, 2025-02-22:
**Describe the bug**
Plan shows freeze charge mode, but batteries are actually charging at full rate

[image]

[image]

I think the problem is that the charge start time is set for 13:30, its currently 13:58 so the inverters start charging.

[image]

**Expected behaviour**
Freeze charge only allows charging from solar, not grid import

**Predbat version**
8.14.4

**Environment details**
HAOS
2x Gen 1 hybrid inverters each with their own battery

**Log file**

darryl2k, 2025-02-25:
I just noticed that I have  asimilar issue, just caught it an dpaused the battery charg

[image]

theblackcat9, 2025-02-27:
I think I've got the same problem with "Hold Charge".  

As there has been more solar around over the last few days, predbat has been planning to "Hold charge" overnight during the cheaper off-peak period, so that the solar tops up the battery to 100% around 4pm to export excess solar during the peak period. I'm on Octopus Flux tariff. This is how I would expect predbat to work.  However, what has actually happened is the battery has charged up during the off-peak period to nearly 100% meaning the solar tops up the battery to 100% around noon and is exporting the "excess" solar during the day export rate (12.74p/kWh) which is less than the overnight off-peak rate(14.47p/kWh), which is not ideal.

When predbat wants to "hold charge", it looks like it is setting the target SOC to a lower value than the current SOC. However, when this happens, my battery is charging from the grid until it reaches 100% or predbat does something different.
I did an experiment to set a timed charged from the givenergy portal, setting the target SOC to lower than the current SOC to see if the same thing happened and it didn't. It worked as expected, the grid supplied the house but the battery did not charge.
When I had a look at the givtcp control entities, predbat was setting number.givtcp_ch9999g999_target_soc as the target SOC and the portal set the entity number.givtcp_ch9999g999_charge_target_soc_1.
As the charge schedule is activated, I think the inverter is looking at the  charge_target_soc_1 value which is at 100% and therefore charging the battery rather than using the target_soc value.

I'm using givtcp 3.0.4 and predbat 8.15.1 and have a single AIO.

Is this a bug in the way predbat is setting the target SOC or just in my apps.yaml?
The relevant parts of my apps.yaml are:-
  geserial: 'ch9999g999'
  givtcp_rest:
    - 'http://192.168.0.xxx:6345'
I've also got a number of non-rest definitions including this one:-
charge_limit:
- number.givtcp_{geserial}_target_soc

Predbat status:-

[image]

Inverter changes from portal remote control log:-

[image]

SOC % history. Initial SOC at 2am was 67%. Target SOC for "Hold charge" was set to 63%

[image]

gcoan, 2025-03-09:
@springfall2008 does #2084 fix this?

gcoan, 2025-03-09:
> > @springfall2008 does #2084 fix this?
> 
> No it's unrelated
> 
> Freeze charge should set the target Soc to the current level, is it not doing that?

In the screenshot above it looks like target SoC was set to the current SoC on both inverters

https://private-user-images.githubusercontent.com/142018870/415910382-078ae781-496c-42c9-9705-cdcc45db7b67.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDE1NTg5MzUsIm5iZiI6MTc0MTU1ODYzNSwicGF0aCI6Ii8xNDIwMTg4NzAvNDE1OTEwMzgyLTA3OGFlNzgxLTQ5NmMtNDJjOS05NzA1LWNkY2M0NWRiN2I2Ny5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMzA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDMwOVQyMjE3MTVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05YzU0ZmFmOTc4N2RjNjAyNDY0YTlhZTQ2OWRmNzcwMGM0ZTIxMjAxMzU3NTZlMDQ2NDgwZDRkNTI0NDMyOTVjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.PUAaLhHQQVlxQR7SQfjd2bfDp47IUQLmCH3GpOXAXDo

But both were still charging at full rate 🥺

sammort, 2025-03-12:
@springfall2008 I'm seeing the same behavior again today, triggered by predbat planning to Hold Charge up to the planned Octopus Saving Session. Instead the AIO is charging at full rate from the Grid.

[predbat.log](https://github.com/user-attachments/files/19209612/predbat.log)
[predbat_plan.html.txt](https://github.com/user-attachments/files/19209614/predbat_plan.html.txt)
[predbat_debug.yaml.txt](https://github.com/user-attachments/files/19209613/predbat_debug.yaml.txt)

WyndStryke, 2025-03-23:
@gcoan I looked into this from the perspective of a Fox inverter, and made some observations.  On the Fox side, the solution is to set the inverter's mode to 'back-up' which will hold it steady at the reserve_soc via the charge_freeze_service, but along the way I found some stuff which _might_ be relevant to other inverters.

https://github.com/springfall2008/batpred/issues/2114#issuecomment-2744553411

springfall2008, 2025-03-23:
From another thread, maybe 'enable AC charge upper limit' needs enabling on the AIO?

gcoan, 2026-08-20:
closing as no recent progress on this, possibly superseded

closed 2026-08-20 by gcoan