springfall2008/batpred #2037: apps.yaml functions

brickatius, 2025-02-23:
Would it be possible to have a comprehensive list, with brief descriptions, of the available/necessary functions for apps.yaml files?
I appreciate that there are annotated template files for various inverters but I think there are things missing. For example, I have a LuxPower Hybrid inverter. There is no soc_kw entity only a soc in % so I had to use a HA template to provide soc_kw as this is required to calculate the charge and discharge curves.
A list of 'must have' functions would be really useful.

gcoan, 2025-02-23:
Have you looked at the apps.yaml part of the documentation ?

https://springfall2008.github.io/batpred/apps-yaml/

This should describe everything in apps.yaml and what it is used for, what is optional and recommended settings.

If there are any gaps then I'm happy to fill them in.  I have on my to-do list to restructure the apps.yaml files to make them follow the same structure as the documentation and to try to reduce the variation of files, but that's a bit of a longer term task.

brickatius, 2025-02-24:
Hi Geoffrey, Thanks for your reply - greatly appreciated. I’ve left a comment on Github.Thank you all for what you are doing with Predbat - sterling work. I came to Predbat because now that the Octopus Flux daytime export rates are less than the nighttime import rates, the previous no-brainer of charging the batteries to 100%, exporting excess solar ASAP and force discharging down to a percentage that would see me through to the next charging slot doesn’t work anywhere near as profitably. Kind regards, Richard

https://springfall2008.github.io/batpred/apps-yaml/

https://springfall2008.github.io/batpred/apps-yaml/

closed 2025-03-08 by brickatius