springfall2008/batpred #2041: Compare using previous day's Agile import tariff rates

CJDumbleton, 2025-02-24:
**Describe the bug**
The midnight run of the new (excellent) compare feature used yesterday's Agile import tariff rates.

**Expected behaviour**
I would expect Predbat's compare feature that ran at midnight this morning to use today's Agile import tariff rates.

**Predbat version**
v8.15.1

**Environment details**
- GivEnergy 9.5 kWh battery and 5kW hybrid inverter
- Standard HAOS installer

**Screenshots**
The left-hand side of the screenshot shows the start of the Agile import plan from today's midnight Compare results. The right-hand side of the screenshot shows yesterday's Agile import tariff downloaded from https://energy-stats.uk/download-historical-pricing-data/. They show that the import tariff rates used by Predbat Compare at midnight this morning were from yesterday.
[image]

**Log file**
[predbat.log](https://github.com/user-attachments/files/18940691/predbat.log)
Log attached from 2025-02-24 00:00 to 2025-02-24 00:30. Some entries of note:
2025-02-24 00:01:38.346517: Compare Tariff: Agile import/Fixed export
2025-02-24 00:01:41.093458: Download Octopus rates from https://api.octopus.energy/v1/products/AGILE-24-10-01/electricity-tariffs/E-1R-AGILE-24-10-01-M/standard-unit-rates/
2025-02-24 00:01:41.093689: Return cached octopus data for https://api.octopus.energy/v1/products/AGILE-24-10-01/electricity-tariffs/E-1R-AGILE-24-10-01-M/standard-unit-rates/ age 0.1 minutes
2025-02-24 00:01:41.094461: Download Octopus rates from https://api.octopus.energy/v1/products/OUTGOING-VAR-24-10-26/electricity-tariffs/E-1R-OUTGOING-VAR-24-10-26-M/standard-unit-rates/
2025-02-24 00:01:41.094618: Return cached octopus data for https://api.octopus.energy/v1/products/OUTGOING-VAR-24-10-26/electricity-tariffs/E-1R-OUTGOING-VAR-24-10-26-M/standard-unit-rates/ age 0.0 minutes

**Predbat debug yaml file**
[predbat_debug_11_20_00.yaml.txt](https://github.com/user-attachments/files/18941009/predbat_debug_11_20_00.yaml.txt)
Note, I manually ran the Compare feature at 11:25 (after I had produced the above debug file). This time Predbat Compare correctly used today's Agile import tariff prices.

bigsmelly, 2025-03-28:
Just a +1, i also see this .

gcoan, 2025-09-07:
Hi @springfall2008 

Did you get a chance to look into this?  It happened again last night, very visible as the Agile rates for Sunday were very low.

[image]

I’m on Agile + Fixed outgoing.

Compare of ‘current’ gives me the correct rates:

[image]

But compare of ‘agile+fixed’ gives the day before rates:

[image]

logfile:

[predbat.3.log](https://github.com/user-attachments/files/22198904/predbat.3.log)

comparisons.yaml:
[comparisons.log.yaml](https://github.com/user-attachments/files/22198905/comparisons.log.yaml)

gcoan, 2025-09-18:
@springfall2008 this is still regularly happening:

[image]

The left most blue line (current rates( should be the same length as the yellow/orange one (agile), but about half the time the compare uses the agile rates from the previous day, thus getting the compare wrong