springfall2008/batpred #2043: Silent Predbat crash on Meteo warning retrieval

PianSom, 2025-02-24:
**Describe the bug**
I had a silent crash this morning at 10.30am, which I only spotted because the plan didn't update. Container restart was successful and showed no issues (other than 429 errors for Solcast and Meteo)

**Expected behaviour**
No crash 

**Predbat version**
8.15.1

**Environment details**
Predbat in standalone docker container
2xAIO

**Screenshots**
Log screen grab:

[image]

**Log file**
As above

**Predbat debug yaml file**
Not repeatable, but if you want to see a generic log file then holler.

theblackcat9, 2025-02-24:
I had the same problem at 10:30 and only noticed at 12:30 that the predbat plan had not updated.  I had to restart HA to get things back up and working again.

This is my predbat log from 10:30 until I restarted at 12:30 :-

2025-02-24 10:30:01.029805: --------------- PredBat - update at 2025-02-24 10:30:00+00:00 with clock skew 0 minutes, minutes now 630
2025-02-24 10:30:01.030390: Saved current settings to /addon_configs/6adb4f0d_predbat/predbat_config.json
2025-02-24 10:30:01.048613: Using cached GITHub data for https://api.github.com/repos/springfall2008/batpred/releases age 55.0 minutes
2025-02-24 10:30:01.048733: Predbat /config/predbat.py version v8.15.1 currently running, latest version is v8.15.1 latest beta v8.15.1
2025-02-24 10:30:01.048782: Autoupdate: Currently up to date
2025-02-24 10:30:01.082704: Configuration: forecast_hours 36 num_cars 0 debug enable is False calculate_plan_every 15
2025-02-24 10:30:01.083167: Predbat mode is set to Control charge & discharge
2025-02-24 10:30:01.229495: Processing alerts for approx position latitude 52.6 longitude -1.2
2025-02-24 10:30:01.229610: Processing alerts from https://feeds.meteoalarm.org/feeds/meteoalarm-legacy-atom-united-kingdom
2025-02-24 12:32:37.256800: Loading apps.yaml
2025-02-24 12:32:37.313051: Predbat: Startup predbat
2025-02-24 12:32:37.345933: --------------- PredBat - update at 2025-02-24 12:30:00+00:00 with clock skew 0 minutes, minutes now 750
2025-02-24 12:32:37.348111: Config root is /config
2025-02-24 12:32:37.348240: --------------- PredBat - update at 2025-02-24 12:30:00+00:00 with clock skew 0 minutes, minutes now 750
2025-02-24 12:32:37.348255: Starting HA interface
2025-02-24 12:32:37.358783: Info: Connected to Home Assistant at http://supervisor/core
2025-02-24 12:32:37.363437: Info: Add-on slug is 6adb4f0d_predbat
2025-02-24 12:32:37.363954: Creating task: <coroutine object HAInterface.socketLoop at 0x7f1eec24fe60>
2025-02-24 12:32:37.364612: Info: Web Socket task started
2025-02-24 12:32:37.364996: Info: Start socket for url http://supervisor/core/api/websocket

I didn't get any error messages. This is the predbat status for this morning:-

[image]

24 February 2025
changed to Error: Exception raised Auto-restart triggered
12:35:24 - 3 hours ago - Supervisor
changed to Inverter 0 unable to read REST data from http://192.168.0.123:6345/readData - REST will be disabled
12:35:06 - 3 hours ago - Supervisor
changed to Demand
05:00:08 - 11 hours ago - Supervisor

The errors at 12:35 only occurred while HA was rebooting.

browellm, 2025-02-24:
I also had this same crash and can't point to anything that would have prompted it.

SwiftRR, 2025-02-24:
Is it all connected to v8.13.0 of predbat?: '**New bad weather alert system'**: 
Check out https://github.com/springfall2008/batpred/releases/tag/v8.13.0.
You will see the line of code that looks as if it is causing the issue.

There was some code to be added to apps.yaml
I added this to my apps.yaml but then commented out all the lines as the alarm seemed a bit 'sensitive'.

There is a .py file that was added to our predbat in this update: `alertfeed.py`
This includes:
`https://feeds.meteoalarm.org/feeds/meteoalarm-legacy-atom-united-kingdom`

The log posted here suggests that the url above was initiated my apps.yaml.

Rob

springfall2008, 2025-02-26:
Not sure I can see an error here?

The REST error means GivTCP wasn't talking correctly

Have you updated your predbat add on container?

PianSom, 2025-02-26:
There was no other error showing at 10:31, when the meter update happened. But the plan then didn't update until Predbat was manually restarted at 14:01. 

It was a silent and deadly crash.

chalfontchubby, 2026-08-20:
Thanks for the report. This issue has been open for over a year with no recent activity - we're volunteers working through a large backlog and need to filter what's still relevant. If this is still a problem for you, please let us know and we'll keep it open. Otherwise, we'll close it in a week (around 2026-08-28).

gcoan, 2026-08-21:
I think this could be the same as #2062 predbat crashes if it can't connect to meteoalarm for weather warnings.

As far as I remember there hasn't been any changes to the error handling of the meteoalarm get, so as this was a silent crash a long time ago and hard to recreate, it may be worth closing this and investigating whether 2062 is fixed.

chalfontchubby, 2026-08-27:
Closing this as believed fixed. Predbat crashing when it can't reach meteoalarm looks to have been addressed since this was reported - the alert-feed fetch now has proper error handling around the network call (`alertfeed.py`'s `run()` wraps the whole fetch in a try/except that logs a warning and returns rather than propagating, with its own inner try/except around the download itself), so an outage there shouldn't take Predbat down with it.

We haven't reproduced a live meteoalarm outage to confirm directly, so if you hit this again on a current version, please open a new ticket with a log covering the failure - happy to take another look.

closed 2026-08-27 by chalfontchubby