springfall2008/batpred #2054: Crash due to connection refused/maximum retries exceeded

nblount, 2025-02-28:
**Describe the bug**
PredBat crashed at 15h00 due to maximum retries exceeded on a connection that was being refused.

**Expected behaviour**
Not to crash or it be more obvious that it has crashed may change the plan to show error?

**Predbat version**
 Add-on version: 1.2.4
 Home Assistant Core: 2025.2.4
 Home Assistant Supervisor: 2025.02.4
 Predbat files are installed correctly for version v8.15.1

**Environment details**
- GivEnergy Invertor 5kG3, Battery 3 * 9.5kG2
- HAOS

**Screenshots**
Battery Plan was stuck at 15h00.

**Log file**
2025-02-27 14:55:08.311774: Info: record_status Demand
2025-02-27 15:00:00.650577: Error: HTTPConnectionPool(host='supervisor', port=80): Max retries exceeded with url: /core/api/states (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f759b6e65c0>: Failed to establish a new connection: [Errno 111] Connection refused'))
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/urllib3/connection.py", line 198, in _new_conn
    sock = connection.create_connection(
  File "/usr/local/lib/python3.10/dist-packages/urllib3/util/connection.py", line 85, in create_connection
    raise err
  File "/usr/local/lib/python3.10/dist-packages/urllib3/util/connection.py", line 73, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/urllib3/connectionpool.py", line 787, in urlopen
    response = self._make_request(
  File "/usr/local/lib/python3.10/dist-packages/urllib3/connectionpool.py", line 493, in _make_request
    conn.request(
  File "/usr/local/lib/python3.10/dist-packages/urllib3/connection.py", line 445, in request
    self.endheaders()
  File "/usr/lib/python3.10/http/client.py", line 1278, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.10/http/client.py", line 1038, in _send_output
    self.send(msg)
  File "/usr/lib/python3.10/http/client.py", line 976, in send
    self.connect()
  File "/usr/local/lib/python3.10/dist-packages/urllib3/connection.py", line 276, in connect
    self.sock = self._new_conn()
  File "/usr/local/lib/python3.10/dist-packages/urllib3/connection.py", line 213, in _new_conn
    raise NewConnectionError(
urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPConnection object at 0x7f759b6e65c0>: Failed to establish a new connection: [Errno 111] Connection refused
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/requests/adapters.py", line 667, in send
    resp = conn.urlopen(
  File "/usr/local/lib/python3.10/dist-packages/urllib3/connectionpool.py", line 841, in urlopen
    retries = retries.increment(
  File "/usr/local/lib/python3.10/dist-packages/urllib3/util/retry.py", line 519, in increment
    raise MaxRetryError(_pool, url, reason) from reason  # type: ignore[arg-type]
urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='supervisor', port=80): Max retries exceeded with url: /core/api/states (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f759b6e65c0>: Failed to establish a new connection: [Errno 111] Connection refused'))
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/config/hass.py", line 190, in timer_tick
    item["callback"](None)
  File "/config/predbat.py", line 1015, in run_time_loop
    self.ha_interface.update_states()
  File "/config/ha.py", line 400, in update_states
    res = self.api_call("/api/states")
  File "/config/ha.py", line 686, in api_call
    response = requests.get(url, headers=headers, timeout=TIMEOUT)
  File "/usr/local/lib/python3.10/dist-packages/requests/api.py", line 73, in get
    return request("get", url, params=params, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/requests/api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/requests/sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python3.10/dist-packages/requests/sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/requests/adapters.py", line 700, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPConnectionPool(host='supervisor', port=80): Max retries exceeded with url: /core/api/states (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f759b6e65c0>: Failed to establish a new connection: [Errno 111] Connection refused'))
2025-02-27 15:00:10.880106: Warn: Web Socket closed, will try to reconnect in 5 seconds
2025-02-27 15:00:15.885409: Info: Start socket for url http://supervisor/core/api/websocket
2025-02-27 15:00:20.985651: Error: Web Socket exception in startup: 400, message='Invalid response status', url=URL('http://supervisor/core/api/websocket')
2025-02-27 15:00:20.985778: Error: Traceback (most recent call last):
  File "/config/ha.py", line 184, in socketLoop
    async with session.ws_connect(url) as websocket:
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 1167, in __aenter__
    self._resp = await self._coro
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 821, in _ws_connect
    raise WSServerHandshakeError(
aiohttp.client_exceptions.WSServerHandshakeError: 400, message='Invalid response status', url=URL('http://supervisor/core/api/websocket')
2025-02-27 15:00:20.986018: Warn: Web Socket closed, will try to reconnect in 5 seconds
2025-02-27 15:00:25.991141: Info: Start socket for url http://supervisor/core/api/websocket
2025-02-27 15:00:25.994814: Info: Web Socket active
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service init-predbat: stopping
s6-rc: info: service init-predbat successfully stopped
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-timezone: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service base-addon-timezone successfully stopped
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service init-predbat: starting
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service init-predbat successfully started
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting

**Predbat debug yaml file**
N/A

gcoan, 2025-02-28:
If predbat crashes it should set status to error

we've had this problem before, although it does seem to be happening more frequently on the newer versions of home assistant
the fix is to create a long lasting access token that predbat can use, see updated docs: https://github.com/gcoan/batpred/blob/main/docs/apps-yaml.md#home-assistant-connection

springfall2008, 2025-03-01:
The log above indicates a restart without error, but you are saying it didn't work?

nblount, 2025-03-01:
I rebooted HA and it came back up. The report was that it crashed.

johnwb87, 2025-03-04:
I had what I think looks like the same silent crash yesterday also. Missed the start of the saving session. I only realised because I checked the GivEnergy App to see the export.

From the log for the full duration. The crash is the same as above when the Web Socket goes back to active. Then hung until I restarted the add-on.

2025-03-03 16:10:00.688829: Error: HTTPConnectionPool(host='supervisor', port=80): Max retries exceeded with url: /core/api/states (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xffffb7f93130>: Failed to establish a new connection: [Errno 111] Connection refused'))
2025-03-03 16:10:02.969321: Warn: Web Socket closed, will try to reconnect in 5 seconds
2025-03-03 16:10:07.975210: Info: Start socket for url http://supervisor/core/api/websocket
2025-03-03 16:10:07.983622: Error: Web Socket exception in startup: Cannot connect to host supervisor:80 ssl:default [Connect call failed ('172.30.32.2', 80)]
2025-03-03 16:10:07.993852: Error: Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/connector.py", line 980, in _wrap_create_connection
    return await self._loop.create_connection(*args, **kwargs)  # type: ignore[return-value]  # noqa
  File "/usr/lib/python3.10/asyncio/base_events.py", line 1076, in create_connection
    raise exceptions[0]
  File "/usr/lib/python3.10/asyncio/base_events.py", line 1060, in create_connection
    sock = await self._connect_sock(
  File "/usr/lib/python3.10/asyncio/base_events.py", line 969, in _connect_sock
    await self.sock_connect(sock, address)
  File "/usr/lib/python3.10/asyncio/selector_events.py", line 501, in sock_connect
    return await fut
  File "/usr/lib/python3.10/asyncio/selector_events.py", line 541, in _sock_connect_cb
    raise OSError(err, f'Connect call failed {address}')
ConnectionRefusedError: [Errno 111] Connect call failed ('172.30.32.2', 80)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/ha.py", line 184, in socketLoop
    async with session.ws_connect(url) as websocket:
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 1167, in __aenter__
    self._resp = await self._coro
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 805, in _ws_connect
    resp = await self.request(
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 562, in _request
    conn = await self._connector.connect(
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/connector.py", line 540, in connect
    proto = await self._create_connection(req, traces, timeout)
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/connector.py", line 901, in _create_connection
    _, proto = await self._create_direct_connection(req, traces, timeout)
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/connector.py", line 1209, in _create_direct_connection
    raise last_exc
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/connector.py", line 1178, in _create_direct_connection
    transp, proto = await self._wrap_create_connection(
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/connector.py", line 988, in _wrap_create_connection
    raise client_error(req.connection_key, exc) from exc
aiohttp.client_exceptions.ClientConnectorError: Cannot connect to host supervisor:80 ssl:default [Connect call failed ('172.30.32.2', 80)]

2025-03-03 16:10:07.994494: Warn: Web Socket closed, will try to reconnect in 5 seconds
2025-03-03 16:10:12.996155: Info: Start socket for url http://supervisor/core/api/websocket
2025-03-03 16:10:24.397708: Error: Web Socket exception in startup: Cannot connect to host supervisor:80 ssl:default [Connect call failed ('172.30.32.2', 80)]
2025-03-03 16:10:24.399515: Error: Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/connector.py", line 980, in _wrap_create_connection
    return await self._loop.create_connection(*args, **kwargs)  # type: ignore[return-value]  # noqa
  File "/usr/lib/python3.10/asyncio/base_events.py", line 1076, in create_connection
    raise exceptions[0]
  File "/usr/lib/python3.10/asyncio/base_events.py", line 1060, in create_connection
    sock = await self._connect_sock(
  File "/usr/lib/python3.10/asyncio/base_events.py", line 969, in _connect_sock
    await self.sock_connect(sock, address)
  File "/usr/lib/python3.10/asyncio/selector_events.py", line 501, in sock_connect
    return await fut
  File "/usr/lib/python3.10/asyncio/selector_events.py", line 541, in _sock_connect_cb
    raise OSError(err, f'Connect call failed {address}')
ConnectionRefusedError: [Errno 111] Connect call failed ('172.30.32.2', 80)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/ha.py", line 184, in socketLoop
    async with session.ws_connect(url) as websocket:
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 1167, in __aenter__
[... 7 lines cut ...]
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/connector.py", line 901, in _create_connection
    _, proto = await self._create_direct_connection(req, traces, timeout)
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/connector.py", line 1209, in _create_direct_connection
    raise last_exc
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/connector.py", line 1178, in _create_direct_connection
[... cut ...]
    self._resp = await self._coro
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 821, in _ws_connect
    raise WSServerHandshakeError(
aiohttp.client_exceptions.WSServerHandshakeError: 400, message='Invalid response status', url=URL('http://supervisor/core/api/websocket')

2025-03-03 16:10:29.429998: Warn: Web Socket closed, will try to reconnect in 5 seconds
2025-03-03 16:10:34.434097: Info: Start socket for url http://supervisor/core/api/websocket
2025-03-03 16:10:34.475581: Info: Web Socket active

gcoan, 2025-03-04:
> I had what I think looks like the same silent crash yesterday also

Might be worth creating a long lived access token see https://github.com/springfall2008/batpred/issues/2054#issuecomment-2690098636

johnwb87, 2025-03-05:
Thanks @gcoan. To confirm even though running as an Add-on follow the documentation which discusses needing the token for a docker install?

gcoan, 2025-03-06:
@johnwb87 yes this works with non-docker installs

In my latest draft documentation I updated it to make it clearer that setting a token is a possible fix to these websocket and other 'disconnect' errors

> If you are using a standard Predbat add-on then this will be automatic and you should normally not need to set this. If you find you get issues where Predbat cannot communicate with Home Assistant after running for a long period of time and you get web socket errors, then creating a HA access token can resolve this.

> If you run Predbat in a Docker container then you will need to set the URL or IP address of Home Assistant and an access key which is the long-lived access token you can create inside Home Assistant:

If docker you have to create the access token, for predbat add-on you shouldn't need to but a number of people have had crashes on newer HA versions and this seems to cure this

sparks1372, 2025-03-20:
I am experiencing very similar issues with v8.16.6 which I thought were due to load on the Raspberry Pi running HA. Last night I set up PredBat to run on a separate machine as a docker container to try mitigate this issue but that also experienced the same but this morning at about 9 AM. The reason why I wondered if it was due to load on the RPi was it seemed to always fail part way through the compare run around midnight.

1852 | 2025-03-20 09:19:00.992804:  Error: Traceback (most recent call last):
1851 | 2025-03-20 09:19:00.968648:   Error: Web Socket exception in startup: Cannot connect to host  xxx:xxx ssl:default [Connect call failed  ('xxx', xxx)]
1850 | 2025-03-20 09:19:00.960836:  Info: Start socket for url http://xxxx:xxx/api/websocket
1849 | 2025-03-20 09:18:55.955305:  Warn: Web Socket closed, will try to reconnect in 5 seconds - error count 1
1847 | aiohttp.client_exceptions.C  lientConnectorError: Cannot connect to host homeassistant.sparks:8123  ssl:default [Connect call failed ('xxxx', xxxx)]
1846 | raise client_error(req. connection_key, exc) from exc
1845 | File "/usr/lib/python3/di st-packages/aiohttp/connector.py", line 1000, in _wrap_create_connection
1844 | ^^^^^^^ ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
1843 | transp, proto = await s elf._wrap_create_connection(
1842 | File "/usr/lib/python3/di st-packages/aiohttp/connector.py", line 1204, in _create_direct_connection
1841 | raise last_exc
1840 | File "/usr/lib/python3/di st-packages/aiohttp/connector.py", line 1235, in _create_direct_connection
1839 | ^^^^^^^^^^^^ ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
1838 | _, proto = await self._ create_direct_connection(req, traces, timeout)
1837 | File "/usr/lib/python3/di st-packages/aiohttp/connector.py", line 911, in _create_connection
1836 | ^^^^^^^^^^^^^^^ ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
1835 | proto = await self._cre ate_connection(req, traces, timeout)
1834 | File "/usr/lib/python3/di st-packages/aiohttp/connector.py", line 544, in connect
1833 | ^^^^^^^^^^^^^^^^ ^^^^^^^^^^^^^^
1832 | conn = await self._conn ector.connect(
1831 | File "/usr/lib/python3/di st-packages/aiohttp/client.py", line 574, in _request
1830 | ^^^^^^^^^^^^^^^^ ^^^
1829 | resp = await self.reque st(
1828 | File "/usr/lib/python3/di st-packages/aiohttp/client.py", line 825, in _ws_connect
1827 | ^^^^^^^^^^ ^^^^^^
1826 | self._resp = await self ._coro
1825 | File "/usr/lib/python3/di st-packages/aiohttp/client.py", line 1187, in __aenter__
1824 | async with session.ws_c onnect(url) as websocket:
1823 | File "/config/ha.py", lin e 198, in socketLoop
1822 | Traceback (most recent call  last):
1820 | The above exception was the  direct cause of the following exception:
1818 | ConnectionRefusedError: [Er rno 111] Connect call failed ('192.168.68.122', 8123)
1817 | raise OSError(err, f'Co nnect call failed {address}')
1816 | File "/usr/lib/python3.12 /asyncio/selector_events.py", line 691, in _sock_connect_cb
1815 | ^^^^^^^^^
1814 | return await fut
1813 | File "/usr/lib/python3.12 /asyncio/selector_events.py", line 651, in sock_connect
1812 | await self.sock_connect (sock, address)
1811 | File "/usr/lib/python3.12 /asyncio/base_events.py", line 1007, in _connect_sock
1810 | ^^^^^^^^^^^^^^^^ ^^^^^^^^^
1809 | sock = await self._conn ect_sock(
1808 | File "/usr/lib/python3.12 /asyncio/base_events.py", line 1104, in create_connection
1807 | raise exceptions[0]
1806 | File "/usr/lib/python3.12 /asyncio/base_events.py", line 1122, in create_connection
1805 | ^^^^^^^^^^^^^^^^ ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
1804 | return await self._loop .create_connection(*args, **kwargs)
1803 | File "/usr/lib/python3/di st-packages/aiohttp/connector.py", line 992, in _wrap_create_connection
1802 | 2025-03-20 09:18:55.954959:  Error: Traceback (most recent call last):
1801 | 2025-03-20 09:18:55.943634:   Error: Web Socket exception in startup: Cannot connect to host  xxxx ssl:default [Connect call failed  ('xxxx', xxxx)]
1800 | 2025-03-20 09:18:55.935234:  Info: Start socket for url http://xxx:xxx/api/websocket
1799 | 2025-03-20 09:18:50.939664:  Error: ('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))
1798 | 2025-03-20 09:18:50.928817:  Warn: Web Socket closed, will try to reconnect in 5 seconds - error count 0

sparks1372, 2025-03-20:
For avoidance of doubt I have the HA URL and Key set up

gcoan, 2025-03-20:
maybe running out of memory/resources in your docker container, either the HA one or predbat?

Do you see any errors in the HA log around the same time that might indicate why HA is disconnecting?

closed 2026-08-28 by chalfontchubby