springfall2008/batpred #2057: Development of predbat to control GivEnergy 3-phase Hybrid Inverter Giv-3HY-XX

LeeFarm11, 2025-02-28:
**Is your feature request related to a problem? Please describe.**
predbat does not work with GivEnergy 3 phase Inverters due it seems to being completely different to previous inverters in the way it is controlled. GivTCP is unable to translate most of the predbat requests into anything which works properly. 

**Describe the solution you'd like**
For predbat to be able to control the Inverter directly without GivTCP. Via REST maybe?

**Describe alternatives you've considered**
I did have a home built automation working, but it is only partially successful. For Winter it could just about cope, but for Summer there is no chance. Some of the needed controls are not available. Combined with lack of any documentation of what all the 3 phase controls available in Giv Portal really do, it is not a viable solution. I did try to work with @britkat1980 to give him access to my Inverter, but we could not get it working. Since then I have moved my HA to a VM on a Windows machine (instead of VM on a iMac) so it might be easier in future to set this up for someone.

**Additional context**
I would like to invite @gcoan or @springfall2008 or @britkat1980 to visit me, or to set up remote access to my Inverter somehow, so that they can try to work out a solution. I have some technical ability, but developing the code needed is beyond me.
I am trying to put together a spreadsheet to identify the impact of different combinations of settings, but it is slow progress.

trevordowling, 2025-02-28:
Hi Lee, 

I've got a GIV-3HY-11 up and running, there are some issues still being worked through on the three phase based systems (there just aren't many of us running these yet), but you can get this to work well now. 

I'm new to this too, but happy to help get you up and running if I can (I have limited knowledge but what I've got working I'm more than happy to share).

I am finding new issues, but this community is great and people really want to help (though my advice is to read the documentaion provided, it is pretty good and to help minimse suport requests we all should try to read the notes first (I've been guilty too! sorry all!))

I've got a GIV-3HY-11 and one (x6 packs) 20.29kWh battery stack.

I'm not running any car charging or iBoost (though I did get the iBoost up and running in the plan)

Trevor

LeeFarm11, 2025-02-28:
Hi @trevordowling I have same as you Inverter and Battery pack.
I am running GivTCP Dev with predbat, with REST disabled in apps.yaml
I would love to know how you have Freeze Export, Freeze Charge and others working because I don't see any possible way.
I also have no car charging or iBoost but I do have 18kwp panels.
I did put together an automation which allowed Charging and Exporting to work from predbat, but latest GivTCP Dev version seems to have made some things better but some things worse.

How are you managing to get predbat to work?

Thanks

trevordowling, 2025-02-28:
Hi Lee,

I've got 20 panels, so less generation than you, about 8.8kw

Have you got anything running or are you trying to set up these specific
items (Freeze Export, Freeze Charge)?

You must be using the GivTCP-beta  3.1.1

I initially used v2, then read the docs, loaded v3 but actually failed to
execute this bit correctly, so be careful. Advise you uninstal, delete old
versions, etc. (I think there were some notes on how to do this in the
docs).

I advise you get REST up and running, I've just logged a new issue possibly
related to this (TBC), where entities are only available via REST, and
remain undefined if not but within GivTCP rather than PB. So please get
REST running.

Not 100% sure on what I have running in terms of Freeze Export and Freeze
Charge, as I've just basically let the system run.

My basic operation is as follows:

   - Before midnight discharge battery as export (sell what I have)
   - Charge to 100% between 00:00 and 05:00
   - Holding at 100% once full until the low import rate has expired
   (import rate is less than export rate)
   - Run from battery, change if solar and export extra

I had initially expected to include Octopus and iBoost items, but at the
moment there is nt reason to with the current tariff.

I've experimented with charge and discharge within a single charge window,
worked but it looked crazy, so I've switched that off. (You can consolidate
things to make this possibly worthwhile).

Current plan as of now

[image: image.png]

Today so far - you can see I charged the battery to full, it held it there
until my 5am window (low cost - lower than export, so makes sense to hold
the battery at 100%).

[image: image.png]

The early attempt to extract more out of the low cost window - but IMHO
it's not worth the extra battery 'hammering', but a good exercise.

[image: image.png]

LeeFarm11, 2025-02-28:
Hi.

I am using GivTCP Dev.

The images don't show in your post.
But that sounds like you are just using a fixed schedule same every day, with manual interventions, not using predbat at all?

trevordowling, 2025-02-28:
Hi Lee,

No - Predbat is full up and running and controlling the system (appologies that the images didn't load)

Current plan as of now

[image]

Today so far - you can see I charged the battery to full, it held it there until my 5am window (low cost - lower than export, so makes sense to hold the battery at 100%).

[image]

The early attempt to extract more out of the low cost window - but IMHO it's not worth the extra battery 'hammering', but a good exercise.

[image]

The plan for the next period

[image]

LeeFarm11, 2025-02-28:
Oh WOW!
This is without any additional automations or aliases and helpers etc?

I have been using GivTCP DEV. Should I run Beta instead?

And Enable REST in apps.yaml?

Anything else?

trevordowling, 2025-02-28:
Hi Lee,

Yes...

You need to upgrade to the Beta version on GivTCP. I'm running PB 8.15.1.

It is PredBat just working...

There are still some gremlins here and there but that's true for the
GivEnergy apps too... so all we can do is all help to address, fix and work
around things as they comlatest Beta version and the latest release of PB
8.15.1.

I'm more than happy to help review your config files.

I've been working on this since Dec but only got my smart meter a couple of
weeks ago.

You can see that the complexity can be set just by adjusting the settings,
but be careful, I ran in demo mode for some time trying to understand how
it all hangs together.

I worked with one developer last week, who's card didn't support the three
phase system. With some help and back and forth, it now works well. Entity
names different between 1 and 3 phase system.

This will work, yes, my plan is now much simpler than expected, but as gas
is 6p, electric 8p and export is 15p, it makes sense.

The iBoost may still make a return on my side, but only as an engineering
project. I'm aiming to extend the control of an iBoost system directly from
HA, and not their clamp.

Trevor

LeeFarm11, 2025-03-01:
Today is very sunny and my predbat plan is suggesting to allow the battery to get full by 10:30. Then all the Solar will try to go to the grid. You see 6.07kwh and 6.03kwh per half hour in the plan being generated but this is obviously above 11kw, so the Inverter will instead limit itself and throw away any excess over 11kw. I would expect it to do some exporting before 100% to keep some capacity in the battery to allow none of the Solar to be wasted.

@gcoan I know this is not the correct place to post this but are you able to help identify why predbat does this strange plan? I assume other users have this issue and I have seen it mentioned on FaceBook but I don't recall a solution. I am unable to identify why this happens.

@trevordowling This also illustrates why I need Freeze Export and Freeze Charge working. I have not yet had chance to test your proposal. With this amount of Sun and without yet having tested, I need today to export manually to prevent clipping.

[image]

gcoan, 2025-03-01:
> You see 6.07kwh and 6.03kwh per half hour in the plan being generated but this is obviously above 11kw, so the Inverter will instead limit itself and throw away any excess over 11kw. I would expect [...]

Crazy amount of solar you are generating and we're only just out of winter!  I thought my predicted 2.6kWh per hour was good, although I'm exceeding that with 7.7kW power at the moment.

This is a problem with predbat, it doesn't handle solar clipping particularly well (yet).  There have been quite a bit of discussion about it, with two possible strategies discussed:
1. Export a bit from the battery earlier on (whilst the sun isn't as strong) to leave room for more battery charging - this is what you suggested
2. Stop battery charging completely earlier on to leave space, then trickle feed charging so as to still fill the battery but not exceed export limits

See #1937 and #1206

Its possible we'll never get a perfect totally automated predbat plan that does everything everyone wants because it needs a more intimate knowledge of setup, shading, limits, battery sizes, etc.  I have particular problems with dissimilar battery sizes on my two inverters that Predbat doesn't recognise as such.  Personally I favour option 2 but others prefer option 1.

The thinking at the moment is about getting predbat as good as it can be and then have automations around the side that control Predbat, possibly via the manual API.
Last summer my approach was to put predbat into read only mode during the day to manage the charging and stop clipping myself.  I'm having to do similar for the power up events we get almost every day at the moment as the dissimilar battery sizes confuse predbat.

trevordowling, 2025-03-01:
Hi @LeeFarm11 ,

Following on from @gcoan comment, you stated you have 18kwp of panels, this exceeds the 15000w max input of the GIV-3HY-11 inverter.

I'd be concerned you are going to exceed the upper input limits which can't be good. Have you checked to ensure this will safely handle the excess? 

I'd of thought you should have no more than 34, 440w panels attached == 15kW input, unless there is a physical arrangment in how these panels are arranged stopping them from all being in the full sun at the same time.

Trevor

LeeFarm11, 2025-03-01:
@trevordowling yes I know 18 is more than 15. :smilyface There are many many people with panels kwp above inverter limit. It is recommended that way. We also have quite a shallow roof.

Installed last March, which is why I have been trying to get predbat and GivTCP to play nice for so long. It is why I tried to work with GivTCP developer to support getting GivTCP working properly by giving him remote access to my Inverter. But GivTCP developer does not use predbat and it really needs predbat and GivTCP people working together to understand how the inverter works and how it can be translated to predbat speak. It is why I did build my own automation to support GivTCP where it can't cope with predbat's instructions.

The inverter is fine with 18kwp and as @gcoan mentioned others have the same probelm with this clipping issue. See those issues numbers 1206 and 1937 mentioned above.

My offer for @gcoan or @springfall2008 or @britkat1980 to visit or set up remote access still stands. Northamptonshire.

britkat1980, 2025-03-01:
Northamptonshire is a bit far away for me, but happy to try and get remote access sorted again. If you can get the vpn up and running

gcoan, 2025-03-01:
Northamptonshire could be possible for me but really Trefor is the predbat developer who knows the code far better than me

trevordowling, 2025-03-01:
I'm in Cambridge (Ely), so also more than happy to help with access, if that's closer for someone

Trevor

LeeFarm11, 2025-03-01:
Now up and running with GivTCP BETA 3.1.1 and predbat 1.2.4 / 8.15.1. Now with REST Enabled.

**Predbat Status - Charging**
Plan is proposing to Charge from 20:00 as expected. (cheap rate)

Timeslots were set correctly, same as with DEV installation.
The 2 Charge switches were set correctly, same as with Dev installation.
Charge Up To is set, but like before it uses the wrong Inverter setting. It is setting this for the GLOBAL Charge Up To %, but I know from experience that this has no impact in this setting. I have not yet found any use for the Change Up To % setting or for the Discharge Down To % setting. It needs to set the AC Charge 1 Upper SOC % to the desired target, but it does not do so.

[image]

Therefore I have re-enabled my automation to set AC Charge 1 Upper SOC % to be set to match Charge Up To % whenever Charge Up To % changes. 

```
alias: Copy Charge Up To % to AC Charge 1 Upper SOC %
description: ""
triggers:
  - entity_id:
      - number.givtcp_td2343g049_target_soc
    trigger: state
conditions: []
actions:
  - action: number.set_value
    data:
      value: "{{ states('number.givtcp_td2343g049_target_soc') }}"
    target:
      entity_id: number.givtcp_td2343g049_charge_target_soc_1
mode: single
trace:
  stored_traces: 50
```

@trevordowling
1) I have not yet found any use for the Change Up To % setting or for the Discharge Down To % setting. Have you found any use for them?
2) Do you also find the same issue with Setting the Charge Up To? Does predbat only ever plan for Charging to 100% for you? Try setting AC Charge 1 Upper SOC % to something other than 100% and see if your setup corrects it? Please let us know.

trevordowling, 2025-03-01:
1. I have not yet found any use for the Change Up To % setting or for the Discharge Down To % setting. Have you found any use for them? **@trevordowling Not used them**
2. Do you also find the same issue with Setting the Charge Up To? Does predbat only ever plan for Charging to 100% for you? Try setting AC Charge 1 Upper SOC % to something other than 100% and see if your setup corrects it? Please let us know. **@trevordowling I see the same issue - see below**

I set these setting in the GivEnergy Portal

[image]

But indeed, the charge up to status within Predbat plan remains at 100%

[image]

gcoan, 2025-03-01:
> But indeed, the charge up to status within Predbat plan remains at 100%
> 

What that is showing is that predbat is planning to charge the battery to 100% in that period, i.e. its the target SoC Predbat is trying to get to, not any kind of sensor value.

To change what sensors predbat writes to, this is done in apps.yaml (non REST mode).
@LeeFarm11 have you tried changing charge_limit from the default 'number.givtcp_xx_target_soc' to the charge_target_soc_1 ?

LeeFarm11, 2025-03-01:
**Predbat Status - Exporting**

Note - Since installation I have been getting alert every 5 minutes - Predbat: Inverter 0 Force export set to False at time xx:xx:00

to test this I set my rates_export to this -
```
  rates_export:
    -  start: "21:00:00"
       end: "21:30:00"
       rate: 500
    -  start: "21:30:00"
       end: "21:00:00"
       rate: 14
```
As expected plan wants to Charge until 21:00. Then Export from 21:00 to 21:30 and then Charge again from 21:30 to 22:00

[image]

AC Charge 1 End Time was successfuly changed from 22:00 to 21:00

At 21:00 the Charging switches were turned off.
Then DC Discharge 1 Start Time and DC Discharge 1 End Time were set to 21:00 and 21:31 (yes 21:31....)
Target SOC was set to 4% which did result in DC Discharge 1 Lower SOC % Limit being set to 4% as required. 
However Enable Force Discharge was NOT successfully turned on.

I then turned Enable Force Discharge on manually in the Portal.
I think predbat got confused at this point and tuned Charging back on for few minutes, but sorted itself out shortly after. (turns out it was confused. See further down)

Predbat Plan then showing Exporting to 67% and Home Assistant Alert said Predbat satatus changed to: Exporting target 79%-57%.
But I see nothing set at 67 or 57, not in GivTCP Overview or in Portal.
AC Discharge 1 Lower SOC % Limit does not show in Portal due to a problem with the Portal. but does show in HA GivTCP Overview as Discharge Target SOC 1. Showing 4%

At 21:15 predbat decides that it wants to start Charging again for no good reason so turns on the Charging settings again - increased AC Charge 1 End Time to 21:20 and sets the SCharging switches On, but leaves the Discharging switch also On (**seems predbat is not able to read or write to/of Enable Force Discharge**)

Then at 21:20 decides to turn off the Charging again which allows the Discharging to start again. Now alert says - Predbat status change to: Exporting target 77% - 63% and the Inverter is successfully exporting.

Then at 21:25 predbat decides to turn Charging back on again. Predbat status change to: Charging target 75%-100%, this time extending AC Charge 1 End Time to 22:00

Here is the resulting Portal Power Graph.

[image]

LeeFarm11, 2025-03-01:
> What that is showing is that predbat is planning to charge the battery to 100% in that period, i.e. its the target SoC Predbat is trying to get to, not any kind of sensor value.
>
> To change what [...]

Yes but the 100% needs to set somewhere. Currently it is setting the wrong value to that target. That is why I have created my small automation shown further up.

LeeFarm11, 2025-03-01:
> * I have not yet found any use for the Change Up To % setting or for the Discharge Down To % setting. Have you found any use for them? **[@trevordowling](https://github.com/trevordowling) Not used them**
> 
> * Do you also find the same issue with Setting the Charge Up To? Does predbat only ever plan for Charging to 100% for you? Try setting AC Charge 1 Upper SOC % to something other than 100% and see if your setup corrects it? Please let us know. **[@trevordowling](https://github.com/trevordowling) I see the same issue - see below**

Thanks. Good to know you see same as me.

LeeFarm11, 2025-03-01:
> > > But indeed, the charge up to status within Predbat plan remains at 100%
> > 
> > 
> > What that is showing is that predbat is planning to charge the battery to 100% in that period, i.e. its the target SoC Predbat is trying to get to, not any kind of sensor value.
> > To change what sensors predbat writes to, this is done in apps.yaml (non REST mode). [@LeeFarm11](https://github.com/LeeFarm11) have you tried changing charge_limit from the default 'number.givtcp_xx_target_soc' to the charge_target_soc_1 ?
> 
> Yes but the 100% needs to be set somewhere. Currently it is setting the wrong value to that target. That is why I have created my small automation shown further up.

Doh! You're right. As usual.
Instead of running my automation I can change it here in app.yaml -

  charge_limit:
    - number.givtcp_{geserial}_target_soc

I have now changed it to 

  charge_limit:
    #- number.givtcp_{geserial}_target_soc
    - number.givtcp_{geserial}_charge_target_soc_1

and I have disabled my automation

Thanks

LeeFarm11, 2025-03-01:
For turning on Export not working, and I suspect relating to the error every 5 minutes (Predbat: Inverter 0 Force export set to False at time xx:xx:00), we need to turn on this switch shown here in the Portal -

[image]

This does toggle this switch in HA Overview (works both ways. Switch it in Portal and it updates HA Overview. Switch it in HA Overview and it updates Portal)

[image]

Details here -

[image]

And in apps.yaml reference appears correct.

[image]

What am I missing?

gcoan, 2025-03-01:
> Doh! You're right. As usual.
> Instead of running my automation I can change it here in apps.yaml -
> 
> I have now changed it to
> 
>   charge_limit:
>     #- number.givtcp_{geserial}_target_soc
>     - number.givtcp_{geserial}_charge_target_soc_1

Thanks, I was confused by your first reply, looking at what the automation was doing, it seemed an obvious fix if the 3-phase doesn't use target_soc.

Obviously when we get to a working configuration I will add stuff to the documentation to help others, or we have a custom inverter type, or whatever.

One off the list I hope !

LeeFarm11, 2025-03-02:
> Thanks, I was confused by your first reply, looking at what the automation was doing, it seemed an obvious fix if the 3-phase doesn't use target_soc.
>
> Obviously when we get to a working [...]

So Predbat Status Charging now works (with modification to apps.yaml)

And Predbat Status Demand works.

Now we need to get the other 8ish statuses working.

First Exporting. See my notes above. Anyone got any ideas what I am missing in this post - https://github.com/springfall2008/batpred/issues/2057#issuecomment-2692441839

trevordowling, 2025-03-02:
Hi @LeeFarm11 ,

What specifically in terms of Export is not working for you?

Is it the export itself, setting the target level or something else?

As Export for me appears to be working OK - Predbat is exporting battery left before charging again at midnight without any apparent issues.

Last nights efforts before charging - exported about 10kwh's

[image]

Log from Inverter

[image]

trevordowling, 2025-03-02:
@LeeFarm11 

Also - to get the Status Charging working, you made this change to the apps.yaml file.

[image]

I assume you also switched of REST to force PredBat to use the entities listed.

I'm I correct?

LeeFarm11, 2025-03-02:
> Hi [@LeeFarm11](https://github.com/LeeFarm11) ,
> 
> What specifically in terms of Export is not working for you?

Setting Enable Force Dicharge does not work for me.
And it looks like it does not work for you either.
Looks like you turned on Enable Force Dicharge manually via the Giv Portal, and predbat did NOT control that change, otherwise it would show Server and External Change Detected. But it shows You and Portal - Remote Control ?-

Did Predbat/ GivTCP set Enable Force Discharge to True at 22:25:38? Or did you achieve it manually or by some other way?

[image]

LeeFarm11, 2025-03-02:
> I assume you also switched of REST to force PredBat to use the entities listed.

No. REST is still Enabled in apps.yaml

LeeFarm11, 2025-03-02:
> [@LeeFarm11](https://github.com/LeeFarm11)
> 
> Also - to get the Status Charging working, you made this change to the apps.yaml file.
> 
> [image]
> 
> I assume you also switched of REST to force PredBat to use the entities listed.
> 
> I'm I correct?

Actually, I did not properly test this last night.
I now see that the change I made in apps.yaml did not mark setting Charge % correct. Because REST is ON, the change I made in apps.yaml had no positive impact.

Because we are trying to get predbat / GivTCP working with REST ON, I will reinstate my automation to set AC Charge 1 Upper SOC % to be set to match Charge Up To % whenever Charge Up To % changes.

@gcoan Back to the drawing board I'm afraid.

I haven't yet found how to identify which REST control predbat tries to use for each action. For example how do I find what REST control predbat is trying to use when it is trying to set Force export to False?

gcoan, 2025-03-02:
OK, it might have been worth preserving with trying to get non-REST mode working for predbat.

Anyway, to answer your question,  the inverter handling is in inverter.py

there are lots and lots of functions in there

for example line 1320 adjust_battery_target() sets the battery charging soc target %

lines 1346-1358:

```
        if current_soc != soc:
            self.base.log("Inverter {} Current charge limit is {}% and new target is {}%".format(self.id, current_soc, soc))
            self.current_charge_limit = soc
            if self.rest_data:
                self.rest_setChargeTarget(soc)
            else:
                self.write_and_poll_value("charge_limit", self.base.get_arg("charge_limit", indirect=False, index=self.id), soc)

            if self.base.set_inverter_notify:
                self.base.call_notify("Predbat: Inverter {} Target SOC has been changed to {}% at {}".format(self.id, soc, self.base.time_now_str()))
            self.mqtt_message(topic="set/target_soc", payload=soc)
        else:
            self.base.log("Inverter {} Current Target SOC is {}%, already at target".format(self.id, current_soc))
```

you can see that the function rest_setChargeTarget() is called to set the charge target

that function is at line 2370 where it calls the REST function /setChargeTarget

so from this it could be reasonable to take the line that Predbat is doing the "right thing", its GivTCP that is not correctly translating the REST call into the correct register updates for your inverter type

which I think brings us back to either getting GivTCP changed so it maps the predbat REST command to the correct register updates, OR we work out what needs to be set in GivTCP and change Predbat.

If we are going down the latter it might be easier to go to non-REST mode as you can more directly change the controls you have worked out that are required for your inverter,  and then maybe with a detailed list of what needs to change in GivTCP we might be better able to get GivTCP changed?

Anyway, that's my read of what the Predbat code does, have a look and see what you think of the code and whether you come to the same conclusion

LeeFarm11, 2025-03-02:
But I thought that even if REST is disabled in apps.yaml, REST is still used for reading? Therefore if we adjust the setting like we did for setting charge Up To in apps.yaml and disable REST, then predbat will still read the wrong register and get itself confused keep trying to change the setting it already changed, and other things like that. Or do I still not fully understand that concept? 

Therefore we should leave REST ON and try to add automations until the code can be modified to do what is needed for these Inverters.

Except - if the issue is that there is no REST register for the required setting, in which case then we do need GivTCP to be modified. 

Potentially we should also be using GivTCP DEV, not BETA, because i think @britkat1980 has created differences between Dev and Beta for 3 phase. Mark, can you elaborate?

I think we need to agree the approach between you, me, and @trevordowling ?

I haven't yet taken a look at inverter.py but I will try to do so soon. I guess that the bit I am missing about 'Force export' here - https://github.com/springfall2008/batpred/issues/2057#issuecomment-2692441839 will be revealed in inverter.py?

[image]

trevordowling, 2025-03-02:
I'm still behind the curve on being able to help decide the best approach, but happy to align and support. 

If possible keeping things within Predbat makes sense, as we have complete control without the need for another 'team' (GIVTCP). 

There is no doubt that there are a number of differences and maybe enough to just consider this another Inverter type. 

I also see things like this in the main HA log

[image]

Which are of concern, until at least they are understood

LeeFarm11, 2025-03-03:
> If possible keeping things within Predbat makes sense, as we have complete control without the need for another 'team' (GIVTCP).

Unfortunately, I don't think this is possible because -

"Predbat will still retrieve information via REST" even when REST is disabled in apps.yaml
So if REST is disabled then Inverter is controlled via HA, but the current state of Inverter controls and other Inverter information is read via REST. So GivTCP REST is vitally important whether we want it or not. Or do I misunderstand?

[image]

LeeFarm11, 2025-03-03:
Is it possible to create an HA automation which will read and write DIRECTLY the GivTCP REST registers? Instead of reading to the GivTCP Control entities? I think I need to do that where the REST register does exist, but there is no HA entity mapped.

An example of how to do it would be appreciated.

Thanks.

trevordowling, 2025-03-03:
@LeeFarm11 

I'll produce a map between MQTT entities to the REST data. I've grabbed both within a second or so, so should be able to align both the names and values as a cross check. 

Might help us understand the size of the problem.

My assumption is: REST data is provided by the Inverter directly, where as MQTT entities are part of the GivTCP implementation.

Is this a correct assumption?

britkat1980, 2025-03-03:
> Is it possible to create an HA automation which will read and write DIRECTLY the GivTCP REST registers? Instead of reading or writing to the GivTCP Control entities? I think I need to do that where the REST register does exist, but there is no HA entity mapped.
> 
> An example of how to do it would be appreciated.
> 
> Thanks.
> 
> Edit: added 'or writing'

You cannot directly write to a numbered register in order to only expose registers which are safe. There are no such things as REST registers. The “raw” topic (in both rest and mqtt) shows the full register map.

LeeFarm11, 2025-03-03:
> You cannot directly write to a numbered register in order to only expose registers which are safe. There are no such things as REST registers. The “raw” topic (in both rest and mqtt) shows the full register map.

Thanks @britkat1980 

So in the example in this comment here - https://github.com/springfall2008/batpred/issues/2057#issuecomment-2692441839

predbat is using GivTCP REST and it is NOT using this -
[image]

When predbat is not using REST, predbat appears to be able to set Enable Force Discharge. 

When predbat is using REST, is someone able to explain why the Enable Discharge Schedule switch on the Inverter is not being switched by predbat? This is when predbat Status is switching to Exporting, as tested and shown in this comment here - https://github.com/springfall2008/batpred/issues/2057#issuecomment-2692431367 
Must it be a problem in the predbat code? My python knowledge is not yet so advanced. Here somewhere -?

**But wait. This function seems never called. 
rest_enableChargeSchedule is called elsewhere in the script, but rest_enableDischargeSchedule is never called in inverter.py
Can that be right?**

```
    def rest_enableDischargeSchedule(self, enable):
        """
        Configure enable discharge schedule via REST (V3.x+)
        """
        url = self.rest_api + "/enableDischargeSchedule"
        data = {"state": "enable" if enable else "disable"}

        for retry in range(5):
            r = self.rest_postCommand(url, json=data)
            # self.sleep(10)
            self.rest_data = self.rest_runAll(self.rest_data)
            new_value = self.rest_data["Control"]["Enable_Discharge_Schedule"]
            if isinstance(new_value, str):
                if new_value.lower() in ["enable", "on", "true"]:
                    new_value = True
                else:
                    new_value = False
            if new_value == enable:
                self.count_register_writes += 1
                self.base.log("Set inverter {} discharge schedule {} via REST successful on retry {}".format(self.id, enable, retry))
                return True

        self.base.log("Warn: Set inverter {} discharge schedule {} via REST failed got {}".format(self.id, enable, self.rest_data["Control"]["Enable_Discharge_Schedule"]))
        self.base.record_status("Warn: Inverter {} REST failed to enableDischargeSchedule".format(self.id), had_errors=True)
        return False
```

trevordowling, 2025-03-03:
@LeeFarm11 

Following on from the discharging question, here is the overview of the dicharge charge cycle 2nd March (20:00) - 3rd March (08:00) (approx).

[image]

Running with REST enabled, the discharge period was determined and executed as expected. I have deliberatly reduced the max export value, to lengthen the time period of discharge, which seemed to worked. Though I had to use the scaling in Predbat config

[image]

as the setting in apps.yaml 

[image]

did not appear to do what I had expected (though possibly I missundertood how this setting is used)

gcoan, 2025-03-03:
> "Predbat will still retrieve information via REST" even when REST is disabled in apps.yaml
> So if REST is disabled then Inverter is controlled via HA, but the current state of Inverter controls and [...]

You are correct, all retrieves are via REST BUT this is only the way GivEnergy inverters are coded.  Non-GE inverters can either use service calls or HA entities, whatever is needed.

So if we do make a new variant inverter for the GivEnergy 3 phase we can set it up to work in any way we want to get it working,  and then either fold it back into the main code or leave it as a custom inverter type.

There's outline instructions here: https://github.com/springfall2008/batpred/blob/main/docs/inverter-setup.md#i-want-to-add-an-unsupported-inverter-to-predbat
(although to be honest its not as well described as it could be)

I'm just thinking it might be easier if we work (mainly) with the GivTCP entities rather than trying to get REST to work.  I probably should have directed towards this route earlier on

> I also see things like this in the main HA log
> 
> [image]
> 
> Which are of concern, until at least they are understood

@trevordowling don't worry about those, this is covered in the FAQ's, these messages can be ignored or supressed https://github.com/springfall2008/batpred/blob/main/docs/faq.md#predbat-is-causing-warning-messages-about-exceed-maximum-size-in-the-home-assistant-core-log

> Potentially we should also be using GivTCP DEV, not BETA, because i think [@britkat1980](https://github.com/britkat1980) has created differences between Dev and Beta for 3 phase. Mark, can you elaborate?

I agree, use the latest code because there are bug fixes in it

> as the setting in apps.yaml
> 
> [image]
> 
> did not appear to do what I had expected (though possibly I missundertood how this setting is used)

this causes quite a few people confusion.  export_limit DOES NOT limit your exporting, it MODELS (in the predbat plan) any DNO imposed limit that is configured in your inverter by the installer.
https://github.com/springfall2008/batpred/blob/main/docs/apps-yaml.md#inverter-control-configurations

If you want to discharge the inverter at a specified level (i.e. not max discharge rate) then set inverter_limit_discharge

trevordowling, 2025-03-03:
@gcoan I'll work on supressing the error messages - thanks for the pointers - T

gcoan, 2025-03-03:
> [@gcoan](https://github.com/gcoan) I'll work on supressing the error messages - thanks for the pointers - T

I will (at some point, its on my list) be updating the documentation with improved guidance in this area because there's a load of guff being unnecessarily written to the Home Assistant database that doesn't need to be and these messages are a side effect of this.  However, the advice above is fine, it works, I can just do better.....

britkat1980, 2025-03-03:
@LeeFarm11 the REST interface does allow control of Force Discharge as per here: 

[image]

My guess if only that predbat doesn't know about them, as its a unique REST call for 3ph inverters

gcoan, 2025-03-04:
> My guess if only that predbat doesn't know about them, as its a unique REST call for 3ph inverters

thanks

at the moment Predbat treats the 3 phase the same as the other givenergy inverters, so won't use this specific REST call

@britkat1980 are there any other 3 phase specific calls predbat should be using?

LeeFarm11, 2025-03-04:
> [@LeeFarm11](https://github.com/LeeFarm11)
>
> Following on from the discharging question, here is the overview of the dicharge charge cycle 2nd March (20:00) - 3rd March (08:00) (approx).
>
> [image] [...]

Hi @trevordowling 
Your Discharge works because you have the Enable Force Discharge switch permanently On. This is possible so long as the time slots are always maintained correctly, which predbat probably does in your case. However, for other users, and whenever predbat decides that it does not want to discharge the same as previous day, it would not identify times to change the discharge slot to, therefore it would not change the time slot. Because you have turned the Force Discharge Enable switch On, and because predbat cannot turn it off, then the Inverter would Discharge anyway, even though predbat did not want it to.

So yes it is working for you. But it is not really working properly. It only discharges because you have the switch permanently On. And it will cause problems for other users. Potentially not for you, while you always have this overnight opportunity every night.

trevordowling, 2025-03-04:
Hi All,

I've done an initial review of the MQTT entities vs REST data (Inverter). The table below shows the entities that have different values (not a match) or are not present (N/A). The search was looking for MQTT values in the REST data set.

[image]

LeeFarm11, 2025-03-04:
@britkat1980 

I was previously running GivTCP Dev, before starting this Issue 2057, and I noticed some issues. 

1) I noticed that Force AC Charge Enable and Force Charge Enable are no longer controllable individually. I now see that there are situations where they are needed individually.
Are you able to add back that functionality, in addition to leaving Enable Charge Schedule switching both of them together?

2) And are you also able to add an entity which matches 'Power Factor Battery Reserve %' as described here - 
https://github.com/britkat1980/giv_tcp/issues/286#issuecomment-2665111843

@gcoan 

I wonder if I have now cleared up a misunderstanding I had. I did read Control as 'sending messages to the Inverter'. So disabling REST in apps.yaml would send controls to the Inverter via the settings in apps.yaml, but continue to read all settings from REST including the Control section. 
But now I think what Control means, is that with REST disabled, it will still read everything from REST EXCEPT the Control section, and instead will read and write those settings mentioned in apps.yaml.
So therefore now I agree I think it makes sense to run GivTCP DEV and with REST OFF in apps.yaml. Yes?

@trevordowling @britkat1980 

I think we will find that some of those settings are available in DEV and are not available in BETA.

Based on trevor's previous list of settings I found some additional information shown in the below prints.

I have also seen 'Export (Paused)'

[image]

[image]

LeeFarm11, 2025-03-04:
@britkat1980 
Here is GivTCP DEV 3.0.51 without REST trying to update DischargeRate to 0. It is not allowed, so I think that is why it fails.
But I have not yet seen it trying with a value greater than 0.
This supports my comment in one of the pictures in my previous post that 0% is not allowed in Charge Rate and Discharge Rate. Correction - 99% IS allowed. (Note - 99% is not allowed in Enable Charge Target)
I guess I need to add a helper to intercept Charge Rate or Discharge Rate and if it is 0, replace it with 1 before passing to the GivTCP entity?

predbat log -
```
2025-03-04 13:26:12.865159: Warn: Inverter 0 Trying to write 0 to discharge_rate didn't complete got 10000
2025-03-04 13:26:12.895425: Warn: record_status Warn: Inverter 0 write to discharge_rate failed
2025-03-04 13:28:02.066915: Warn: Inverter 0 Trying to write 0 to discharge_rate didn't complete got 10000
2025-03-04 13:28:02.096114: Warn: record_status Warn: Inverter 0 write to discharge_rate failed
2025-03-04 13:31:13.921307: Warn: Inverter 0 Trying to write 0 to discharge_rate didn't complete got 10000
2025-03-04 13:31:13.951827: Warn: record_status Warn: Inverter 0 write to discharge_rate failed
```

GivTCP log -
```
2025-03-04 13:24:13,240 - GivTCP - read        -  [DEBUG   ] - Error num 1 in watch loop execute command: ('Exception', 'read.py', 201)
2025-03-04 13:24:13,241 - GivTCP - read        -  [DEBUG   ] - Not running handler
2025-03-04 13:24:13,242 - GivTCP - read        -  [DEBUG   ] - Running partial refresh
2025-03-04 13:24:36,781 - GivTCP - read        -  [DEBUG   ] - Timeout Error: 
2025-03-04 13:24:36,781 - GivTCP - read        -  [DEBUG   ] - Error num 2 in watch loop execute command: ('Exception', 'read.py', 201)
2025-03-04 13:24:36,783 - GivTCP - read        -  [DEBUG   ] - Not running handler
2025-03-04 13:24:36,784 - GivTCP - read        -  [DEBUG   ] - Running partial refresh
2025-03-04 13:25:00,327 - GivTCP - read        -  [DEBUG   ] - Timeout Error: 
2025-03-04 13:25:00,328 - GivTCP - read        -  [DEBUG   ] - Error num 3 in watch loop execute command: ('Exception', 'read.py', 201)
2025-03-04 13:25:00,328 - GivTCP - read        -  [DEBUG   ] - Not running handler
2025-03-04 13:25:00,329 - GivTCP - read        -  [DEBUG   ] - Running partial refresh
2025-03-04 13:25:12,542 - GivTCP - mqtt        -  [DEBUG   ] - MQTT Message Recieved: GivEnergy/control/TD2343G049/setDischargeRate= 0
2025-03-04 13:25:12,542 - GivTCP - mqtt        -  [DEBUG   ] - Requesting Control Action: setDischargeRate - {'dischargeRate': '0'}
2025-03-04 13:25:22,590 - GivTCP - mqtt        -  [DEBUG   ] - MQTT Message Recieved: GivEnergy/control/TD2343G049/setDischargeRate= 0
2025-03-04 13:25:22,590 - GivTCP - mqtt        -  [DEBUG   ] - Requesting Control Action: setDischargeRate - {'dischargeRate': '0'}
2025-03-04 13:25:23,928 - GivTCP - read        -  [DEBUG   ] - Timeout Error: 
2025-03-04 13:25:23,928 - GivTCP - read        -  [DEBUG   ] - Error num 4 in watch loop execute command: ('Exception', 'read.py', 201)
2025-03-04 13:25:23,929 - GivTCP - read        -  [DEBUG   ] - Not running handler
2025-03-04 13:25:23,929 - GivTCP - read        -  [DEBUG   ] - Write Request recieved
2025-03-04 13:25:23,929 - GivTCP - read        -  [DEBUG   ] - Command: setDischargeRate was recieved: {'dischargeRate': '0'}
2025-03-04 13:25:32,647 - GivTCP - mqtt        -  [DEBUG   ] - MQTT Message Recieved: GivEnergy/control/TD2343G049/setDischargeRate= 0
2025-03-04 13:25:32,648 - GivTCP - mqtt        -  [DEBUG   ] - Requesting Control Action: setDischargeRate - {'dischargeRate': '0'}
2025-03-04 13:25:42,710 - GivTCP - mqtt        -  [DEBUG   ] - MQTT Message Recieved: GivEnergy/control/TD2343G049/setDischargeRate= 0
2025-03-04 13:25:42,710 - GivTCP - mqtt        -  [DEBUG   ] - Requesting Control Action: setDischargeRate - {'dischargeRate': '0'}
2025-03-04 13:25:48,488 - GivTCP - write       -  [ERROR   ] - Setting Discharge Rate failed: ('Exception', 'write.py', 612)
2025-03-04 13:25:48,793 - GivTCP - read        -  [DEBUG   ] - Command: setDischargeRate was recieved: {'dischargeRate': '0'}
2025-03-04 13:25:52,773 - GivTCP - mqtt        -  [DEBUG   ] - MQTT Message Recieved: GivEnergy/control/TD2343G049/setDischargeRate= 0
2025-03-04 13:25:52,775 - GivTCP - mqtt        -  [DEBUG   ] - Requesting Control Action: setDischargeRate - {'dischargeRate': '0'}
2025-03-04 13:26:02,836 - GivTCP - mqtt        -  [DEBUG   ] - MQTT Message Recieved: GivEnergy/control/TD2343G049/setDischargeRate= 0
2025-03-04 13:26:02,837 - GivTCP - mqtt        -  [DEBUG   ] - Requesting Control Action: setDischargeRate - {'dischargeRate': '0'}
2025-03-04 13:26:13,350 - GivTCP - write       -  [ERROR   ] - Setting Discharge Rate failed: ('Exception', 'write.py', 612)
2025-03-04 13:26:13,656 - GivTCP - read        -  [DEBUG   ] - Write Commands lengths: 2 -> 6
2025-03-04 13:26:13,657 - GivTCP - read        -  [DEBUG   ] - Looping back to mop up incoming write commands
2025-03-04 13:26:13,659 - GivTCP - read        -  [DEBUG   ] - Write Request recieved
2025-03-04 13:26:13,661 - GivTCP - read        -  [DEBUG   ] - Write Commands lengths: 0 -> 0
2025-03-04 13:26:13,663 - GivTCP - read        -  [DEBUG   ] - No new writes, removing writerequest file
2025-03-04 13:26:13,668 - GivTCP - read        -  [DEBUG   ] - Running full refresh
2025-03-04 13:26:44,073 - GivTCP - read        -  [DEBUG   ] - Timeout Error: 
[... cut ...]
2025-03-04 13:28:24,597 - GivTCP - write       -  [ERROR   ] - Setting Discharge Rate failed: ('Exception', 'write.py', 612)
2025-03-04 13:28:24,903 - GivTCP - read        -  [DEBUG   ] - Command: setDischargeRate was recieved: {'dischargeRate': '0'}
2025-03-04 13:28:49,465 - GivTCP - write       -  [ERROR   ] - Setting Discharge Rate failed: ('Exception', 'write.py', 612)
2025-03-04 13:28:49,772 - GivTCP - read        -  [DEBUG   ] - Write Commands lengths: 2 -> 2
2025-03-04 13:28:49,772 - GivTCP - read        -  [DEBUG   ] - No new writes, removing writerequest file
```

britkat1980, 2025-03-04:
If you share the limits of allowable values for each of those entities I can make the code work with them as per others (4-100%)

LeeFarm11, 2025-03-04:
Hi @britkat1980 

Here are the allowed values, as far as I can make out -

[image]

Are you also able to make these below changes please?

> I was previously running GivTCP Dev, before starting this Issue 2057, and I noticed some issues.
>
>     1. I noticed that Force AC Charge Enable and Force Charge Enable are no longer controllable [...]

britkat1980, 2025-03-04:
Thanks, are you sure you can't set AC rate to 100%? I can on my Gen1 single phase hybrid

britkat1980, 2025-03-04:
Looking at the latest dev code I think you can switch both those controls individually

britkat1980, 2025-03-04:
please try Dev 3.0.52 with fixes for the above

LeeFarm11, 2025-03-04:
Sorry. no idea why I said 1-99. I did mean 1 -100. Thanks.

trevordowling, 2025-03-05:
Hi @britkat1980 ,

I've just ran using Dev 3.0.52

[image]

I only installed the latest version of the GivTCP and then restarted HA (reboot). If I missed any other steps, please shout.

Some errors, maybe worth waiting to see what @LeeFarm11 finds as well

Thanks

@trevordowling

The log file looks like this...

2025-03-05 06:47:10,012 - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2025-03-05 06:47:10,029 - startup     -  [INFO    ] - Network Found: 192.168.1.147/24
2025-03-05 06:47:12,030 - startup     -  [INFO    ] - Searching for Inverters
2025-03-05 06:47:12,030 - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2025-03-05 06:47:20,510 - client      -  [INFO    ] - Connection established to 192.168.1.120:8899
2025-03-05 06:47:20,511 - client      -  [INFO    ] - Detecting plant
2025-03-05 06:47:24,928 - client      -  [INFO    ] - Plant Detected
2025-03-05 06:47:36,971 - startup     -  [INFO    ] - Inverter TD2335G007 which is a Gen1 - Hybrid_3ph with 6 batteries has been found at: 192.168.1.120
2025-03-05 06:47:36,986 - startup     -  [INFO    ] - Setting up invertor: 1
2025-03-05 06:47:37,057 - startup     -  [INFO    ] - ==============================================================
2025-03-05 06:47:37,057 - startup     -  [INFO    ] - ====             Web Gui Config is at                     ====
2025-03-05 06:47:37,057 - startup     -  [INFO    ] - ====     http://192.168.1.147:8099/config.html             ====
2025-03-05 06:47:37,057 - startup     -  [INFO    ] - ==============================================================
2025-03-05 06:47:37,058 - startup     -  [INFO    ] - Running Invertor 1 (TD2335G007) read loop every 30/120s
2025-03-05 06:47:37,062 - startup     -  [INFO    ] - Serving Web Dashboard from port 3000
2025-03-05 06:47:37,077 - startup     -  [INFO    ] - Setting daily charge target forecast job to run at: 23:50
[2025-03-05 06:47:37 +0000] [50] [INFO] Starting gunicorn 23.0.0
[2025-03-05 06:47:37 +0000] [50] [INFO] Listening at: http://0.0.0.0:6350 (50)
[2025-03-05 06:47:37 +0000] [50] [INFO] Using worker: sync
[2025-03-05 06:47:37 +0000] [66] [INFO] Booting worker with pid: 66
[2025-03-05 06:47:37 +0000] [60] [INFO] Starting gunicorn 23.0.0
[2025-03-05 06:47:37 +0000] [60] [INFO] Listening at: http://0.0.0.0:6345 (60)
[2025-03-05 06:47:37 +0000] [60] [INFO] Using worker: sync
[2025-03-05 06:47:37 +0000] [67] [INFO] Booting worker with pid: 67
[2025-03-05 06:47:37 +0000] [68] [INFO] Booting worker with pid: 68
[2025-03-05 06:47:37 +0000] [69] [INFO] Booting worker with pid: 69
2025-03-05 06:47:37,768 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
2025-03-05 06:47:37,769 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.1.120
2025-03-05 06:47:37,770 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-03-05 06:47:56,030 - GivTCP - read        -  [ERROR   ] - Error processing Three Phase data: ((<class 'KeyError'>, KeyError('force_discharge_enable'), <traceback object at 0x7fda02b31380>), 1864)
2025-03-05 06:47:56,031 - GivTCP - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7fda02b47fc0>), 1939)
2025-03-05 06:47:56,032 - GivTCP - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 2120)
2025-03-05 06:48:30,499 - GivTCP - read        -  [ERROR   ] - Error processing Three Phase data: ((<class 'KeyError'>, KeyError('force_discharge_enable'), <traceback object at 0x7fda02f79000>), 1864)
2025-03-05 06:48:30,499 - GivTCP - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7fda02f78fc0>), 1939)
2025-03-05 06:48:30,499 - GivTCP - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 2120)

LeeFarm11, 2025-03-05:
Just updated to 3.0.52 and I have same issue. Repeating same error loop every 30 seconds or so.

```
2025-03-05 08:16:20,032 - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2025-03-05 08:16:20,066 - startup     -  [INFO    ] - Network Found: 192.168.1.204/24
2025-03-05 08:16:22,068 - startup     -  [INFO    ] - Searching for Inverters
2025-03-05 08:16:22,068 - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2025-03-05 08:16:23,413 - startup     -  [INFO    ] - GivEVC 11288853538258 found at: 192.168.1.102
2025-03-05 08:16:29,889 - client      -  [INFO    ] - Connection established to 192.168.1.101:8899
2025-03-05 08:16:29,889 - client      -  [INFO    ] - Detecting plant
2025-03-05 08:16:34,312 - client      -  [INFO    ] - Plant Detected
2025-03-05 08:16:39,842 - startup     -  [INFO    ] - Inverter TD2343G049 which is a Gen1 - Hybrid_3ph with 5 batteries has been found at: 192.168.1.101
2025-03-05 08:16:39,855 - startup     -  [INFO    ] - Setting up invertor: 1
2025-03-05 08:16:39,974 - startup     -  [INFO    ] - ==============================================================
2025-03-05 08:16:39,974 - startup     -  [INFO    ] - ====             Web Gui Config is at                     ====
2025-03-05 08:16:39,975 - startup     -  [INFO    ] - ====     http://192.168.1.204:8099/config.html             ====
2025-03-05 08:16:39,975 - startup     -  [INFO    ] - ==============================================================
2025-03-05 08:16:39,980 - startup     -  [INFO    ] - Running Invertor 1 (TD2343G049) read loop every 30/120s
2025-03-05 08:16:39,989 - startup     -  [INFO    ] - Serving Web Dashboard from port 3000
[2025-03-05 08:16:40 +0000] [50] [INFO] Starting gunicorn 23.0.0
[2025-03-05 08:16:40 +0000] [50] [INFO] Listening at: http://0.0.0.0:6350 (50)
[2025-03-05 08:16:40 +0000] [50] [INFO] Using worker: sync
[2025-03-05 08:16:40 +0000] [62] [INFO] Booting worker with pid: 62
[2025-03-05 08:16:41 +0000] [58] [INFO] Starting gunicorn 23.0.0
[2025-03-05 08:16:41 +0000] [58] [INFO] Listening at: http://0.0.0.0:6345 (58)
[2025-03-05 08:16:41 +0000] [58] [INFO] Using worker: sync
[2025-03-05 08:16:41 +0000] [63] [INFO] Booting worker with pid: 63
[2025-03-05 08:16:41 +0000] [64] [INFO] Booting worker with pid: 64
[2025-03-05 08:16:41 +0000] [65] [INFO] Booting worker with pid: 65
2025-03-05 08:16:44,916 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
2025-03-05 08:16:44,917 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.1.101
2025-03-05 08:16:44,923 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-03-05 08:17:07,693 - GivTCP - read        -  [DEBUG   ] - Running full refresh
2025-03-05 08:17:07,693 - GivTCP - read        -  [DEBUG   ] - Running processData
2025-03-05 08:17:07,693 - GivTCP - read        -  [DEBUG   ] - Beginning parsing of Inverter data
2025-03-05 08:17:07,744 - GivTCP - read        -  [DEBUG   ] - Getting Battery Details
2025-03-05 08:17:07,746 - GivTCP - read        -  [DEBUG   ] - Building battery output: 
2025-03-05 08:17:07,748 - GivTCP - read        -  [DEBUG   ] - Battery HY2343G257 added
2025-03-05 08:17:07,748 - GivTCP - read        -  [DEBUG   ] - Building battery output: 
2025-03-05 08:17:07,751 - GivTCP - read        -  [DEBUG   ] - Battery HY2343G446 added
2025-03-05 08:17:07,751 - GivTCP - read        -  [DEBUG   ] - Building battery output: 
2025-03-05 08:17:07,753 - GivTCP - read        -  [DEBUG   ] - Battery HY2343G302 added
2025-03-05 08:17:07,754 - GivTCP - read        -  [DEBUG   ] - Building battery output: 
2025-03-05 08:17:07,755 - GivTCP - read        -  [DEBUG   ] - Battery HY2343G025 added
2025-03-05 08:17:07,756 - GivTCP - read        -  [DEBUG   ] - Building battery output: 
2025-03-05 08:17:07,757 - GivTCP - read        -  [DEBUG   ] - Battery HY2343G144 added
2025-03-05 08:17:07,758 - GivTCP - read        -  [DEBUG   ] - Getting mode control figures
2025-03-05 08:17:07,759 - GivTCP - read        -  [DEBUG   ] - Calculating Mode...
2025-03-05 08:17:07,760 - GivTCP - read        -  [DEBUG   ] - Mode is: Timed Export
2025-03-05 08:17:07,761 - GivTCP - read        -  [DEBUG   ] - Force Export is not Running
2025-03-05 08:17:07,761 - GivTCP - read        -  [ERROR   ] - Error processing Three Phase data: ((<class 'KeyError'>, KeyError('force_discharge_enable'), <traceback object at 0x7f0426072840>), 1864)
2025-03-05 08:17:07,762 - GivTCP - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7f0426072880>), 1939)
2025-03-05 08:17:07,763 - GivTCP - read        -  [DEBUG   ] - processData result: processData Error processing registers: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7f0426072880>), 1939)
2025-03-05 08:17:07,763 - GivTCP - read        -  [DEBUG   ] - Running pubFromPickle
2025-03-05 08:17:07,763 - GivTCP - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 2120)
2025-03-05 08:17:37,811 - GivTCP - read        -  [DEBUG   ] - Running partial refresh
2025-03-05 08:17:40,485 - GivTCP - read        -  [DEBUG   ] - Data get was successful, now running handler if needed: 
2025-03-05 08:17:40,486 - GivTCP - read        -  [DEBUG   ] - Running processData
[... cut ...]
2025-03-05 08:17:40,518 - GivTCP - read        -  [DEBUG   ] - processData result: processData Error processing registers: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7f0426076340>), 1939)
2025-03-05 08:17:40,518 - GivTCP - read        -  [DEBUG   ] - Running pubFromPickle
2025-03-05 08:17:40,519 - GivTCP - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 2120)
2025-03-05 08:18:10,557 - GivTCP - read        -  [DEBUG   ] - Running partial refresh
```

LeeFarm11, 2025-03-05:
> I guess I need to add a helper to intercept Charge Rate or Discharge Rate and if it is 0, replace it with 1 before passing to the GivTCP entity?

To do this, I need to point the relevant setting to some kind of helper/automation in apps.yaml. I can do that.

The helper/automation needs to intercept the value sent, and if it is 0 change to 1, and then send on the revised value to the relevant entity. That I think I can build.

BUT, because we're not using REST, I assume predbat also needs to read the value via the same helper/ automation? I don't think I've seen that before. How do I build that into the helper/automation please? I guess some kind of Return statement? And while doing that read, predbat needs to think that the value has been set to 0, even though it has been set to 1. Sorry for the newbie question.

britkat1980, 2025-03-05:
> Just updated to 3.0.52 and I have same issue. Repeating same error loop every 30 seconds or so.
> 
Apologies, a simple error by me, I'll push out an update this evening after work

britkat1980, 2025-03-05:
> > I guess I need to add a helper to intercept Charge Rate or Discharge Rate and if it is 0, replace it with 1 before passing to the GivTCP entity?

Latest Dev will not allow 0 to be sent, so write request will fail

LeeFarm11, 2025-03-05:
> > > I guess I need to add a helper to intercept Charge Rate or Discharge Rate and if it is 0, replace it with 1 before passing to the GivTCP entity?
> 
> Latest Dev will not allow 0 to be sent, so write request will fail

But we anyway need to intercept at predbat, because predabt will try to send 0, to the alias in apps.yaml, and then the alias will have to do something with it. But predbat will still be upset if it tries to read (via the alias) and finds 1 instead of 0.

LeeFarm11, 2025-03-06:
Strange readings -

[image]

[9 more replies]

closed 2026-09-09 by chalfontchubby