springfall2008/batpred #2058: Agile prices in compare not right

mogons57, 2025-02-28:
**Describe the bug**
Predbat compare doesn't seem to using the correct agile import or export prices in the midnight runs.

**Expected behaviour**

Doing the API call in a browser in the morning i.e.
https://api.octopus.energy/v1/products/AGILE-24-10-01/electricity-tariffs/E-1R-AGILE-24-10-01-K/standard-unit-rates/
gives me

{
  "count": 7968,
  "next": "https://api.octopus.energy/v1/products/AGILE-24-10-01/electricity-tariffs/E-1R-AGILE-24-10-01-K/standard-unit-rates/?page=2",
  "previous": null,
  "results": [
    {
      "value_exc_vat": 20.02,
      "value_inc_vat": 21.021,
      "valid_from": "2025-02-28T22:30:00Z",
      "valid_to": "2025-02-28T23:00:00Z",
      "payment_method": null
    },
    {
      "value_exc_vat": 20.75,
      "value_inc_vat": 21.7875,
      "valid_from": "2025-02-28T22:00:00Z",
      "valid_to": "2025-02-28T22:30:00Z",
      "payment_method": null
    },
etc.

using the octopus plugin I get
22:00 21.8p
22:30 21.0p

which agrees with the API call

in Predbat compare for my current tariff I get
22:00 21.79p
22:30 21.02p

again agrees with Agile tariff I'm on

However in the compare table for Agile import/Fixed export or Agile import/Agile export

the agile import prices are showing as
22:00 13.40p
22:30 9.34p

which are very different to the API numbers

the agile export prices in the compare tables are also odd at
22:00 6.42p
22:30 4.73p

As far as I can tell the prices in other tables match the API calls - just the Agile ones look odd from the run done at midnight
I've tried trying to find the compare reported agile prices but cant find them in a different region or on a previous day.

When compare is run manually (in this case at about 10:30 during the day the compare was originally done for) the agile tariffs in the tables are now correct

i.e.
Agile Import
  22:00 21.79p
  22:30 21.02p
Agile Export
  22:00 9.84p
  22:30 9.52p

It looks like the numbers coming into compare when it runs at midnight are wrong or they are not being processed properly.

**Predbat version**

5.15.1

**Environment details**

- Givenergy with 9.5kW battery
- Standard install

springfall2008, 2025-03-01:
Do you have Nordpool enabled, are the rates for that period computed rather than actual?

mogons57, 2025-03-02:
Nordpool is enabled and the 2 times I picked are just examples. The whole day shows the wrong agile prices.

mogons57, 2025-03-12:
Not sure what has changed behind the scenes but everything seems to be working now. Agile prices coming into compare correctly now. Problem fixed.

closed 2025-03-12 by mogons57