springfall2008/batpred #2061: Exception raised while updating state of sensors dc_status, system_mode and battery_priority (3 phase based system)

trevordowling, 2025-02-28:
**Describe the bug**

An exception is raised while updating state of sensor dc_status, system_mode and battery_priority.

(**Note**: possibly this is not a bug in PredBat but more an issue with the GivTCT MQTT defs and the differences between the single and three phase systems)

Error entry is only generated if not running REST (as far as I can tell)

(ERROR) MQTT

[image]

**Home Assistant Core - Log Details**

      Logger: homeassistant.components.mqtt.models
      Source: components/mqtt/models.py:366
      integration: MQTT (documentation, issues)
      First occurred: 27 February 2025 at 22:26:36 (8 occurrences)
      Last logged: 07:37:13
      
      Exception raised while updating state of sensor.givtcp_{geserial]_dc_status, topic: 'GivEnergy/{geserial]/{geserial]/DC_Status' with payload: b'Normal'
      Exception raised while updating state of sensor.givtcp_{geserial]_system_mode, topic: 'GivEnergy/{geserial]/{geserial]/System_Mode' with payload: b'Grid_tied'
      Exception raised while updating state of sensor.givtcp_{geserial]_battery_priority, topic: 'GivEnergy/{geserial]/{geserial]/Battery_Priority' with payload: b'Load'
      Exception raised while updating state of sensor.givtcp_{geserial]_battery_priority, topic: 'GivEnergy/{geserial]/{geserial]/Battery_Priority' with payload: b'Battery'
      Exception raised while updating state of sensor.givtcp_{geserial]_dc_status, topic: 'GivEnergy/{geserial]/{geserial]/DC_Status' with payload: b'Waiting'
      Traceback (most recent call last):
        File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 635, in state
          numerical_value = int(value)
      ValueError: invalid literal for int() with base 10: 'Grid'
      
      During handling of the above exception, another exception occurred:
      
      Traceback (most recent call last):
        File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 638, in state
          numerical_value = float(value)
      ValueError: could not convert string to float: 'Grid'
      
      The above exception was the direct cause of the following exception:
      
      Traceback (most recent call last):
        File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 366, in process_write_state_requests
          entity.async_write_ha_state()
          ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
        File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1023, in async_write_ha_state
          self._async_write_ha_state()
          ~~~~~~~~~~~~~~~~~~~~~~~~~~^^
        File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1148, in _async_write_ha_state
          self.__async_calculate_state()
          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
        File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1085, in __async_calculate_state
          state = self._stringify_state(available)
        File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1029, in _stringify_state
          if (state := self.state) is None:
                       ^^^^^^^^^^
        File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 642, in state
          raise ValueError(
          ...<5 lines>...
          ) from err
      ValueError: Sensor sensor.givtcp_{geserial]_battery_priority has device class 'None', state class 'None' unit '' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: 'Grid' (<class 'str'>)

Looking these entities up in the Developer Status shows that they are listed as State 'unknown'

[image]

However reviewing the REST data dump (http://[192.168.1.147:6345/readData]) all three entities are visible and defined.
    
    "System_Mode": "Grid_tied"
    "DC_Status": "Normal"
    "Battery_Priority": "Load"

**Expected behaviour**
No MQTT error reported

**Predbat version**

v8.15.1

**Environment details**

GIV-3HY-11
GivEnergy battery module stack - 51.8 Ah | 20.29 kWh | (6 Packs)
HAOS on PC hardware (dedicated)

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

**Predbat debug yaml file**
Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to predbat_debug_XX_XX_XX.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

[predbat_debug_15_40_00.txt](https://github.com/user-attachments/files/19029927/predbat_debug_15_40_00.txt)

gcoan, 2025-03-01:
definitely looks like a givtcp issue not predbat

predbat is issuing the command to givtcp and for whatever reason its not working correctly for your inverter

chalfontchubby, 2026-08-29:
Thanks for the report. This issue has been open for over 6 months with no recent activity - we're volunteers working through a large backlog and need to filter what's still relevant. If this is still a problem for you, please let us know and we'll keep it open. Otherwise, we'll close it in a week (around 2026-09-05).

chalfontchubby, 2026-09-09:
Closing as flagged - no response to the note above, so we're assuming this is no longer live.

If you're still hitting this, please reopen or open a fresh issue with a current version number and a `predbat_debug_*.yaml` captured while it's happening; we can replay that directly against current code.

_Posted by Claude on Rik's behalf._

closed 2026-09-09 by chalfontchubby