springfall2008/batpred #2062: Predbat crashes when can't connect to metoalarm

gcoan, 2025-02-28:
**Describe the bug**
In brief (longer explanation below), when predbat tries to connect to meteoalarm (e.g. internet down), if this fails, then predbat crashes and stops executing the plan.

Could be similar to #2340

Longer version:

Took me a bit to work this out, but internet went down earlier today (twice) first time at about 13:28.  I was unable to connect remotely and the GivEnergy app and portal showed that there were no updates received after 13:25.
I surmised (correctly) that the internet had gone down.

Looking at the predbat logs I can see it attempts to connect to meteoalarm, fails as the internet is down, crashes, and this process repeats until the internet comes back 2 hours later.

My inverters were left stuck in export and didn't start charging in the power up event at 13:30.
When internet came back at about 15:00 predbat correctly swapped inverters to charging mode.

**Expected behaviour**
Failure to connect to meteoalarm or indeed any external service including Octopus, Solcast, Github, etc should not be a fatal error.  Wherever possible Predbat should continue executing the plan with the latest data it has

**Predbat version**
8.14.4

**Environment details**
GivTCP 3.04
HAOS 2025.2.5
Gen 1 hybrid inverter x2

**Screenshots**
GivTCP log shows no activity betweem 13:25 and 15:10
[image]

Home Assistant entity state show that HA was connected to the inverters OK, receiving PV power and exporting it, and receiving regular inverter time statii updates.  However predbat stuck in force export mode
[image]

I don't have the actual plan, but it was to force export up to 13:30, then charge from 13:30 to 16:00 (2 hour power up then Cosy cheap rate), then Demand mode for the peak.

**Log file**
I'll try to capture the full logfile when I have better internet connection, but screenshots of the error:

[image]
[image]
[image]
[image]

gcoan, 2025-07-18:
@springfall2008 any chance you could look at this Trefor?

chalfontchubby, 2026-08-27:
Closing this as believed fixed. Predbat crashing when it can't reach meteoalarm looks to have been addressed since this was reported - the alert-feed fetch now has proper error handling around the network call (`alertfeed.py`'s `run()` wraps the whole fetch in a try/except that logs a warning and returns rather than propagating, with its own inner try/except around the download itself), so an outage there shouldn't take Predbat down with it.

We haven't reproduced a live meteoalarm outage to confirm directly, so if you hit this again on a current version, please open a new ticket with a log covering the failure - happy to take another look.

closed 2026-08-27 by chalfontchubby

gcoan, 2026-08-27:
thanks, I wasn't sure whether this had had proper error handling added.  I stopped using meteoalarm because of this, I'll try it again

chalfontchubby, 2026-08-27:
Code inspection indicates there is support in place for exceptions, and it's now fairly old with no recent comments.