springfall2008/batpred #2066: Actual charge rates not in HA service calls

robertugo2, 2025-03-02:
**Describe the bug**
For service calls the power for charging and discharging is not included. Only max/min values are included as {power}, see:
https://github.com/springfall2008/batpred/blob/725c1e82ca2a78f3929e41c7f48a021528fe0250/apps/predbat/inverter.py#L2126

**Expected behaviour**
I would expect, that the calculated charge/discharge values are passed to service call. So as you will see in logs, the caluculated charge rate is 1200W and unfortunately this value is not included as {power} value. Is it possible to add this? I tried to do that by my own, but unfortunately the code seems to be too complicated for me :(
Value is successfully passed to https://github.com/springfall2008/batpred/blob/725c1e82ca2a78f3929e41c7f48a021528fe0250/apps/predbat/inverter.py#L1239 function, so maybe it would be possible to store this variable as class property, that can be picked up and set to {power}? I also see many possible workarounds (set mqtt_api to true and get charging power from there and so on), but ideally I would like to have that available in service call. Or at least as something like `sensor.predbat_current_charge_power` in HA, because as I see it is not exposed as sensor.

Please let me know if I can help you somehow.

**Predbat version**

v8.15.1 Minor fixes & improvements, new compare_active entity

**Environment details**

I'm using a Sofar inverter controlled by the Solax Modbus integration. Thus that I defined my inverter as "MINE" and I would like to control some helper instances in HA. 
My config you will find here: https://gist.github.com/robertugo2/f422316253f8a0f81760eb11633d532c

**Screenshots**
Not applicable

**Log file**
```
2025-03-02 10:35:03.940525: Inverter 0 Target SOC 83 (this inverter 83) Battery temperature 20 Select charge rate 1200w (real 1200.0w) current charge rate 2600
(...)
2025-03-02 10:35:03.941100: Inverter 0 adjust target soc for charge to 83% (8.5kWh/10.24kWh 10.24kWh) based on going from 25% -> 83% total add is 5.94kWh and this battery needs to add 5.94kWh to get to 8.5kWh
2025-03-02 10:35:03.941115: Inverter 0 Current Target SOC is 83%, already at target
2025-03-02 10:35:03.941134: Current SOC 25% is less than Target SOC 83. Grid charging enabled with charge current set to 65.00
2025-03-02 10:35:03.941167: Inverter 0 Calling service discharge_stop_service domain discharge service_name input_select/select_option with data {'entity_id': 'input_select.predbat_discharge_mode', 'option': 'Stop'}
2025-03-02 10:35:03.949040: Inverter 0 Calling service charge_start_service domain charge service_name input_select/select_option with data {'entity_id': 'input_select.predbat_charge_mode', 'option': 'Start'}
2025-03-02 10:35:03.956250: Inverter 0 Calling service charge_start_service domain charge service_name input_number/set_value with data {'entity_id': 'input_number.predbat_charge_rate', 'value': '5000'}
2025-03-02 10:35:03.963715: Inverter 0 Calling service discharge_stop_service domain discharge service_name input_select/select_option with data {'entity_id': 'input_select.predbat_discharge_mode', 'option': 'Stop'}
```

**Predbat debug yaml file**
Not applicable I think.

springfall2008, 2025-03-09:
Fix on 'main' for testing

chalfontchubby, 2026-08-20:
Believed fixed by #2088. Please open a new issue if there is still a problem.

closed 2026-08-20 by chalfontchubby