springfall2008/batpred #2069: Prebat strange behaviour around Savings Session

johnwb87, 2025-03-03:
**Describe the bug**
Predbat is making poor decisions and intervening too much once enrolled in Savings Session.

It is a sunny day, before the Saving Session came through we had done some early discharging and where sat in Demand, charging the battery for the rest of the day, until planned export later. I use carbon metric so the export was planned in what would become the Saving Session window.

Post enrolling in the saving session I notice Predbat status changed briefly to Charging(!) and has now planned a big block of Freeze Export through the afternoon leaving no battery to export during the Saving Session.

**Expected behaviour**
Produce a sensible plan that makes sensible interventions.

**Predbat version**

8.15.1

**Environment details**

- Gen3 Hybrid 5.0 
- Predbat as Add-on

**Screenshots**

[image]

**Log file**

[predbat (3).log](https://github.com/user-attachments/files/19053148/predbat.3.log)

**Predbat debug yaml file**
I turned on debug and will have that file soon after the next run.

johnwb87, 2025-03-03:
And now this shocker. After which I have promptly used the forced demand selectors.

[image]

Debug for that plan run.

[predbat_debug_12_40_00.yaml.txt](https://github.com/user-attachments/files/19053575/predbat_debug_12_40_00.yaml.txt)

springfall2008, 2025-03-08:
I had a look at this, its a combination of factors, but you have a 45p/kg weighting on Carbon so exporting during high carbon times can be selected ahead of raw cost. You however also have best_soc_keep set to 1.0 which means if the export takes you below 1 kWh then it can trigger a small charge as a result.

Setting best_soc_keep back to 0 seems to clean up the plan (as would turning off the carbon).

springfall2008, 2025-03-09:
I also found an issue with calculate_export_oncharge when its disabled, fix in the next release

closed 2026-08-21 by johnwb87