springfall2008/batpred #2071: Hold for Car - But not car charging taking place - IOG

robyholmes, 2025-03-03:
**Describe the bug**
Predbat is putting itself into hold for car, even though the car isn't charging and there are no IGO slots planned let along currently active.

**Expected behaviour**
Predbat should be following it's normal plan.

**Predbat version**

v8.15.1 (Running on HA via Addon)

**Environment details**

- GE Hybrid 5KW Gen 3
- Standard HAOS
- Hypervolt Home Pro 3 Charger - Integration via HA

**Screenshots**

[image]

**Log file**

[predbat.log](https://github.com/user-attachments/files/19056439/predbat.log)

**Predbat debug yaml file**

[predbat_debug_16_40_00.yaml.txt](https://github.com/user-attachments/files/19056568/predbat_debug_16_40_00.yaml.txt)

robyholmes, 2025-03-03:
The issue seems to be that Predbat is planning the car charging, which is what it used to do before we moved to IGO for car charging.

robyholmes, 2025-03-03:
Might have discovered the issue, I had the following commented out in my apps.yaml
`  octopus_intelligent_slot: 're:(binary_sensor.octopus_energy([0-9a-z_]+|)_intelligent_dispatching)'
  octopus_ready_time: 're:(time.octopus_energy([0-9a-z_]+|)_intelligent_ready_time)'
  octopus_charge_limit: 're:(number.octopus_energy([0-9a-z_]+|)_intelligent_charge_limit)'`

Will report back if this does fix it. Currently it's not planning charge so I don't know for sure if it's sorted it.

gcoan, 2025-03-03:
If you have moved from predbat-led charging to iGo led charging then there are a number of reconfigurations you need to make to your predbat setup so that predbat takes the car charging slots from Octopus rather than plan them itself

uncommenting this line is one of them!

please check the docs against your setup

https://springfall2008.github.io/batpred/car-charging/

robyholmes, 2025-03-03:
Yes I've done all the others. It's been running fine all winter. I suspect because Predbat was planning charges during the night when IGO was. Now the sun has come out, predbat is planning more during the day so the issue has appeared.

This, plus our cars SOC isn't updating as fast (due to a number of reason, not needed for IGO) meant predbat wasn't seeing the car charged when it was.

chalfontchubby, 2026-08-20:
Thanks for the report. This issue has been open for over a year with no recent activity - we're volunteers working through a large backlog and need to filter what's still relevant. If this is still a problem for you, please let us know and we'll keep it open. Otherwise, we'll close it in a week (around 2026-08-28).

robyholmes, 2026-08-20:
Not seen the issue again so I believe clean up of Apps.yaml resolved it.

Thanks Team!

closed 2026-08-20 by robyholmes