springfall2008/batpred #2072: Unnecessary battery cycling

johnwb87, 2025-03-04:
**Describe the bug**
Sunny day and Predbat is choosing to cycle between Demand and Export within the same 30 min slot, result in a held SOC for the slot.

**Expected behaviour**
Just Freeze Export to hold the SOC

**Predbat version**

8.15.1

**Environment details**

- GE Gen3 Hybrid 5kW + 9.5kW Battery
- Standard HAOS installer and Add-on.
- Anything else?

**Screenshots**

[image]

**Log file**

[predbat (4).log](https://github.com/user-attachments/files/19069123/predbat.4.log)

**Predbat debug yaml file**
Just turned on. WIll upload, but have also forced Freeze export now.

SwiftRR, 2025-03-04:
My eyes are drawn towards the iBoost column. 
I do not use a solar diverter myself, but doesn't this column suggest a request for 4.67 kWh in every slot?

Rob

johnwb87, 2025-03-04:
No. That means that much has been delivered today. I don’t have iBoost on solar switched on. I use it on energy rates.

johnwb87, 2025-03-05:
I know that predbat plans in 5 minute slots, which is what allows the splits 30 minute slots to occur.

I was thinking on this and wondered if there is any filtering that occurs to remove largely redundant split slots as a single action would be more efficient and economical from a battery cycle and inverter registry write perspective.

springfall2008, 2025-03-08:
Can you capture a debug yaml next time you see this, you can just download it from the web UI you don't need to turn on debug?

closed 2026-08-20 by chalfontchubby