springfall2008/batpred #2073: Zero-rated import not causing battery to discharge beforehand

chris-y, 2025-03-04:
**Describe the bug**
When I have a zero-rated period (power-up) Predbat was discharging as much as possible before the session, so it could charge the battery back up during the session.  For some reason this is no longer happening the last couple of times.  I'm using 8.16.0 beta but it started before I tried that.

**Expected behaviour**
Predbat should discharge the battery as much as possible to maximise revenue, and then charge it back up whilst it is free.

**Predbat version**

8.16.0 beta

**Environment details**

- Solis inverter with custom scripts
- Standard HAOS installer

**Screenshots**

Plan HTML:
[predbat_plan.zip](https://github.com/user-attachments/files/19074364/predbat_plan.zip)

**Predbat debug yaml file**

[predbat_debug.zip](https://github.com/user-attachments/files/19074463/predbat_debug.zip)

springfall2008, 2025-03-08:
Setting metric_min_improvement_export to 3 appears to fix this

[image]

chris-y, 2025-03-17:
It's weird, this has been ok, but there's another session tomorrow and Predbat decided on no export beforehand and no charging during the session.

I've force export for the half hour before the free electricity, and now it is exporting beforehand (although quite a long time beforehand, not immediately before) and charging during the session.

gcoan, 2025-03-17:
> It's weird, this has been ok, but there's another session tomorrow and Predbat decided on no export beforehand and no charging during the session.

would need to see the plan, but one reason could be that if the solar forecast is greater than your inverter charge capacity then there's no benefit to grid importing at 0p

I'm seeing some of this happening tomorrow for the power up, my max charge rate is 2x2.6kW, so 2.6kW every half hour, but solar generation is more than this

[image]

chris-y, 2025-03-17:
Could be; there's a lot of solar predicted tomorrow.

closed 2026-08-28 by chalfontchubby