springfall2008/batpred #2077: Integrate Sigenergy Signestore

azebro, 2025-03-05:
Would it be possible to integrate Sigenergy Sigenstore?
The Sigenergy is already integrated with HA via: [](https://github.com/TypQxQ/Sigenergy-Home-Assistant-Integration)

gcoan, 2025-03-06:
since none of us have that inverter so can't try it out, the best bet (assuming you have one) is to follow the Predbat instructions for adding an unsupported inverter https://springfall2008.github.io/batpred/inverter-setup/#i-want-to-add-an-unsupported-inverter-to-predbat

If you have specific issues then we can try to enhance Predbat, but you should be able to make the service calls to charge/discharge/hold etc through configuring predbat in this way

Once you get a working apps.yaml then do share and we can add it to the documentation for other users

azebro, 2025-03-06:
> since none of us have that inverter so can't try it out, the best bet (assuming you have one) is to follow the Predbat instructions for adding an unsupported inverter [...]

Thanks, started doing that.
I am getting exception though: "Error: Inverter type MINE not defined" when using the template in the link. I have tried inputitng different values but all the same output. Do I need to set it to one of the supported inverters?

azebro, 2025-03-06:
I have followed the link and based myself on Huawei template. The attached apps.yaml seems to work for Sigenstore providing that the integration via Moddus is enabled as per: [Sigenergy HA](https://github.com/TypQxQ/Sigenergy-Home-Assistant-Integration). The issue still remains on the inverter_type, I have left it to "HU" as any other value gives me error as "inverter not defined".

[apps.zip](https://github.com/user-attachments/files/19108850/apps.zip)

azebro, 2025-03-16:
Hi @gcoan and @springfall2008 , here are updated config files for Sigenergy Sigenstore tested. There were some issues in the previous ones so please incorporate the latest ones here.

[apps.yaml.zip](https://github.com/user-attachments/files/19269262/apps.yaml.zip)

```
"SIG": {
        "name": "Sigenergy Sigenstor",
        "has_rest_api": False,
        "has_mqtt_api": False,
        "output_charge_control": "power",
        "charge_control_immediate": True,
        "has_charge_enable_time": False,
        "has_discharge_enable_time": False,
        "has_target_soc": False,
        "has_reserve_soc": False,
        "has_timed_pause": False,
        "charge_time_format": "S",
        "charge_time_entity_is_option": False,
        "soc_units": "%",
        "num_load_entities": 1,
        "has_ge_inverter_mode": False,
        "time_button_press": False,
        "clock_time_format": "%Y-%m-%d %H:%M:%S",
        "write_and_poll_sleep": 4,
        "has_time_window": False,
        "support_charge_freeze": False,
        "support_discharge_freeze": False,
        "has_idle_time": False,
        "can_span_midnight": True,
        "charge_discharge_with_rate": False,

```

gcoan, 2025-03-16:
will do, thanks, I'll ping you when I have it drafted for review

fbradyirl, 2025-05-08:
The yaml based package by @TypQx has since been superseded by a proper full HACS integration. 

https://github.com/TypQxQ/Sigenergy-Local-Modbus

Does the example in Predbat docs for SigenStor need to be updated?

Evaske, 2025-05-13:
> The yaml based package by @TypQx has since been superseded by a proper full HACS integration.
> 
> https://github.com/TypQxQ/Sigenergy-Local-Modbus
> 
> Does the example in Predbat docs for SigenStor need to be updated?

Did you get anywhere with working with the newer version? I wasn't sure how to get it to actually make the control changes that wasn't using the modbus service.

fbradyirl, 2025-05-13:
> Did you get anywhere with working with the newer version? I wasn't sure how to get it to actually make the control changes that wasn't using the modbus service.

Not sure I follow. 

The newer version is a proper integration but still uses Modbus to read and write to the device.

gcoan, 2025-05-13:
I'd need someone with a signestore inverter to work out what changes are needed to apps.yaml and the install instructions as I don't have one !

Evaske, 2025-05-14:
> Not sure I follow.
>
> The newer version is a proper integration but still uses Modbus to read and write to the device.

When trying to use the newer integration it didn't seem to know about `modbus.register_write` as is in the Predbat documentation for writing to the inverter. I did get it all working in the end, just had to manually add the modbus entry to the configuration.yaml and make some changes to the predbat app.yaml

azebro, 2025-05-14:
I am the author of the integration guide and was thinking of updating but:

  1.
Afaik HACS integration with Sigen is flaky - I have it on my test deployment and not convinced
  2.
To integrate with batpred you need values in W and not KW - this requires even more modifications
  3.
The modbus spec for Sigen did not change much (only added EV stuff) so tbh I do not see added value in moving to HACS

Happy to be corrected though.

TypQxQ, 2025-06-13:
Hi!
I'm the author of both integrations. The new HACS integration adds AC Charger, DC Charger and multiple Inverters.
One other major improvement is that it polls the modbus values in groups and not all values all the time, this means less trafic and faster updates.
If you find anything not working properly, please don't hesitate to put in a issue or even do a pull req 😉 

Came here to check out this addon and if it works with Sigenergy 😄

F17HEH, 2025-11-04:
@TypQxQ im testing this for you, having to change a few things but will update.  Not got my export yet so will be a few weeks.

main changes so far

`  load_today:
    - sensor.sigen_plant_consumed_power
  import_today:
    - sensor.sigen_plant_daily_grid_import_energy
  export_today:
    - sensor.sigen_plant_daily_grid_export_energy
  pv_today:
    - sensor.sigen_plant_daily_pv_energy
  battery_power:
    - sensor.sigen_plant_battery_power_w
  pv_power:
    - sensor.sigen_plant_pv_power_w
  load_power:
    - sensor.sigen_plant_consumed_power_w
  grid_power:
    - sensor.sigen_plant_grid_active_power_w
  soc_percent:
    - sensor.sigen_plant_battery_state_of_charge
  soc_max:
    - sensor.sigen_plant_battery_power
  battery_temperature:
    - sensor.sigen_inverter_battery_average_cell_temperature`

gcoan, 2025-11-04:
> Battery power is not correct so will keep looking what battery_power should be set to.

battery power is power flow in and out of the battery https://springfall2008.github.io/batpred/apps-yaml/#power-data

But is not essential to Predbat, its only used for the power flow graph on the predbat web UI

The error you are getting is about battery SIZE, you're getting a negative battery size returned to Predbat, and clearly that's wrong

F17HEH, 2025-11-04:
So i had to create 3 helpers in HA to remove some errors.

```input_number:
  charge_rate:
    name: Charge Rate
    min: 0
    max: 8000
    step: 100
    unit_of_measurement: "W"
    icon: mdi:battery-charging
    
  discharge_rate:
    name: Discharge Rate
    min: 0
    max: 8000
    step: 100
    unit_of_measurement: "W"
    icon: mdi:battery-arrow-down
```
and

```input_select:
  predbat_requested_mode:
    name: Predbat Requested Mode
    options:
      - "Demand"
      - "Charging"
      - "Discharging"
      - "Freeze Charging"
      - "Freeze Discharging"
    icon: mdi:battery-cog
```

F17HEH, 2025-11-05:
So looks like the plan is calculating but its not auctioning as expected.  When the plan shows battery discharge the grid was feeding the house.   Not sure what is going on so any help would be great.

Its now on freeze charge and grid is feeding house, but solar is feeding the battery and not the house at all.

F17HEH, 2025-11-05:
So HA is now talking to the plant, now just looking into why none of the changes made in batpred dont action changes

F17HEH, 2025-11-05:
Right all working, will review the docs and update here the changes i had to make (reading the docs again was key - rather a lot of picnic)

DFEvans, 2025-11-13:
Working through the docs and various threads as best I can, it's quite a lot to try and understand with the change of the Sigenergy integration, changes by both users and the integration author to do the same, etc.!

The only steps I have had to do to make it appear sensible, beyond what is currently documented in BatPred's docs and templates, is:
* Change `load_today` to `sensor.sigen_plant_consumed_power`
* Enable various sigen sensors that were disabled (following whatever the logs complained about)

One surprise is that I don't see obvious issues with power units, despite extensive discussion on the topic. E.g. currently BatPred shows my use as 600W in line with the MySigEn app, without an obvious unit reporting error like 0.6W or 600 kW. Is this the result of an update on one side or the other since this discussion started? Or does the live display on BatPred using the correct units hide an underlying problem where the units are misinterpreted elsewhere?

gcoan, 2025-11-13:
thanks for the feedback @DFEvans do you know what sigen sensors you had to enable?

Predbat is a lot more tolerant of different units of measure now than it used to be, so (hopefully) the prior discussion about units isn't so relevant.  Certainly recent people configuring Sig with Predbat haven't raised any concerns over this.
But do double check the plan and that it looks sensible, that load, PV etc match what you expect

DFEvans, 2025-11-14:
Looking back through the logs, I believe these are the ones that needed enabling - they're ones that gave errors for "returned value None" and that are still in my app.yaml:
```
sensor.sigen_plant_available_max_discharging_capacity
sensor.sigen_plant_daily_consumed_energy
```

(I'm slightly uncertain about the list, as replies here and in other threads suggested other sensors which then needed enabling, but in all cases except `load_today` I eventually reverted to the sensor in the template as it seemed to work)

> But do double check the plan and that it looks sensible, that load, PV etc match what you expect

Yep, a few days of careful watching to follow. Now that it's daytime, I've spotted that `sigen_plant_pv_power` is permanently reading 0 despite the system currently generating solar, which I've enquired about on the integration: https://github.com/TypQxQ/Sigenergy-Local-Modbus/discussions/261

It also sounds like the control side may have some quirks that I won't yet have encountered - and by the sounds of it, those can be silent failures.

F17HEH, 2025-11-14:
I did have to enable number.sigen_plant_ess_backup_state_of_charge to allow the Set reserve Rate to work correctly and update an automation.  One to look out for

DFEvans, 2025-11-28:
Making a note of a few additional controls that we could probably get Predbat to make use of. The config is currently controlling max battery charge and discharge rate, and max **system** grid export rate (the corresponding import rate is unused). These other parameters are available by the integration, I suspect several are recent additions (the Modbus manual copy I have from February doesn't include several):
* Backup SOC level
* Charging SOC cutoff
* Discharging SOC cutoff
* Power Conversion System (= battery and solar?) Grid Export Limit
* Power Conversion System (= battery and solar?) Grid Import Limit

The "other limit settings that seem fishy" I mentioned above was an odd case I encountered the other day where I think my solar generation was being artificially limited. From continued investigation, my current belief is:
* The solar generation was (amazingly, for winter) above the available battery charge rate at the time, due to some combination of battery SOC and temperature. Not a Predbat problem, but;
* The grid export limit was set to zero as Predbat's aim was to charge + use own solar

I need to sit down and read some more about HA automations to make full use of the new settings - particularly dealing with Predbat giving a single SOC target value in some modes only, while the system needs two correctly set values in all modes.

DFEvans, 2025-11-28:
In fact, a brief flash of inspiration for a very basic solution that gets Freeze Discharge working - use "Self Consumption" mode for Freeze Charge/Discharge, and just set the Battery SOC Charge/Discharge limits to forbid charge or discharge. Those parameters appear to always apply, avoiding the issue F17HEH encountered with other parameters only applying in some modes. The revised `predbat_requested_mode_action` action with that concept is below - I'll keep an eye on it myself for a while.

(This simple bodge doesn't use Predbat's target SOC values, which would be an improvement)

```
- id: predbat_requested_mode_action
  alias: "Predbat Requested Mode Action"
  description: "Acts as a mapper for the input_select.predbat_requested_mode to the select.sigen_plant_remote_ems_control_mode"
  mode: restart
  triggers:
    - trigger: state
      entity_id:
        - input_select.predbat_requested_mode
  conditions: []
  actions:
    - action: select.select_option
      metadata: {}
      target:
        entity_id: select.sigen_plant_remote_ems_control_mode
      data:
        option: >
          {% if is_state('input_select.predbat_requested_mode', "Demand") %}Maximum Self Consumption
          {% elif is_state('input_select.predbat_requested_mode', "Charging") %}Command Charging (PV First)
          {% elif is_state('input_select.predbat_requested_mode', "Freeze Charging") %}Maximum Self Consumption
          {% elif is_state('input_select.predbat_requested_mode', "Discharging") %}Command Discharging (PV First)
          {% elif is_state('input_select.predbat_requested_mode', "Freeze Discharging") %}Maximum Self Consumption
          {% endif %}
    - choose:
        # Set charging limit to 0 when requested mode is Freeze Charging
        # Docs:
        #  Freeze charging - The battery is charging but the current battery level (SoC) is frozen (held). Think of it
        #  as a charge to the current battery level. The grid or solar covers any house load. If there is a shortfall of
        #  Solar power to meet house load, the excess house load is met from grid import, but if there is excess Solar
        #  power above the house load, the excess solar will be used to charge the battery
        # In Sigenergy, this is effectively "self consumption" mode with discharging prohibited
        - conditions:
            - condition: state
              entity_id: input_select.predbat_requested_mode
              state: "Freeze Charging"
          sequence:
            - service: number.set_value
              data_template:
                entity_id: number.sigen_plant_ess_charge_cut_off_state_of_charge
                value: 100
            - service: number.set_value
              data_template:
                entity_id: number.sigen_plant_ess_discharge_cut_off_state_of_charge
                value: 100

        # Set charging limit to 0 when requested mode is Freeze Discharging
        # Docs:
        #  Freeze exporting (mapped to Freeze Discharging in sigenergy_sigenstor.yaml) - The battery is in demand mode,
        #  but with charging disabled. The battery or solar covers the house load. As charging is disabled, if there is
        #  excess solar generated, the current SoC level will be held and the excess solar will be exported. If there is
        #  a shortfall of generated solar power to meet the house load, the battery will discharge to meet the extra load.
        # In Sigenergy, this is effectively "self consumption" mode with charging prohibited
        - conditions:
            - condition: state
              entity_id: input_select.predbat_requested_mode
              state: "Freeze Discharging"
          sequence:
            - service: number.set_value
              data_template:
                entity_id: number.sigen_plant_ess_charge_cut_off_state_of_charge
                value: 0
            - service: number.set_value
              data_template:
                entity_id: number.sigen_plant_ess_discharge_cut_off_state_of_charge
                value: 0

        # If neither of the above conditions are met, set the limits to the input numbers
        - conditions:
            - condition: not
              conditions:
                - condition: state
                  entity_id: input_select.predbat_requested_mode
                  state: "Freeze Charging"
                - condition: state
                  entity_id: input_select.predbat_requested_mode
                  state: "Freeze Discharging"
          sequence:
            - service: number.set_value
              data_template:
                entity_id: number.sigen_plant_ess_charge_cut_off_state_of_charge
                value: 100
            - service: number.set_value
              data_template:
                entity_id: number.sigen_plant_ess_discharge_cut_off_state_of_charge
                value: 0
```

F17HEH, 2025-11-28:
ah yes @azebro is correct the max consumption cannot set those values.  This is the issues i ran into

F17HEH, 2025-11-28:
@DFEvans i find running with freeze charging turned off and the below is working for me well.  Ive also turned on the double pass as find its better with Agile.

```- id: predbat_requested_mode_action
  alias: Predbat Requested Mode Action
  description: Acts as a mapper for the input_select.predbat_requested_mode to the
    select.sigen_plant_remote_ems_control_mode
  triggers:
  - trigger: state
    entity_id:
    - input_select.predbat_requested_mode
  conditions: []
  actions:
  - action: select.select_option
    metadata: {}
    target:
      entity_id: select.sigen_plant_remote_ems_control_mode
    data:
      option: '{% if is_state(''input_select.predbat_requested_mode'', "Demand") %}Maximum
        Self Consumption {% elif is_state(''input_select.predbat_requested_mode'',
        "Charging") %}Command Charging (PV First) {% elif is_state(''input_select.predbat_requested_mode'',
        "Freeze Charging") %}Maximum Self Consumption {% elif is_state(''input_select.predbat_requested_mode'',
        "Discharging") %}Command Discharging (PV First) {% elif is_state(''input_select.predbat_requested_mode'',
        "Freeze Discharging") %}Maximum Self Consumption {% endif %}

        '
  - choose:
    - conditions:
      - condition: state
        entity_id: input_select.predbat_requested_mode
        state: Freeze Charging
      sequence:
      - data_template:
          entity_id: number.sigen_plant_ess_max_discharging_limit
          value: 0
        action: number.set_value
      - data_template:
          entity_id: number.sigen_plant_grid_export_limitation
          value: 100
        action: number.set_value
    - conditions:
      - condition: state
        entity_id: input_select.predbat_requested_mode
        state: Freeze Discharging
      sequence:
      - data_template:
          entity_id: number.sigen_plant_ess_max_charging_limit
          value: 0
        action: number.set_value
    - conditions:
      - condition: not
        conditions:
        - condition: state
          entity_id: input_select.predbat_requested_mode
          state: Freeze Charging
        - condition: state
          entity_id: input_select.predbat_requested_mode
          state: Freeze Discharging
      sequence:
      - data_template:
          entity_id: number.sigen_plant_ess_max_charging_limit
          value: '{{ [(states(''input_number.charge_rate'') | float / 1000) | round(2),
            states(''sensor.sigen_inverter_ess_rated_charge_power'') | float] | min}}'
        action: number.set_value
      - data_template:
          entity_id: number.sigen_plant_ess_max_discharging_limit
          value: '{{ [(states(''input_number.discharge_rate'') | float / 1000) | round(2),
            states(''sensor.sigen_inverter_ess_rated_discharge_power'') | float] |
            min}}'
        action: number.set_value
      - data_template:
          entity_id: number.sigen_plant_grid_export_limitation
          value: 100
        action: number.set_value
  mode: restart
- id: automation_sigen_ess_max_charging_limit_input_number_action
  alias: Predbat max charging limit action
  description: Mapper from input_number.charge_rate to number.sigen_plant_ess_max_charging_limit
  triggers:
  - trigger: state
    entity_id: input_number.charge_rate
  action:
  - action: number.set_value
    target:
      entity_id: number.sigen_plant_ess_max_charging_limit
    data:
      value: '{{ (states(''input_number.charge_rate'')| float / 1000) | round(2) }}'
  mode: single
- id: automation_sigen_ess_max_discharging_limit_input_number_action
  alias: Predbat max discharging limit action
  description: Mapper from input_number.discharge_rate to number.sigen_plant_ess_max_discharging_limit
  triggers:
  - trigger: state
    entity_id: input_number.discharge_rate
  action:
  - action: number.set_value
    target:
      entity_id: number.sigen_plant_ess_max_discharging_limit
    data:
      value: '{{ (states(''input_number.discharge_rate'')| float / 1000) | round(2)
        | int }}'
  mode: single```

DFEvans, 2025-11-28:
> i only have the same docs, with those setting for me the battery charges from the grid, this is the problem ive been having.

> Max discharge will stop the battery being used, but there is no way to set so the grid feeds the house, unless solar is meeting demand then additional solar tops battery up. Just cant get that behavior.

Just to check, are we all talking about the same settings? The ones I have seen discussed before (with the issues you state) are controls for the charge and discharge **power** in kW:
* `number.sigen_plant_ess_max_charging_limit`
* `number.sigen_plant_ess_max_discharging_limit`

whereas the ones I'm trying are controls for the charge and discharge **SOC target** in percent:
* `number.sigen_plant_ess_charge_cut_off_state_of_charge`
* `number.sigen_plant_ess_discharge_cut_off_state_of_charge`

(also, for @gcoan, those are both items that require enabling)

DFEvans, 2025-11-28:
Hmm, again, not the behaviour I'm seeing right now in demand mode - now the sun has set, with Maximum Self Consumption set and Charge Cut-Off SOC at 100%, the battery is powering the load and no grid is being pulled:

[image]

I did change the mode mapping for Freeze Charge and Freeze Discharge to Maximum Self Consumption in both cases (so the SigEnergy system isn't trying to import/export), rather than the Command Charge/Command Discharge. With the old mapping, I'd definitely expect the behaviour you mention!

F17HEH, 2025-11-28:
> Hmm, again, not the behaviour I'm seeing right now in demand mode - now the sun has set, with Maximum Self Consumption set and Charge Cut-Off SOC at 100%, the battery is powering the load and no grid is being pulled:
> <img alt="Image" width="735" height="740" src="https://private-user-images.githubusercontent.com/28623784/520236122-66a81cfc-df32-4fdf-a541-91bac7be5f21.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjQzNTA4NzAsIm5iZiI6MTc2NDM1MDU3MCwicGF0aCI6Ii8yODYyMzc4NC81MjAyMzYxMjItNjZhODFjZmMtZGYzMi00ZmRmLWE1NDEtOTFiYWM3YmU1ZjIxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTExMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUxMTI4VDE3MjI1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc0ZWI4NGJkYmIyYjAxMDU3NTgzNDU2MDM3ZjAxYjI4MzI1ZmQyYjM1ODMxMjBiMjVkM2MxY2M3ZjEyMDIyZGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.pkkCzChpHEakOCEzAota30AGAN4exKVNI4fCP6k1n5A">
> 
> I did change the mode mapping for Freeze Charge and Freeze Discharge to Maximum Self Consumption in both cases (so the SigEnergy system isn't trying to import/export), rather than the Command Charge/Command Discharge. With the old mapping, I'd definitely expect the behaviour you mention!

what is your battery SOC tho?  as mine is not 100

F17HEH, 2025-11-29:
> I believe for the Freeze charging and Freeze discharging modes to work correctly, you need to set the discharge cut off state of charge to be the current battery SOC value for Freeze Charging, and set the charge cut off state of charge to be the current battery SOC value for Freeze Discharging:
> 
>         #  Freeze charging - The battery is charging but the current battery level (SoC) is frozen (held). Think of it
>         #  as a charge to the current battery level. The grid or solar covers any house load. If there is a shortfall of
>         #  Solar power to meet house load, the excess house load is met from grid import, but if there is excess Solar
>         #  power above the house load, the excess solar will be used to charge the battery
>         # In Sigenergy, this is effectively "self consumption" mode with discharging prohibited
>         - conditions:
>             - condition: state
>               entity_id: input_select.predbat_requested_mode
>               state: "Freeze Charging"
>           sequence:
>             - service: number.set_value
>               data_template:
>                 entity_id: number.sigen_plant_ess_charge_cut_off_state_of_charge
>                 value: 100
>             - service: number.set_value
>               data:
>                 value: "{{ states('sensor.sigen_plant_battery_state_of_charge') }}"
>               target:
>                 entity_id: number.sigen_plant_ess_discharge_cut_off_state_of_charge
> 
>         # Set charging limit to 0 when requested mode is Freeze Discharging
>         # Docs:
>         #  Freeze exporting (mapped to Freeze Discharging in sigenergy_sigenstor.yaml) - The battery is in demand mode,
>         #  but with charging disabled. The battery or solar covers the house load. As charging is disabled, if there is
>         #  excess solar generated, the current SoC level will be held and the excess solar will be exported. If there is
>         #  a shortfall of generated solar power to meet the house load, the battery will discharge to meet the extra load.
>         # In Sigenergy, this is effectively "self consumption" mode with charging prohibited
>         - conditions:
>             - condition: state
>               entity_id: input_select.predbat_requested_mode
>               state: "Freeze Discharging"
>           sequence:
>             - service: number.set_value
>               data:
>                 value: "{{ states('sensor.sigen_plant_battery_state_of_charge') }}"
>               target:
>                 entity_id: number.sigen_plant_ess_charge_cut_off_state_of_charge
>             - service: number.set_value
>               data_template:
>                 entity_id: number.sigen_plant_ess_discharge_cut_off_state_of_charge
>                 value: 0
> 

Can you confirm the behaviour of excess solar with these settings?

jamiekets, 2025-11-29:
> Can you confirm the behaviour of excess solar with these settings?

I have not been able to test with excess solar yet as I am unlikely to have any today but should be able to confirm tomorrow. With Freeze Charge set last night, the battery SOC was maintained and all house load was met by the grid. With excess solar in Freeze Charging mode, the battery should continue to charge as the cut off state of charge is set to 100%. In Freeze Discharge mode, excess solar should be exported once the house load is met

gcoan, 2025-11-29:
thanks for all the collaborative development and problem solving guys, I'll await your test results 👍

jamiekets, 2025-11-29:
@DFEvans - I guess it depends how you interpret Freeze Charging. It read it as the current SOC will be maintained at the point when the mode was triggered. Yes if you have excess solar your battery will charge above that SOC and if your solar later does not cover your load, the battery will then discharge back to that SOC when Freeze Charging was triggered

DFEvans, 2025-11-29:
Hmm, interesting, it is somewhat ambiguous in the docs as to whether Predbat intends to store excess energy for later, or to instead allow discharged back down to the initial level. I'd read it as the opposite of Freeze Discharge, which states explicitly that Charging is forbidden.

DFEvans, 2025-12-02:
I believe the configuration is in a good place - I've not yet found a case where it wasn't (mostly) doing as Predbat desired. The various suggestions about slightly better SOC control seem sensible improvements - I'll probably grab Toby's change above myself - but I've not much time to dig in deeper.

On the temperature, thanks for the pointer. It looks like the chart I tried to attach got eaten by gremlins, hopefully there this time. The pattern is interesting as the maximum charge rate is a function of both SOC and temperature simultaneously - for a given SOC, the charge rate might be 100% of maximum if the temperature is high enough, or it might be reduced. Similarly, a given temperature doesn't impact charge rate at low SOC, but will impact it at high SOC.

[image]

gcoan, 2025-12-02:
I had already raised the PR with the prior version of the template etc in, so these changes will get appended to this PR, so if it can be checked and any issues please let me know soon.

I have merged the automation from @DFEvans https://github.com/springfall2008/batpred/issues/2077#issuecomment-3591563384 into the documentation

And added the revised discharging lower level as per @toby200 https://github.com/springfall2008/batpred/issues/2077#issuecomment-3603124857

One thing I noticed was that the action calls in the main predbat_requested_mode_action automation were 'service: number.set_value'; per HA 2024 naming convention they should have been 'action: number.set_value' as Toby had them.  It was only this automation that was like this, the other two were OK, so I have changed them all to actions.

In the documentation the indentation for the automations and helper entities looked wrong, I have aligned things back to the left.

Latest template: https://github.com/gcoan/batpred/blob/main/templates/sigenergy_sigenstor.yaml

install docs:
https://github.com/gcoan/batpred/blob/main/docs/inverter-setup.md#sigenergy-sigenstor

Thanks

toby200, 2025-12-03:
> Sigenergy say it's OK to discharge Sigen batteries to 0%. It's one of their selling points.
> 
> Source: [Technical Note – Sigen Battery 100% Depth of Discharge](https://www.sigenergy.com/en/support/files/2212)

Oh, thanks for that and for providing the link, very interesting. 10% more capacity, happy days - my installer suggested setting it to 20% so I shall have words!

toby200, 2025-12-03:
`number.sigen_plant_ess_charge_cut_off_state_of_charge`  is the SoC it will charge up to , then it will stop charging
`number.sigen_plant_ess_discharge_cut_off_state_of_charge` is the SoC it will discharge down to, then it will stop discharging

These are what we're using above already, so those can be set to make it charge up to a value or discharge down to a value if we know what the limit is to set it to. I couldn't see those limit % values in any entities but there are a lot to look through!

Isn't soc_max the battery's max capacity in kwh rather than a target %? I see it set as
```
  soc_max:
    - sensor.sigen_plant_rated_energy_capacity
```
which seems right but isn't something that can be set, it's a read-only value.

Slightly unrelated but I think we can also set this to pull a value from Sigen rather than hardcoding to 8k as a default
From 
```
  # Set the maximum charge/discharge rate of the battery
  battery_rate_max:
    - 8000
```
to
```
  # Set the maximum charge/discharge rate of the battery
  battery_rate_max:
    - sensor.sigen_plant_ess_rated_charging_power
```

rather than hardcoding to a value it can pull from Sigen what the rated power is. One slight caveat is Sigen have a charging power and a discharging power and they aren't the same. Mine shows 4.4kw charging and 4.8kw discharging. Assuming charging is normally lower then I guess that's the safer option of the two.

gcoan, 2025-12-03:
> number.sigen_plant_ess_charge_cut_off_state_of_charge is the SoC it will charge up to , then it will stop charging
> number.sigen_plant_ess_discharge_cut_off_state_of_charge is the SoC it will [...]

great, but at the moment predbat in the apps.yaml template

https://github.com/gcoan/batpred/blob/main/templates/sigenergy_sigenstor.yaml

predbat is not configured to set these entities

Would need the lines:

```
reserve:
  -  number.sigen_plant_ess_discharge_cut_off_state_of_charge
charge_limit:
  -  number.sigen_plant_ess_charge_cut_off_state_of_charge
```

adding to the template so predbat sets these with the charge / discharge limits

and the automation changing because it is at the moment setting these to 100 and 0 (or reserve min) when discharging

> Isn't soc_max the battery's max capacity in kwh rather than a target %?

yes you are absolutely correct, its charge_limit that predbat sets

and yes we could use that entity for the charge rate max rather than hard coding it

gcoan, 2025-12-03:
if we do set the charge limit based on what predbat plans it to be to meet the plan, then this will be more efficient as it'll prevent over charging.  Maybe a minor point but if we can I'd suggest we do remove the fixed limits from the automation and let the predbat entities define the limits

example from my own plan tonight, charging to 65% and then holding at 68% thereafter as that's all predbat calculates I need to support the battery before the next charge period:
[image]

the units that an apps.yaml or entity requires should be documented in the predbat documentation (either the apps.yaml or customisation sections are the main parts to look at). Wherever possible give predbat what it wants.  There are some places where it will determine the unit of measure and adjust, and again these should be documented.

If not, let me know examples where the docs are wrong

F17HEH, 2025-12-18:
one thing to note in the template the 

inverter_limit:

This needs updating to use `sensor.sigen_plant_available_max_active_power`

F17HEH, 2025-12-18:
So ive fully updated as per the new setup, will monitor over the next few days (got a good solar day tomorrow) and update here

reedy, 2025-12-18:
There's definitely a lot of data hidden in there!

I guess the discharge can't be right as it's more than the 15kW inverter... But 11.34 isn't far off (9.2/2)*3

reedy, 2025-12-18:
There is "ESS Max Charging Limit" and "ESS Max Discharging Limit" which are config

reedy, 2025-12-18:
> I have updated the issue title as well

SigenStor ;)

F17HEH, 2025-12-18:
>It should be set to a static number, from the inverter or just hard code it, but whatever your battery max charge/discharge rate is.
predbat will use this when setting full/partial charging rates, so it needs to be a rate that can be used for both solar and grid charging (so for some people it might be their inverter limit)

OK have set this to use `sensor.sigen_plant_available_max_active_power` which is a static number of the inverter max.

gcoan, 2025-12-18:
> > I have updated the issue title as well
> 
> SigenStor ;)

Blame @azebro he raised the issue.  Should know what kit he has.

gcoan, 2025-12-19:
ignore the above commit, wrong issue number on my part

> I added the extra log line you requested above, I've also updated to the latest docker version. When setting Freeze Charge the line logs:
> 
> > Warn: inverter does not have charge_enable_time, status=Demand, current_charge_limit=0, setting soc target to 0

I wanted the extra log to confirm that it is definitely going down this path, bit strange that the status says Demand, I would have expected F-C

It wasn't just updating to the latest docker, it was updating the predbat app within it to the latest version.  You were currently about 13 versions behind and there was a lot of logfile changes a few versions ago

Be useful to see the full log, and confirm on what statii the branch is executed

DFEvans, 2025-12-25:
The automations.yaml snippet in the docs defines `value` twice for the charging limit action, which is invalid. I think the first entry is the desired one.

Also, I am seeing the charge cut off being set at 4% when Predbat is requesting freeze charge, which seems wrong - it's preventing any battery charge. The automation looks correct; is it fighting with Predbat setting something via this line in apps.yaml?
```
charge_limit:
    - number.sigen_plant_ess_charge_cut_off_state_of_charge
```
Annoyingly, that's the parameter that causes the latest Predbat versions to crash if not set, so I can't test unsetting it.

darkorb, 2026-02-04:
Just so you are aware, setting `sigen_plant_ess_charge_cut_off_state_of_charge` - which happen with `Predbat Requested Mode Action` outlined earlier in the thread makes the EVDC (and maybe the EVAC, but I don't have one of those) unhappy.

In some instances it fails to save settings (like changing from PV Surplus to V2X for example) because of that, even if you are not using battery boost - unfortunately a setting that can't be changed via OCPP just yet. It's not a huge issue, just something to note maybe. I need to check if setting it to something other than 0 makes it less cranky.

[image]
<br />
[image]

F17HEH, 2026-03-05:
This might change a few things with the modbus update
update

Will need to have a play
play
[image]

F17HEH, 2026-03-22:
worth a note that with the new code release ive had to change the battery_rate_max which did use sensor.sigen_plant_available_max_active_power to a integer as this is now being checked and would need to be a float to keep working as it was.

@springfall2008 i have noticed that inverter_limit is still a float so not sure if best to bring them all to floats?

gcoan, 2026-03-22:
inverter limit can either be a fixed number or a sensor in apps.yaml

it used to be validated to be a number, but that was changed to a float in https://github.com/springfall2008/batpred/commit/da02736dbfff755ef92ba6c321dae96034fd03c2

is not immediately clear why this change was made, but there were changes made to the fox cloud integration as part of this, including retrieving inverter_limit from fox cloud so it may be a side effect of that.
To my mind this is wrong, inverter limit is in watts https://springfall2008.github.io/batpred/apps-yaml/#inverter_limit and even if some service provides it in decimals, its spurious planning action to carry that into predbat as a float

so I suggest this is a bug, that inverter_limit should be reverted back to an int, and if fox integration returns a float then that should be rounded for predbat.  I'd suggest you raise this as a separate bug @F17HEH for Trefor to correct

DFEvans, 2026-04-06:
I've spent some time over the Easter weekend trying to refine the charge/discharge limit logic we prototyped a while back.

Two things have stood out as being a little awkward to handle. Firstly, as noted before, Predbat provides a single output Target SOC value, without specifying whether it is a charge target ("ensure SOC stays lower than X") or a discharge target ("ensure SOC stays higher than X"). I've inferred the following mapping:
* Demand - discharge target. Always allow charging from excess solar.
* Charging - charge target.
* Discharging - discharge target.
* Freeze Charging - discharge target. Always allow charging from excess solar.
* Freeze Discharging/Export - charge target. Always allow discharging to cover excess load.

Secondly, the order in which the Requested Mode and Charge Limit are set is not guaranteed, nor is the delay between them being changed, or the delay between a change and HA triggering an action based on that change. Additionally, Predbat may not modify both together. The possibilities are:
* Charge Limit changes first, and then Requested Mode. This seems to be the typical behaviour on a Predbat mode change, with about a 4 second delay for my setup.
* Requested Mode changes first, then Charge Limit.
* Only Requested Mode changes. Charge Limit is unchanged, but its _meaning_ may change because the mode has changed.
* Only Charge Limit changes. The meaning remains the same.

I've therefore updated the existing Requested Mode Action to set the Charge/Discharge SOC Limit parameters, in case the meaning has changed, and added a new Charge/Discharge SOC Limit action that updates the limit based on the current mode. In the first case, this means that the Charge/Discharge limit is first updated based on the previous requested mode, and then updated again when the new requested mode is detected.

I have defined a new `input_number` in `configuration.yaml` to store Predbat's request:
```
input_number:
  predbat_charge_limit:
    name: Battery Charge or Discharge Cutoff SOC
    initial: 0
    min: 0
    max: 100
    unit_of_measurement: "%"
```

In `apps.yaml`, Predbat is now pointed at that:
```
  charge_limit:
    - input_number.predbat_charge_limit
```

To test the new mode, one will need to modify `config.py` to set `"has_target_soc": True` and `"target_soc_used_for_discharge": True"` for `"SIG"` systems: https://github.com/springfall2008/batpred/blob/105e48400767d7e38bec12504703763371fae9c7/apps/predbat/config.py#L1818-L1845

[ am I right to think that those parameters cannot be modified by config files? ]

My requested mode action is now:
```
- id: predbat_requested_mode_action
  alias: "Predbat Requested Mode Action"
  description: "Acts as a mapper for the input_select.predbat_requested_mode to the select.sigen_plant_remote_ems_control_mode"
  mode: restart
  triggers:
    - trigger: state
      entity_id:
        - input_select.predbat_requested_mode
  conditions: []
  actions:
    - action: select.select_option
      metadata: {}
      target:
        entity_id: select.sigen_plant_remote_ems_control_mode
      data:
        option: >
          {% if is_state('input_select.predbat_requested_mode', "Demand") %}Maximum Self Consumption
          {% elif is_state('input_select.predbat_requested_mode', "Charging") %}Command Charging (PV First)
          {% elif is_state('input_select.predbat_requested_mode', "Freeze Charging") %}Maximum Self Consumption
          {% elif is_state('input_select.predbat_requested_mode', "Discharging") %}Command Discharging (PV First)
          {% elif is_state('input_select.predbat_requested_mode', "Freeze Discharging") %}Maximum Self Consumption
          {% endif %}
    
    - choose:
        # Freeze Charging
        # Docs:
        #  Freeze charging - The battery is charging but the current battery level (SoC) is frozen (held). Think of it
        #  as a charge to the current battery level. The grid or solar covers any house load. If there is a shortfall of
        #  Solar power to meet house load, the excess house load is met from grid import, but if there is excess Solar
        #  power above the house load, the excess solar will be used to charge the battery
[... 124 lines cut ...]
                entity_id: number.sigen_plant_ess_charge_cut_off_state_of_charge
              data:
                value: "{{ (states('input_number.predbat_charge_limit') | float) }}"
        # Demand, Discharging, or Freeze Charging - set Discharge Limit
        - conditions:
            or:
            - condition: state
              entity_id: input_select.predbat_requested_mode
              state: "Demand"
            - condition: state
              entity_id: input_select.predbat_requested_mode
              state: "Discharging"
            - condition: state
              entity_id: input_select.predbat_requested_mode
              state: "Freeze Charging"
          sequence:
            - action: number.set_value
              target:
                entity_id: number.sigen_plant_ess_discharge_cut_off_state_of_charge
              data:
                value: "{{ (states('input_number.predbat_charge_limit') | float) }}"
  mode: single
```

Finally, the SigEnergy integration does expose a reserve SOC control, `number.sigen_plant_ess_backup_state_of_charge`. However, I found that this parameter was ignored in at least one mode (discharging, IIRC, definitely a case where it ought to be respected!), and Predbat reliably sets its target SOC value accounting for any reserve the user has requested. As a result, I've not made use of it.

chalfontchubby, 2026-09-02:
I've got a sigenergy and iog running ok with a hypervolt charger. 

I'd suggest this ticket is kinda old and has meandered a bit. Could you raise a specific ticket for car automations and potentially freeze etc and I'll use it to kick myself to write it all up.

tdbegley28, 2026-09-02:
> I've got a sigenergy and iog running ok with a hypervolt charger.
> 
> I'd suggest this ticket is kinda old and has meandered a bit. Could you raise a specific ticket for car automations and potentially freeze etc and I'll use it to kick myself to write it all up.

Thank you. Raised here: https://github.com/springfall2008/batpred/issues/4911

chalfontchubby, 2026-09-02:
Thx. I’m afraid it will be a few days - I’m away for a while.

chalfontchubby, 2026-10-19:
_Posted by Claude on behalf of @chalfontchubby._

Closing as done: SigEnergy SigenStor now has its own section in the [inverter setup docs](https://springfall2008.github.io/batpred/inverter-setup/) with a `sigenergy_sigenstor.yaml` template, plus a built-in SigEnergy Cloud integration. The car-automation discussion here moved to #4911.

closed 2026-10-19 by chalfontchubby