springfall2008/batpred #2079: Predbat crash. Restart fail.

jrwm47, 2025-03-06:
**Describe the bug**
predbat has crashed - twice at random times in the past 2 weeks - after running  continuously  for several months.
set to automatically restart, but didn't

[predbat log up to the crash.pdf](https://github.com/user-attachments/files/19116265/predbat.log.up.to.the.crash.pdf)

[apps.yaml.pdf](https://github.com/user-attachments/files/19116267/apps.yaml.pdf)

[predbat plan at crash.pdf](https://github.com/user-attachments/files/19116269/predbat.plan.at.crash.pdf)

[predbat config.pdf](https://github.com/user-attachments/files/19116274/predbat.config.pdf)

[predbat addon setup.pdf](https://github.com/user-attachments/files/19116275/predbat.addon.setup.pdf)

restarted manually and appears to be planning/controlling normally, but when will it crash again?

**Expected behaviour**
not to crash

**Predbat version
8.15.1

**Environment details**
GE AIO without gateway
Standard HAOS installer

**Screenshots**
attached

**Log file**
attached

**Predbat debug yaml file**
not done yet but will do if it helps

gcoan, 2025-03-07:
In the logfile its a websocket connection failure

I see in apps.yaml that you have added a ha_key which is what we've been advising people to do that have this issue, but the ha_key line is commented out.  Is there a reason for it being commented out?

Thanks

jrwm47, 2025-03-07:
Thanks Geoff

I must have put it in and then commented it out a while ago but I've forgotten when and why. (note to self: date code mods)
On re-reading the docs I guess I did so because I was originally using AppDaemon but no longer am.

Thinking through things last night, I suspect the websocket failure co-incided with me running an apt update/upgrade, with a sizeable download, on another machine connected to the same wifi mesh a/p. It's a bit surprising as tests always show plenty of WAN/LAN bandwidth.

Anyway, I've uncommented the HA IP address and token and predbat is simmering nicely. So fingers crossed.

Cheers, Julian

closed 2025-03-16 by springfall2008