springfall2008/batpred #2090: Account for baseline calculations for Saving Sessions

al090187, 2025-03-09:
Octopus Saving Sessions use your baseline usage over the previous 10 equivalent slots to work out your energy "saved" over the session. However, Predbat just adds the extra value to the slot for the full export amount.

E.g. Over the past 10 days I have exported at the full 6kW I am allowed by my DNO, but only on 5 days. Therefore, my baseline "import" is negative 3kWh. If I then export at the full 6kW again during the session, I will only receive the bonus for the extra 3kWh over my baseline. Predbat adds the bonus for the full 6kWh, so predictions are not correct, and neither is the total cost value.

The Octopus Plugin does have a baseline entity, but this is no good for prediction, because it only shows a single value for the first half hour slot, then updates for the second half hour slot. So, you would need to calculate the baseline figure manually from the historic data in HA, or from the API data. Maybe contact this developer as they already have the calculations worked out from the API.

https://savingsessions.streamlit.app/

gcoan, 2025-03-09:
Agreed, at the moment Predbat doesn't take account of the baseline import/export figure in working out the projected savings.

Is it worthwhile making this change?

In my view, calculating a more accurate figure would be 'better', but since the DFS savings figures have been so low (12/13p), its a bit of a moot improvement as participating in DFS (this year) brings in little real financial benefit.  I've personally not exported anything in the last few sessions as I'd result in emptying the batteries earlier and then have to import later on at a higher rate.  Your example of a 6kW export vs a 3kW baseline is DFS difference of 78p vs 39p.

We are also coming to the end of saving sessions for this year.  There might be one or two more, but probably not many more.  And whether they run next year, we will have to see.

Also a practical aspect of doing a baseline calculation using HA data, you'd need to extend the days history retained. HA default is 10 days history but DFS is calculated over the last 10 weekdays / 4 weekend days.  I purge a lot of my entity history with shorter durations than this to shrink the HA database size.
Ideally would use Long Term Statistics in HA but since this is aggregated to hourly figures they aren't suitable for calculating the half-hourly baseline.
Could get the data from Octopus if Predbat knew your Octopus API key, but adds loads more complexity.

Nilogax, 2025-03-12:
I've built something to do this (similar method to the Solcast adaptive damping @gcoan). I keep track of export during the peak hours over the previous 10 non saving session days (or 4 weekend days) then work out an adjusted "octopoints per penny" figure for each half hour slot. It's still a manual operation to work out an appropriate single rate to use for the slot, then amend apps.yaml to change the "octopoints per penny" value.

I built this last year when rewards were better and I was on Flux so my export coincided with the typical saving session times. I'm now on IOG so don't export much during the saving session times and with the smaller rewards I'm not sure the extra (attempted) accuracy makes that much difference.

headbouyJB, 2025-03-14:
Hey @gcoan the good news is that Saving Sessions (or rather the underlying DFS) is year-round now! Of course we would expect the grid conditions to make a 'Saving Session' necessary are more likely in winter, but I'll be opting in for each of them whoever they pop up. 
I think it's been brilliant this winter to have PredBat up and running to decide if, and how much, I export for it.
[New look Demand Flexibility Service to go live next week](https://www.neso.energy/news/new-look-demand-flexibility-service-go-live-next-week)

TCWORLD, 2026-08-27:
Another saving session today. Predbat just auto-joined it, and immediately the two baseline sensors make the data required available:

* Import (`sensor.octopus_energy_electricity_xxx_xxx_octoplus_power_down_baseline`):
    ```
    start: "2026-08-27T17:00:00+00:00"
    end: "2026-08-27T17:30:00+00:00"
    total_baseline: 0.0014444444444444444
    baselines:
      - start: "2026-08-27T17:00:00+00:00"
        end: "2026-08-27T17:30:00+00:00"
        baseline: 0
        is_incomplete_calculation: true
      - start: "2026-08-27T17:30:00+00:00"
        end: "2026-08-27T18:00:00+00:00"
        baseline: 0.0014444444444444444
        is_incomplete_calculation: true
    ```
* Export (`sensor.octopus_energy_electricity_xxx_xxx_export_octoplus_power_down_baseline`):
    ```
    start: "2026-08-27T17:00:00+00:00"
    end: "2026-08-27T17:30:00+00:00"
    total_baseline: 1.727666666666667
    baselines:
      - start: "2026-08-27T17:00:00+00:00"
        end: "2026-08-27T17:30:00+00:00"
        baseline: 1.143888888888889
        is_incomplete_calculation: true
      - start: "2026-08-27T17:30:00+00:00"
        end: "2026-08-27T18:00:00+00:00"
        baseline: 0.5837777777777778
        is_incomplete_calculation: true
    ```

(Note the incomplete marker is because Mondays data is unavailable due to an octopus API issue, so the baseline is not exactly correct, though is still reasonably accurate).

gcoan, 2026-08-27:
Thanks for the investigation, should be possible to do this 

Would need to be done for both octopus integration and predbat octopus direct

springfall2008, 2026-08-27:
🤖 *This is an automated first-pass triage. A maintainer will review before any action is taken.*

**Classification:** `enhancement` (already applied by a human — retained). No priority set, as this is a feature request. Added `Octopus`.

## Confirmed against current main

Checked out `main` at `v8.53.3-15-gc228a252`. The report is accurate — there is no baseline term anywhere in the saving-session path:

- `Octopus.load_saving_slot()` — `apps/predbat/octopus.py:2786` — does `rate_dict[minute] += rate` for **every minute** of the session window, on both the import and export rate tables (called from `apps/predbat/fetch.py:1032` and `fetch.py:1055`).
- The slot dict built at `apps/predbat/octopus.py:3422-3428` carries only `start`/`end`/`rate`/`state`; there is nowhere to put a baseline even if one were fetched.
- `grep -n baseline apps/predbat/octopus.py` returns nothing.

I ran the existing test (`tools/triage_test.sh saving_session`) — **PASSED**. Worth noting the flat-rate behaviour is *asserted*, not just incidental: `apps/predbat/tests/test_saving_session.py:126` pins the full 44.8 p/kWh across the whole window, so any baseline work needs that expectation updated too.

## A modelling distinction that might affect how this is scoped

The baseline splits into two effects that behave quite differently, and it may be worth deciding which one is actually being fixed:

1. **The constant part.** Reward ≈ `(baseline − delivered) × rate`. Predbat models `−delivered × rate`. The difference is `baseline × rate`, which is the *same constant for every candidate plan*, so it does not change which plan the optimiser picks — it only shifts the predicted cost and the reported saving. That is precisely what @al090187 ("neither is the total cost value") and @TCWORLD ("reports savings as being all export") are describing.
2. **The clipping part.** Once you are already delivering more than baseline there is no further reward to gain or lose, but Predbat keeps charging +rate/kWh on import and crediting +rate/kWh on export regardless. A per-minute additive rate cannot represent that kink, so this one *does* distort the plan — over-penalising import and over-crediting export past the baseline.

I have not confirmed the exact clipping semantics from Octopus (per half-hour slot vs netted across the session, and whether over-baseline half-hours net off under-baseline ones) — that would need checking before (2) is implemented.

If the goal is just "the numbers Predbat reports should match what Octopus pays", (1) alone is a much smaller change: subtract `baseline × rate` as a session-level constant in the cost/savings reporting, leaving `load_saving_slot()` and the planner untouched.

## On data availability

@TCWORLD's finding does look like it resolves @gcoan's 2025 objection about HA history retention — Predbat would not need to derive the baseline from history at all, just read `baselines` / `total_baseline` off the two `..._octoplus_power_down_baseline` sensors. That is the same attribute-read pattern the code already uses for `joined_events` / `available_events` at `apps/predbat/octopus.py:3300`, so the HA-integration side is a modest change. The `is_incomplete_calculation: true` flag in the sample would want handling (or at least logging), since it marks a baseline Octopus itself considers provisional.

The Octopus Direct path is the harder half, as @gcoan notes. The `octoplus_saving_session_query` at `apps/predbat/octopus.py:308` requests only `id`, `code`, `rewardPerKwhInOctoPoints`, `startAt`, `endAt` and the `joinedEvents` block — no baseline field. Bottle Cap Dave's integration derives the figure from its own accumulated consumption data rather than reading it from an API, so Predbat's Direct component would need either an equivalent Kraken query (if one exists — I could not confirm one) or its own half-hourly accumulation.

No duplicates found; searched open and closed issues for baseline/saving-session matches.

headbouyJB, 2026-08-27:
> 2\. **The clipping part.** Once you are already delivering more than baseline there is no further reward to gain or lose, but Predbat keeps charging +rate/kWh on import and crediting +rate/kWh on export regardless.

Unless mistaken, I believe the opposite is true- in a 'usually exporting' slot, the additional reward is only obtained once the baseline export is exceeded. For example on Octopus Agile a 'Power Down' is to our detriment in the 16:00-19:00 period, as we would usually be exporting. We need to export over and above our usual to achieve 'Power Down' rewards.

This means the value over and above the normal export: Reward ≈ (delivered - baseline) × rate 

Result for the overall plan and optimisation is that Power Up and Power Down slots become less valuable.

WhatIsThisAllAboutSilly, 2026-08-27:
I'd agree with  Reward ≈ (delivered - baseline) × rate

If the baseline for a slot is 2kWh, predbat thinks we can deliver 2.3kWh, and the octopoints rate is 10p/kWh, then the export rate for that slot is actually:

2.3-2=0.3kWh

0.3*10=3 pence

3 pence over 2.3kWh delivered is +1.3p/kWh.

On some days, I don't think predbat would even bother as I have a 4p/kWh battery cost set.

gcoan, 2026-08-28:
for those that regularly export in the peakrate when saving sessions occur, having a more accurate estimate of the actual benefit would be useful in order for predbat to decide whether to export or not.

BTW there is now a saving session threshold limit you can configure in apps.yaml so that 'low value' sessions are not joined.

WhatIsThisAllAboutSilly, 2026-08-28:
> for those that regularly export in the peakrate when saving sessions occur, having a more accurate estimate of the actual benefit would be useful in order for predbat to decide whether to export or [...]

Only way I think you can do that is calculating the effective additional rate as I did above rather than just blindly adding p/kWh.

gcoan, 2026-08-28:
> Only way I think you can do that is calculating the effective additional rate as I did above rather than just blindly adding p/kWh.

agreed, its not a straight forward rate addition which is how predbat treats it today

it would calculating the projected export, subtracting house load, subtracting base load,  to calculate a projected export, then dividing that backwards to work out the rate adjustment

and then from that rate adjustment it goes through predbat's normal calculation engine to decide whether to export or hold etc