springfall2008/batpred #2091: Planned export times seem affected by use of the Automation API

mpartington, 2025-03-10:
**Describe the bug**
I have been experimenting with the Automation API to reduce the battery discharge rate for planned export (just prior to the IOG cheap rate slot)

I've noticed that setting inverter_limit_discharge(0)=2000 changes the timings of forced export on the plan.  Instead of as late as possible, it moves it much earlier. If i cancel the override, then it moves back to the expected 'late as possible' slots

**Expected behaviour**
Export to be as late as possible on a 2 rate tariff with fixed export

**Predbat version**

v8.16.3

**Environment details**

- GivEnergy AC3 + 2x5.2 batteries
- Standard HAOS installer 

**Screenshots**

[image]

[image]

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

[apps.yaml.txt](https://github.com/user-attachments/files/19161370/apps.yaml.txt)

[predbat.log](https://github.com/user-attachments/files/19161373/predbat.log)

**Predbat debug yaml file**

[predbat_debug.yaml.txt](https://github.com/user-attachments/files/19161348/predbat_debug.yaml.txt)
[predbat_debug (1).yaml.txt](https://github.com/user-attachments/files/19161349/predbat_debug.1.yaml.txt)

mpartington, 2025-03-10:
As an additional observation (as it's just happened). Every day the freeze discharges disappear from my plan, but it maintains the forced exports, which doesn't make sense. Debug attached

[predbat_debug.yaml (2).txt](https://github.com/user-attachments/files/19161430/predbat_debug.yaml.2.txt)

springfall2008, 2025-03-11:
I think this is as expected, if you can only export at a lower rate then you need longer so must start earlier.

Why are you doing this ?

mpartington, 2025-03-11:
I understand it needing to take longer, but why is export set 3 til 4 pm in the plan, then a long pause until it commences again at 11pm? Wouldn't we expect it to be ordered as late as possible, so starting at 10pm for a longer block?

Reason I was doing it was to change discharge speed, as it was causing a spurious soc adjustment on 3020 firmware around 40%. I wanted to see if changing the power prevented this - spoiler, it didn't!

However it may still be useful for a kinder slower discharge

closed 2026-08-21 by mpartington