springfall2008/batpred #2093: Charging schedule created notification every time

sparkyhughes, 2025-03-11:
**Describe the bug**
Not a bug per se but noticed a big increase in notifications on the latest update. Every time batpred is run I get a “Charging Schedule Created at <time run>”
If this is expected then I’ll probably remove my device from the notify section in apps.yaml file

**Expected behaviour**
didnt see this many notifications before 

**Predbat version**

8.16.3

gcoan, 2025-03-11:
I'm not seeing any different behaviour in notifications on my 8.16.3

Looking back I have a Demand notification at 16:00, Hold Charging at 13:00 and Demand at 07:00 - these all correspond to the start/end of Cosy cheap/peak periods.  There was a couple of Charging/Hold Charging overnight but nothing unusual.

Check your Predbat notification settings and if it keeps happening please supply a logfile and debug file

sparkyhughes, 2025-03-12:
[predbat.log](https://github.com/user-attachments/files/19207190/predbat.log)

[predbat_debug.yaml.txt](https://github.com/user-attachments/files/19207203/predbat_debug.yaml.txt)

springfall2008, 2025-03-16:
What is the exact notification you get?

Do you have an errors in the logfile?

sparkyhughes, 2025-03-17:
[image]

This is what I get every time it runs

gcoan, 2025-03-18:
your problem is that you have switch.predbat_set_inverter_notify turned on

https://springfall2008.github.io/batpred/customisation/#inverter-control-options

sparkyhughes, 2025-03-19:
Many thanks.  That switch has done the trick.  Only getting the usual notifications now.