springfall2008/batpred #2094: Inverter went from scheduled discharge to charge when plan indicated NoChrg

sblee84, 2025-03-11:
**Describe the bug**
I'm on Octopus Flux, so my plan had the inverter doing a full battery discharge to grid up until 7pm, then it should have switched to NoChrg. Instead, it went to a full battery charge from grid. A restart of predbat stopped charging.

**Expected behaviour**
I expected the inverter to be put back in Eco mode for NoChrg, rather than a full charge.

**Predbat version**
V8.16.3

**Environment details**
GivEnergy 3.6kW Gen1 hybrid, 9.5kWh battery
Predbat installed as a HA addin

**Log file**
Attached - predbat.log is from before I restarted, predbat-1.log is from after the restart

[predbat.log](https://github.com/user-attachments/files/19196798/predbat.log)

[predbat-1.log](https://github.com/user-attachments/files/19196807/predbat-1.log)

springfall2008, 2025-03-15:
No Charge isn't a real state its a charge where the battery level isn't increasing, so scheduled charge is expected.

sblee84, 2025-03-15:
The inverter was charging the battery at 2.6kW when it was in this state, and a restart of Predbat stopped it. If NoChrg is supposed to be charging with no increase in SoC, shouldn't it have set the charge rate lower than 100%?

closed 2026-08-21 by sblee84