springfall2008/batpred #2095: Support for Kostal plenticore inverter

mbuhansen, 2025-03-12:
**Is your feature request related to a problem? Please describe.**
I m try to use Predbat to control my Kostal Plenticore inverter, i have made a APPS. yaml and testing now.

I can get kostal to charge and discharge by home assistat service call. i have also made a freeze charge button that work, inverter can charge battery but not use from battery.
When Predbat say hold in state, does it stop charge and discharge from battery completely..?? Can i call this by service api..??

The inverter is control by modbustcp, from home assistant, so i have made button i HA that start inverter to charge, dicharge or hold/lock battry use.
when inverter dont get any modbus message the battery control went back to internal use, and can both charge and discharge battery.

load_power sensor does is this power from grid or hous load both from griad and inverter..?? I have now set is to the last it measure the home power inverter calculate.

so does this apps yaml look correct..??

https://pastebin.com/unMeY171

mbuhansen, 2025-03-12:
Here the plan change from freeze charge to demand, but predbat dont call charge stop service..?? how can i be shure it turn off my no discharge button i use as freze charge function..??

2025-03-12 05:35:06.813198: Import threshold used for optimisation was 230.9p
2025-03-12 05:35:07.028433: Low import rate windows [0: 03-12 05:30:00 - 03-12 06:00:00 @ 224.0, 1: 03-12 06:00:00 - 03-12 06:30:00 @ 283.0, 2: 03-12 06:30:00 - 03-12 07:00:00 @ 283.0, 3: 03-12 07:00:00 - 03-12 07:30:00 @ 299.0, 4: 03-12 07:30:00 - 03-12 08:00:00 @ 299.0, 5: 03-12 08:00:00 - 03-12 08:30:00 @ 287.0, 6: 03-12 08:30:00 - 03-12 09:00:00 @ 287.0, 7: 03-12 09:00:00 - 03-12 09:30:00 @ 265.0, 8: 03-12 09:30:00 - 03-12 10:00:00 @ 265.0, 9: 03-12 10:00:00 - 03-12 10:30:00 @ 244.0, 10: 03-12 10:30:00 - 03-12 11:00:00 @ 244.0, 11: 03-12 11:00:00 - 03-12 11:30:00 @ 242.0, 12: 03-12 11:30:00 - 03-12 12:00:00 @ 242.0, 13: 03-12 12:00:00 - 03-12 12:30:00 @ 240.0, 14: 03-12 12:30:00 - 03-12 13:00:00 @ 240.0, 15: 03-12 13:00:00 - 03-12 13:30:00 @ 242.0, 16: 03-12 13:30:00 - 03-12 14:00:00 @ 242.0, 17: 03-12 14:00:00 - 03-12 14:30:00 @ 243.0, 18: 03-12 14:30:00 - 03-12 15:00:00 @ 243.0, 19: 03-12 15:00:00 - 03-12 15:30:00 @ 258.0, 20: 03-12 15:30:00 - 03-12 16:00:00 @ 258.0, 21: 03-12 16:00:00 - 03-12 16:30:00 @ 279.0, 22: 03-12 16:30:00 - 03-12 17:00:00 @ 279.0, 23: 03-12 17:00:00 - 03-12 17:30:00 @ 380.0, 24: 03-12 17:30:00 - 03-12 18:00:00 @ 380.0, 25: 03-12 19:00:00 - 03-12 19:30:00 @ 399.0, 26: 03-12 19:30:00 - 03-12 20:00:00 @ 399.0, 27: 03-12 20:00:00 - 03-12 20:30:00 @ 370.0, 28: 03-12 20:30:00 - 03-12 21:00:00 @ 370.0, 29: 03-12 21:00:00 - 03-12 21:30:00 @ 272.0, 30: 03-12 21:30:00 - 03-12 22:00:00 @ 272.0, 31: 03-12 22:00:00 - 03-12 22:30:00 @ 261.0, 32: 03-12 22:30:00 - 03-12 23:00:00 @ 261.0, 33: 03-12 23:00:00 - 03-12 23:30:00 @ 251.0, 34: 03-12 23:30:00 - 03-13 00:00:00 @ 251.0, 35: 03-13 00:00:00 - 03-13 00:30:00 @ 228.0, 36: 03-13 00:30:00 - 03-13 01:00:00 @ 228.0, 37: 03-13 01:00:00 - 03-13 01:30:00 @ 227.0, 38: 03-13 01:30:00 - 03-13 02:00:00 @ 227.0, 39: 03-13 02:00:00 - 03-13 02:30:00 @ 227.0, 40: 03-13 02:30:00 - 03-13 03:00:00 @ 227.0, 41: 03-13 03:00:00 - 03-13 03:30:00 @ 224.0, 42: 03-13 03:30:00 - 03-13 04:00:00 @ 224.0, 43: 03-13 04:00:00 - 03-13 04:30:00 @ 218.0, 44: 03-13 04:30:00 - 03-13 05:00:00 @ 218.0, 45: 03-13 05:00:00 - 03-13 05:30:00 @ 215.0, 46: 03-13 05:30:00 - 03-13 06:00:00 @ 215.0, 47: 03-13 06:00:00 - 03-13 06:30:00 @ 245.0, 48: 03-13 06:30:00 - 03-13 07:00:00 @ 245.0, 49: 03-13 07:00:00 - 03-13 07:30:00 @ 275.0, 50: 03-13 07:30:00 - 03-13 08:00:00 @ 275.0, 51: 03-13 08:00:00 - 03-13 08:30:00 @ 305.0, 52: 03-13 08:30:00 - 03-13 09:00:00 @ 305.0, 53: 03-13 09:00:00 - 03-13 09:30:00 @ 311.0, 54: 03-13 09:30:00 - 03-13 10:00:00 @ 311.0, 55: 03-13 10:00:00 - 03-13 10:30:00 @ 293.0, 56: 03-13 10:30:00 - 03-13 11:00:00 @ 293.0, 57: 03-13 11:00:00 - 03-13 11:30:00 @ 271.0, 58: 03-13 11:30:00 - 03-13 12:00:00 @ 271.0, 59: 03-13 12:00:00 - 03-13 12:30:00 @ 259.0, 60: 03-13 12:30:00 - 03-13 13:00:00 @ 259.0, 61: 03-13 13:00:00 - 03-13 13:30:00 @ 257.0, 62: 03-13 13:30:00 - 03-13 14:00:00 @ 257.0, 63: 03-13 14:00:00 - 03-13 14:30:00 @ 257.0, 64: 03-13 14:30:00 - 03-13 15:00:00 @ 257.0, 65: 03-13 15:00:00 - 03-13 15:30:00 @ 253.0, 66: 03-13 15:30:00 - 03-13 16:00:00 @ 253.0, 67: 03-13 16:00:00 - 03-13 16:30:00 @ 251.0, 68: 03-13 16:30:00 - 03-13 17:00:00 @ 251.0, 69: 03-13 17:00:00 - 03-13 17:30:00 @ 346.0, 70: 03-13 17:30:00 - 03-13 18:00:00 @ 346.0, 71: 03-13 18:00:00 - 03-13 18:30:00 @ 358.0, 72: 03-13 18:30:00 - 03-13 19:00:00 @ 358.0, 73: 03-13 19:00:00 - 03-13 19:30:00 @ 377.0, 74: 03-13 19:30:00 - 03-13 20:00:00 @ 377.0, 75: 03-13 20:00:00 - 03-13 20:30:00 @ 374.0, 76: 03-13 20:30:00 - 03-13 21:00:00 @ 374.0, 77: 03-13 21:00:00 - 03-13 21:30:00 @ 279.0, 78: 03-13 21:30:00 - 03-13 22:00:00 @ 279.0, 79: 03-13 22:00:00 - 03-13 22:30:00 @ 263.0, 80: 03-13 22:30:00 - 03-13 23:00:00 @ 263.0, 81: 03-13 23:00:00 - 03-13 23:30:00 @ 255.0, 82: 03-13 23:30:00 - 03-14 00:00:00 @ 255.0, 83: 03-14 00:00:00 - 03-14 00:30:00 @ 225.0, 84: 03-14 00:30:00 - 03-14 01:00:00 @ 225.0, 85: 03-14 01:00:00 - 03-14 01:30:00 @ 228.0, 86: 03-14 01:30:00 - 03-14 02:00:00 @ 228.0, 87: 03-14 02:00:00 - 03-14 02:30:00 @ 226.0, 88: 03-14 02:30:00 - 03-14 03:00:00 @ 226.0, 89: 03-14 03:00:00 - 03-14 03:30:00 @ 221.0, 90: 03-14 03:30:00 - 03-14 04:00:00 @ 221.0, 91: 03-14 04:00:00 - 03-14 04:30:00 @ 219.0, 92: 03-14 04:30:00 - 03-14 05:00:00 @ 219.0, 93: 03-14 05:00:00 - 03-14 05:30:00 @ 219.0, 94: 03-14 05:30:00 - 03-14 06:00:00 @ 219.0]
2025-03-12 05:35:07.053154: Include original charge start 03-12 01:30:00, keeping this instead of new start 03-12 05:30:00
2025-03-12 05:35:07.053311: Inverter 0 Charge window will be: 2025-03-12 01:30:00+01:00 - 2025-03-12 06:00:00+01:00 - current soc 19 target 5
2025-03-12 05:35:07.053433: Inverter 0 Target SOC 5 (this inverter 5) Battery temperature 7 Select charge rate 8300w (real 8300.0w) current charge rate 8756
2025-03-12 05:35:07.053466: Inverter 0 current charge rate is 8756W and new target is 8300W
2025-03-12 05:35:09.098182: Inverter 0 write_and_poll_value: Wrote 8300 to charge_rate, successfully now 8300
2025-03-12 05:35:09.098498: Adjust idle time, charge 01:30:00-06:00:00 discharge 00:00:00-00:00:00
2025-03-12 05:35:09.099992: Adjust demand (idle) time computed is 06:00:00-23:59:00
2025-03-12 05:35:09.100142: Inverter 0 current discharge rate is 11940.0W and new target is 0W
[... cut ...]
2025-03-12 05:35:11.175469: Inverter 0 Skipped service discharge_stop_service domain discharge service_name switch.turn_off as it was previously called.
2025-03-12 05:35:11.175496: Inverter 0 Skipped service charge_freeze_service domain charge service_name switch.turn_on as it was previously called.
2025-03-12 05:35:11.175535: Inverter 0 Skipped service discharge_stop_service domain discharge service_name switch.turn_off as it was previously called.
2025-03-12 05:35:11.175553: Inverter 0 count register writes 0
2025-03-12 05:35:11.224902: Will not recompute the plan, it is 5.0 minutes old and max age is 10 minutes

springfall2008, 2025-03-16:
Your apps.yaml looks pretty good

You might want to try making the charge services set both options as my example below.

Also the discharge start appeared to be have an off not an on?

  **charge_start_service:
    - service: input_boolean.turn_on
      entity_id: "input_boolean.plenticore_g3_charge_from_grid"
    - service: switch.turn_off
      entity_id: "input_boolean.plenticore_g3_no_discharge"     
  charge_freeze_service:
    - service: input_boolean.turn_on
      entity_id: "input_boolean.plenticore_g3_no_discharge"
    - service: input_boolean.turn_off
      entity_id: "input_boolean.plenticore_g3_charge_from_grid"**
  charge_stop_service:
    - service: input_boolean.turn_off
      entity_id: "input_boolean.plenticore_g3_charge_from_grid"
    - service: switch.turn_off
      entity_id: "input_boolean.plenticore_g3_no_discharge"     
  discharge_start_service:
    **service: input_boolean.turn_on**
    entity_id: "input_boolean.plenticore_g3_discharge_to_grid"
  discharge_stop_service:
    service: input_boolean.turn_off
    entity_id: "input_boolean.plenticore_g3_discharge_to_grid"

mbuhansen, 2025-03-16:
Thank you, i will try your configuration.
Does predbat change charge/discharge rate setpoint, to different rates, or is it only full power charge/discharge or zero. I can set my charge and discharge automation to look at the charge rate predbat desire...??

with the configuration you suggest, does charging from grid also go on when charge freeze is call..??
now i have set ha automation to look at STATUS, because i have see  charge sensor is turn on when battery is on hold. ( This is with has_service_api: false in apps.yaml )

gcoan, 2025-03-16:
> Does predbat change charge/discharge rate setpoint, to different rates, or is it only full power charge/discharge or zero. I can set my charge and discharge automation to look at the charge rate predbat desire...??

predbat generally does full rate charge and discharge, but if you turn slow rate charging switch on it will charge at the lowest rate it can to achieve the plan, e.g. for long cheap rate charging overnight 

> with the configuration you suggest, does charging from grid also go on when charge freeze is call..??

no charge freeze allows charging from solar but stops the battery discharging (freezes the SoC).  Have a look at the documentation 'what does predbat do' to understand the modes

mbuhansen, 2025-03-31:
Now most of the function is running still in test.
But a question I have difine battery power i apps yaml, but i log i get this line is that correct, in apps yaml battery power always stay on zero.
2025-03-31 11:15:06.151523:     BAT: [   g>be~,    g>be~,    g>be~,    g>be~,    g<be+,    g>be+,    g>be+,    g>be+,    g>be+,    g~be+,    g<be+,    g>be+,    g>be~,    g>be~,    g<be-,    g>be+,    g>bf-,    g>bf-,    g~be-,    g~be-,    g~be-,    g~be-,    g~be-,    g~be-,    g<be-,    g~be-,    g~be-,    g~be-,    g~be-,    g~be-,    g>be-,    g~be-,    g>be-,    g~be-,    g~be-,    g~be-,    g~be-,    g~be-,    g~be-,    g~be-,    g>be-,    g>be-,    g~be+,    g>bf-,    g>be~,    g>be~,    g>be~,    g>be+,    g>be~,    g>be~,    g>be+,    g>be+,    g>be+,    g<be+,    g>be+,    g<be+,    g<be+,    g~be+,    g<be+,    g~be+,    g>be+,    g>be~,    g>be~,    g>be-,    g>be-,    g<be-,    g~be-,    g~be-,    g~be-,    g~be-,    g~be-,    g~be-,    g~be-,    g~be-,    g<be-,    g<be-,    g<be-,    g<be-,    g~be~,    g<be-,    g<be-,    g<be-,    g<be-,    g<be-,    g<be-,    g<be-,    g~be-,    g~be-,    g<be-,    g>be-,    g~be+,    g>be+,    g<be+,    g<be+,    g<be+,    g>be+]

Is the problem that battery power both is in positive and negative..???

[predbat (2).log](https://github.com/user-attachments/files/19533037/predbat.2.log)

[image]

gcoan, 2025-03-31:
battery power can be positive and negative for charge and discharge, that's what my givenergy inverters do as well

[image]

mbuhansen, 2025-03-31:
okay thanks, maybe i see the problem my battery power is in watt and your is in kw. can you post the line from you log with the battery power.

gcoan, 2025-04-01:
I have all of my entities (battery, solar, import, export) set to kW or kWh just because its easy to read like that, but predbat doesn't care, it will read the sensors in either watts or kW, as long as the unit of measurement is set correctly.

Here's a BAT line from my logfile:

```
2025-04-01 12:58:11.525598:     BAT: [   g>be-,    g>be-,    g>be-,    g<be-,    g<be-,    g~be-,    g<be-,    g<be-,    g<bf+,    g<bf+,    g<bf+,    g<bf+,    g<bf+,    g<bf+,    g~be+,    g~be+,    g>be+,    g>be+,    g>be+,    g>be+,    g>be+,    g>be+,    g>be+,    g>be+,    g>be+,    g>be+,    g>bf+,    g>bf+,    g>bf+,    g>be~,    g>be~,    g>be~,    g>be~,    g>be~,    g>be~,    g>be~,    g>be~,    g>be~,    g>be-,    g<be-,    g~be-,    g~be-,    g~be-,    g~be-,    g<bf+,    g<bf+,    g<bf+,    g<bf+]
```

but I don't think it really helps you.  The BAT line shows what is happening to the battery over time, its a bit codified, but:

g> and g< show whether power is flowing to or from the grid

be and bf show (I think) whether the battery is in Eco (Demand) mode or being forced to do something

and +, - and ~ show whether the battery is increasing, decreasing or holding level

Something like that anyway

Your best bet to understand what the plan is is to look at the predbat HTML plan - see https://springfall2008.github.io/batpred/predbat-plan-card/

trying to work it out from the logfile is hard !

mbuhansen, 2025-04-02:
okay i understand. i normal also use the predbat plan, this is just to bee sure i got the right entinites from kostal to predbat. It take some time to figure out how this great software work.

It looks like my inverter kontrol is working now. but i have another problem with the iboost function.

I have a MYPV ACthor 9s solar diverter. I have try to config iboost function, i have add sensor from consumed diverter power setup manual gasrate to 72 pence ( this is my wood pellets heat KW price).
when i turn on iboost enable ,  i see iboost colum in predplan. But predbat dont calculate a plan, it is only showing the consumed power from solar diverter.
I want so setupr use diverter on solar and when export price is below gas price.

Now predbat want to export power when my export price is below gas price, it dont look correct. What can i have done wrong in the setup.

[image]

[predbat_debug.txt](https://github.com/user-attachments/files/19563885/predbat_debug.txt)
[predbat (3).log](https://github.com/user-attachments/files/19563887/predbat.3.log)
[apps.txt](https://github.com/user-attachments/files/19563886/apps.txt)

gcoan, 2025-04-02:
> Now predbat want to export power when my export price is below gas price, it dont look correct. What can i have done wrong in the setup.

First thing I noticed is that you have not correctly configured your gas prices in apps.yaml

you have:

```
  rates_gas:
  - start: "00:00:00"
    end: "23:30:00"
    rate: 72.0
```

this isn't valid YAML and the end time should be midnight otherwise you are saying gas is free from 23:30:00 to 00:00:00

should be:

```
  rates_gas:
    - start: "00:00:00"
      end: "00:00:00"
      rate: 72.0
```

You also don't have switch.predbat_iboost_gas turned on which is needed if you want predbat to decide whether to heat using iboost or gas based on what's cheaper

Finally, is the Kostal inverter working correctly with predbat now?
If so, can I take the apps.yaml configuration to add to the documentation and is there any other guidance/template sensors, etc needed?
Can you give a link to the integration you are using to control the inverter

mbuhansen, 2025-04-03:
I have change the gas_rate endtime as you describe to 00:00:00, and this is the same result, iboost wont calculate a plan. Because i set 23:30:00 was what i read from Predbat Docu.

Rate Bands to manually configure Energy Rates
If you are not an Octopus Energy customer, or you are but your energy rates repeat simply, you can configure your rate bands in apps.yaml using rates_import/rates_export/rates_gas.

Add the following entries to apps.yaml to define the pattern of rates over 24 hours:

rates_import:
  - start: "HH:MM:SS"
    end: "HH:MM:SS"
    rate: pence
rates_export:
  - start: "HH:MM:SS"
    end: "HH:MM:SS"
    rate: pence
rates_gas:
  - start: "HH:MM:SS"
    end: "HH:MM:SS"
    rate: pence
start and end are in the time format of "HH:MM:SS" e.g. "12:30:00" and should be aligned to 30 minute slots normally,``` i.e. end with ":30:00" or ":00:00". rate is in pence e.g. 4.2```

I have forgot to turn on switch.predbat_iboost_gas again, after i have look on Trefor video about iboost. I have activate iboost on solar switch, will iboost so turn on when battery is fully charging and use solar surplus...?? this don't happend here.
One time have i see iboost yellow on the plan, was when time change from winter to summer time, there was one hour where it show up like a plan, but later same day plan was normal and have change to summer time, this vent away again.

[image]

[image]

Yes the apps.yaml is correct for kostal now, so if you want you can upload. 
The kostal integration is the official integration in home assistant. https://www.home-assistant.io/integrations/kostal_plenticore/
BUt have manually made the toggle switch predbat service call activate, and have a automation that talk to inverter by modbus tcp. I think i need to make more documentation how to setup, becaus e you need to setup all the modbus. The integration can't control inverter.

i will try to work on a package to HA on all this i have made to get it work.

but now i want to get predbat fully working with all my stuff, still have some minor change in my kostal automation to stop and start charge/discharge.

mbuhansen, 2025-04-09:
I think i now have try every combination with the iboost function, and iboost won't calculate a plan in my setup.
What can i do more to debug this problem..??? 
I can see iboost turn on in log but plan only show accumulated power from my sun diverter.

[predbat (4).log](https://github.com/user-attachments/files/19671608/predbat.4.log)

mbuhansen, 2025-07-04:
I have now made a description on my setup to the kostal Plenticore inverter.

I have attached 2 files, on with setup i HA, and the apps yaml file i use.
I hope you can use it on Predbat documentation to other kostal users.

[Kostal setup HA.txt](https://github.com/user-attachments/files/21049352/Kostal.setup.HA.txt)

[Apps.txt](https://github.com/user-attachments/files/21049354/Apps.txt)