springfall2008/batpred #2100: Predbat charging (from grid) when plenty of solar is expected

freddanastrom, 2025-03-14:
**Describe the bug**
Predbats goes in to charging mode on days when PV energy is more than enough to fully charge the battery.
If clouds arrive or PV is not enough to cover the max charging allowed to the battery it will draw power from the grid.

**Expected behaviour**
Expecting that predbat stays in Demand mode during days with enough sunshine.

**Predbat version**

8.16.3

**Environment details**

- Solax inverter/battery
- Standard HAOS installer
- Swedish Nordpool (SE3) prices

**Screenshots**
Plan:
[image]

predbat.status switching between demand/charge/hold when sun is shining at full blast:
[image]

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

**Predbat debug yaml file**

[predbat_debug_12_00_00.yaml.txt](https://github.com/user-attachments/files/19246416/predbat_debug_12_00_00.yaml.txt)

freddanastrom, 2025-03-15:
Adding some info from today's plan that shows the same behavior with charging at 12-13.
I also reverted metric_self_sufficiency back to 0 in this one, which was set to 20 in my initial post.
So all settings are pretty much default now, except for metric_min_improvement_export which i don't think matters in this case

[predbat_debug_09_25_00.yaml.txt](https://github.com/user-attachments/files/19260142/predbat_debug_09_25_00.yaml.txt)

[predbat.log](https://github.com/user-attachments/files/19260146/predbat.log)

Plan:
[image]

springfall2008, 2025-03-15:
Adding a more complete picture

The second charge at 16:30 is a hold at 100%, its used to run from grid at around 125 otherwise you import more at 138/140 range later on, you still run out of battery for the 132 price even with that hold.

The only question then is why the charge to 79 at 12:00 - let me run the plan without it next

[image]

springfall2008, 2025-03-15:
There's something odd here, if I put the two side by side as the plan with the charge appears to export _more_ rather than less, will debug further

[image]

springfall2008, 2025-03-15:
So you have a hybrid inverter with a 20kw limit, an export limit of 11kw and a battery charge rate of 6kw? This means the PV is more than enough to charge the battery at full rate regardless of if charge is enabled or not.

Your battery charge curve tails off at 91% onwards and with the given export limit you will be getting clipping (lost solar you can't export)

All very odd...

freddanastrom, 2025-03-15:
> Adding a more complete picture
>
> The second charge at 16:30 is a hold at 100%, its used to run from grid at around 125 otherwise you import more at 138/140 range later on, you still run out of [...]

Yes, I understand the second charge at 100% and that one seems reasonable. At least as long as the prices for tomorrow are unknown.

It’s those mid day charges are a bit strange and by looking at the history of predbat.status it is also switching between hold and charge at times not shown in the plans I uploaded. So the plan must have changed a bit during the day.

[image]

Let me know if you need any more debug files that match those time stamps.

freddanastrom, 2025-03-15:
> So you have a hybrid inverter with a 20kw limit, an export limit of 11kw and a battery charge rate of 6kw? This means the PV is more than enough to charge the battery at full rate regardless of if [...]

Correct.
The charge curve is a bit uncertain though, but somewhat close to reality.

The battery capacity will be expanded with 120 kWh EV battery’s which I haven’t had time to connect yet and I actually tried setting the capacity to 120 kWh in Predbat just to see the plan, from what I can remember it still had some charging planned at that occasion. But also a lot higher prices in the evening that made it profitable to export.

springfall2008, 2025-03-15:
Okay I've found the issue, there's an assumption in the code that is probably wrong that if the battery is set to charge to say 80% that if there is more solar than load and the battery is about to reach that target it won't store the extra solar in the battery. In reality I think it will actually do so, I'll make a fix

springfall2008, 2025-03-15:
Fix on 'main'

springfall2008, 2025-03-15:
https://github.com/springfall2008/batpred/releases/tag/v8.16.4

springfall2008, 2025-03-16:
The hold charge is a little odd however as the battery is already full, I'd be interested to see the debug YAML if you get that again

freddanastrom, 2025-03-17:
> The hold charge is a little odd however as the battery is already full, I'd be interested to see the debug YAML if you get that again

This one is from 09:25 the same day, but should have the same hold charge.
(I have adjusted my export limit from 11 kW to 13 kW in this debug file, since 11 kW was a limit I had earlier, before the expansion of my PV system)
[predbat_debug_09_25_00.yaml.txt](https://github.com/user-attachments/files/19280659/predbat_debug_09_25_00.yaml.txt)

chalfontchubby, 2026-08-20:
Replayed the attached debug.yaml (predbat_debug_07_00_00.yaml.txt) against current main - the symptom still reproduces. At the 07:00 (UTC) / 12:00 (local) slot, PV forecast is 5.96kWh against a load forecast of only 1.64kWh (clear solar surplus), yet the plan still shows a grid `Chrg` slot there (reason: `charge_low_rate`, target 65%). Leaving this open since it's confirmed on current code, not just the version originally reported.

chalfontchubby, 2026-08-20:
Dug into this a bit further using the `predbat_debug_07_00_00.yaml.txt` file - found the actual mechanism behind it.

Predbat doesn't just plan against a single "most likely" solar forecast. It also runs a more pessimistic, worst-case forecast (a low-percentile PV estimate) alongside the normal one, and blends the two together when deciding whether a charge window is worth using - controlled by **input_number.predbat_pv_metric10_weight** ("Metric 10 Weight" in expert mode), which defaults to 0.15 (15%).

In your debug file, at that 12:00 slot: against the *normal* solar forecast, charging from the grid there makes no difference at all to the outcome - the battery ends up in exactly the same place either way, because the sun does fill it up anyway. But against the *pessimistic* forecast, the sun doesn't show up enough, and the battery would need a top-up. Since 15% of the decision is weighted toward that pessimistic case, and the grid rate at 12:00 was relatively cheap for the day, Predbat opts to buy a small amount of cheap insurance - rather than gamble on the sun turning up as forecast.

So it's not ignoring your solar forecast - it's deliberately hedging a little against the forecast being wrong, and on a day where the sun does turn up, that hedge looks unnecessary in hindsight.

One thing that's probably sharpening this in your case specifically: your config also has **input_number.predbat_pv_metric90_weight** (the equivalent optimistic-case weight) set to 0, which switches off the "good case" side of that blend entirely. Normally the optimistic and pessimistic weights pull against each other a bit; with only the pessimistic side active, the hedging is more one-sided. If you'd like Predbat to lean more on the normal forecast and do less of this pre-emptive topping up, try raising `pv_metric90_weight` back towards its default (0.15) alongside `pv_metric10_weight`, or lower `pv_metric10_weight` itself if you're comfortable trusting the forecast more.

This is working as designed rather than a bug, so I'll close this one out - but it's a useful data point, thank you for the detailed debug files. Feel free to reopen if adjusting those two weights doesn't explain what you're seeing.

closed 2026-08-20 by chalfontchubby

gcoan, 2026-08-20:
Thanks for the investigation Rik

> One thing that's probably sharpening this in your case specifically: your config also has input_number.predbat_pv_metric90_weight (the equivalent optimistic-case weight) set to 0, which switches off the "good case" side of that blend entirely

Hey @chalfontchubby I think a minor correction.   This issue was logged a year ago and the PV90 feature has only just been recently introduced.  Previously predbat only had PV50 and PV10 weight as you explained.

I guess that the debug log supplied wouldn't have had PV90 defined in it, so when you loaded it up to debug, it defaulted PV90 weight to 0 as it wasn't configured in that old debug file.

Anyway, PV90 is now available and could be used to weight towards a better forecast as you say.