springfall2008/batpred #2101: Charge command not executed

azebro, 2025-03-15:
**Describe the bug**
When the charge window opens battery charge should start and finish when the window closes.
I can see in logs that the service is called but values are not chnaged, hence the charging does not start / does not finish
**Expected behaviour**
command should start/stop charging

**Predbat version**

latest

**Environment details**

standard hacs install

I have the following input_select that controls charging:
```
  set_sigen_remote_ems_control_mode:
    name: Set Sigen Remote EMS control mode
    options:
      - "PCS remote control"
      - "Standby"
      - "Maximum self-consumption (default)"
      - "Command charging (consume power from the grid first)"
      - "Command charging (consume power from the PV first)"
      - "Command discharging (output power from PV first)"
      - "Command discharging (output power from the battery first)"
    # icon: mdi:remote
    icon: mdi:battery-unknown
```
if controlled by UI works fine.

in the apps.yaml:

```
   # Services to control charging/discharging
  charge_start_service:
    service: input_select.select_option
    entity_id: "input_select.set_sigen_remote_ems_control_mode"
    option: "Command charging (consume power from the PV first)"
  charge_stop_service:
    service: input_select.select_option
    entity_id: "input_select.set_sigen_remote_ems_control_mode"
    option: "Maximum self-consumption (default)"
  discharge_start_service:
    service: input_select.select_option
    entity_id: "input_select.set_sigen_remote_ems_control_mode"
    option: "Command discharging (output power from PV first)"
```

in the logs:

Warn: Service call select/select data {'entity_id': 'input_select.set_sigen_remote_ems_control_mode', 'option': 'Maximum self-consumption (default)'} failed with response None

The issues is that the values in the input select seems not to be changed by predbat and the charging either does not start or does not finish.

springfall2008, 2025-03-15:
Can you do go 'Developer/Actions' and then use the GUI to create the charge start action, then switch to YAML mode and copy it here?

[image]

azebro, 2025-03-15:
Hi @springfall2008, is this what are you looking for?

[image]

springfall2008, 2025-03-16:
Right, but you error is  "Warn: Service call select/select data {'entity_id': 'input_select.set_sigen_remote_ems_control_mode', 'option': 'Maximum self-consumption (default)'} failed with response None"

So try creating the action for this one and check it works and compare the YAML against your config