springfall2008/batpred #2114: Hold charge does not adjust max charge rate

azebro, 2025-03-17:
When hold chanrge is execured, only discharge rate is adjusted as per the code below:

```
if self.set_soc_enable and self.soc_percent >= target_soc:
                                status = "Hold charging"
                                self.log("Inverter {} Hold charging as soc {}% is above target {}% set_discharge_during_charge {}".format(inverter.id, self.soc_percent, target_soc, self.charge_limit_percent_best[0], self.set_discharge_during_charge))

                                if (target_soc < 100.0) and (abs(inverter.soc_percent - target_soc) <= 1.0):
                                    # If we are within 1% of the target but not at 100% then we can hold charge
                                    # otherwise keep charging enabled
                                    if self.set_soc_enable and ((self.set_reserve_enable and self.set_reserve_hold and inverter.reserve_max >= inverter.soc_percent) or inverter.inv_has_timed_pause):
                                        inverter.disable_charge_window()
                                        disabled_charge_window = True

                                        if self.set_reserve_enable and not inverter.inv_has_timed_pause:
                                            inverter.adjust_reserve(min(inverter.soc_percent + 1, 100))
                                            resetReserve = False
                                    else:
                                        inverter.adjust_charge_window(charge_start_time, charge_end_time, self.minutes_now)

                                    # Pause?
                                    if inverter.inv_has_timed_pause:
                                        inverter.adjust_pause_mode(pause_discharge=True)
                                        resetPause = False
                                    else:
                                        **inverter.adjust_discharge_rate(0)**
                                        
                                        resetDischarge = False
```
this caused the inverter to keep charging.

Suggest changing it to:

```
          if self.set_soc_enable and self.soc_percent >= target_soc:
                                status = "Hold charging"
                                self.log("Inverter {} Hold charging as soc {}% is above target {}% set_discharge_during_charge {}".format(inverter.id, self.soc_percent, target_soc, self.charge_limit_percent_best[0], self.set_discharge_during_charge))

                                if (target_soc < 100.0) and (abs(inverter.soc_percent - target_soc) <= 1.0):
                                    # If we are within 1% of the target but not at 100% then we can hold charge
                                    # otherwise keep charging enabled
                                    if self.set_soc_enable and ((self.set_reserve_enable and self.set_reserve_hold and inverter.reserve_max >= inverter.soc_percent) or inverter.inv_has_timed_pause):
                                        inverter.disable_charge_window()
                                        disabled_charge_window = True

                                        if self.set_reserve_enable and not inverter.inv_has_timed_pause:
                                            inverter.adjust_reserve(min(inverter.soc_percent + 1, 100))
                                            resetReserve = False
                                    else:
                                        inverter.adjust_charge_window(charge_start_time, charge_end_time, self.minutes_now)

                                    # Pause?
                                    if inverter.inv_has_timed_pause:
                                        inverter.adjust_pause_mode(pause_discharge=True)
                                        resetPause = False
                                    else:
                                        **inverter.adjust_discharge_rate(0)
                                        inverter.adjust_charge_rate(0)**
                                        resetDischarge = False
```
to make sure the charge is stopped.

springfall2008, 2025-03-18:
I don't set the charge rate to 0 as it would cause solar to be exported rather than it being sent to the battery. Hold charge should disable the charge window instead.

What is the issue you have?

sammort, 2025-03-18:
Is this issue related? https://github.com/springfall2008/batpred/issues/2035

I've added a debug log for the behaviour in the last comment. I appreciate you get a lot of issues reported so I'm really just trying to bump this back to the front page 😊

WyndStryke, 2025-03-21:
I have something similar, where both hold charge and freeze charge still charge the battery at full rate during the overnight cheap period.  Discharge rate goes to zero for unclear reasons (it does not need to be zeroed during charging), but charge rate is still maxed out at 7kW.  

Probably inverter configuration in my case, so I'm going to talk it through in a discussion thread first.   Thinking of using the inverter's max_soc hard limit for target_soc, but switch.predbat_inverter_soc_reset doesn't seem to set it back to 100% after charging is complete (it drops right down to the same as the reserve/min_soc for the rest of the day).  I'll give that a go tonight.

[image]

WyndStryke, 2025-03-24:
> I have Sigenergy Sigenstore. I could introduce the freeze potentially, but it actually works well if I change the code as above.

According to the manual, the SigenStor has a 'backup reserve' mode.  This sounds very similar to what I used to implement the charge_freeze_service.

> Backup Reserve If there is a Gateway in the network, you can manually set the "Backup Reserve" value in mySigen App. When the grid is connected, the battery stops discharging when the set backup SOC is reached; when the grid is powered down, the battery power from the backup can be used. Example: Self-Consumption Mode involves backup SOC. 

Can this be controlled via the integration (the predbat preserve_soc would be configured to use the backup_soc control, and the mode would be set by charge_freeze_service)?