springfall2008/batpred #2116: Referenced entities switch.predbat_active are missing or not currently available

kukumagi, 2025-03-17:
**Describe the bug**
Referenced entities switch.predbat_active are missing or not currently available
I get this when switching predbat_active
In entities I get this
This entity ('switch.predbat_active') does not have a unique ID, therefore its settings cannot be managed from the UI. See the [documentation](https://www.home-assistant.io/faq/unique_id) for more detail.

**Expected behaviour**
It should switch predbat active on but after awhile it resets to off

**Predbat version**
latest

**Environment details**

**Screenshots**

**Log file**

**Predbat debug yaml file**

[predbat_debug.yaml.txt](https://github.com/user-attachments/files/19285026/predbat_debug.yaml.txt)

WyndStryke, 2025-03-17:
> It should switch predbat active on but after awhile it resets to off

If I remember correctly, once predbat has finished calculating it's plan, it will set predbat_active off, on the basis that it is now going to sleep for another 10 minutes.  Once the 10 minutes is up, and predbat has woken up, it will set active to true again, and so forth.

gcoan, 2025-03-17:
> If I remember correctly, once predbat has finished calculating it's plan, it will set predbat_active off, on the basis that it is now going to sleep for another 10 minutes. Once the 10 minutes is up, and predbat has woken up, it will set active to true again, and so forth.

you are absolutely correct, see https://springfall2008.github.io/batpred/output-data/#basic-status

@kukumagi I don't quite understand what the issue is you are getting.  Can you please share a logfile or screenshot of the problem

kukumagi, 2025-03-18:
> you are absolutely correct, see https://springfall2008.github.io/batpred/output-data/#basic-status
>
> [@kukumagi](https://github.com/kukumagi) I don't quite understand what the issue is you are [...]

I got confused that it was supposed to stay in active state. Since there is no problem we can close this issue.

closed 2025-03-18 by kukumagi