springfall2008/batpred #2119: Manual forced export not working in control charge mode

mpartington, 2025-03-18:
**Describe the bug**
If Predbat is set to Control Charge mode, then the manual override for export does not work.  No export occurs for the selected time period

select.predbat_manual_export

**Expected behaviour**
A manual forced export should occur in any controlling mode (excluding monitor or read-only)

Confirmed with Trefor on FB this is a bug

**Predbat version**

8.16.6

**Environment details**

- GivEnergy
- Standard HAOS installer or Docker
- Anything else?

**Screenshots**

[image]

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

**Predbat debug yaml file**
Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to predbat_debug_XX_XX_XX.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

springfall2008, 2025-03-18:
It should work, and when tested it works fine for me. I don't have anything to go on without logfiles or debug YAML, can you please try again and capture these?
Thanks

springfall2008, 2025-03-18:
Oh wait, you are in control charge mode, not control charge/discharge - this means it won't export...

gcoan, 2025-03-18:
I understand your rationale for this, but I'd argue this is working as designed https://springfall2008.github.io/batpred/customisation/#predbat-control-charge-mode

In Control charge mode Predbat will set the charge times and charge percentages according to the Best plan, charging can be enabled and disabled by Predbat. Predbat will set the inverter into Eco mode when required to enable the battery to support house load, but **it will not plan any forced discharging of the battery for export purposes**.

When you do a manual force export you are telling Predbat to change the plan,  and in this case you are telling it to plan something it doesn't do.

Same would apply if in monitor mode or Control SOC mode and you did force charge, etc

mpartington, 2025-03-18:
I think it is open to interpretation, I would argue that part of the manual is written around how it works in 'automatic / hands off mode'. If I use a manual force, then I am planning the behaviour (i.e. consciously overriding the Predbat plan).  Maybe the manual forces should work in those other modes as well ...  

At least it doesnt cancel the discharge if I set via GivTCP -just tested

gcoan, 2025-03-18:
You could argue it either way. 

One option I thought of is to leave in Control Charge and Discharge but set the export rate to zero.  Predbat shouldn't do any export as no financial benefit but you've got the option to do manual export if you want to

mpartington, 2025-03-18:
That's not a bad shout

mpartington, 2025-03-18:
Closing as normal behaviour

closed 2025-03-18 by mpartington