springfall2008/batpred #2124: Freeze discharge slots disappear from the plan in favour of forced export

mpartington, 2025-03-18:
**Describe the bug**
myself.and another user have noticed that Predbat plans periods of freeze discharge, which are sensible. Then despite having sufficient solar (even considering best10 plan) , these slots disappear and an export is planned for later instead.

**Expected behaviour**
Freeze discharge where appropriate to minimise round trip losses

I caught the debug file late, but can use this to upload more data as available

**Predbat version**
V 8.16.6

**Environment details**

- Inverter and battery setup - Givenergy AC3 and 2x5.2kWh batteries 
- Standard HAOS

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

[image]

[image]

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

**Predbat debug yaml file**

[predbat (1).log.txt](https://github.com/user-attachments/files/19315710/predbat.1.log.txt)

[predbat_debug.yaml.txt](https://github.com/user-attachments/files/19315682/predbat_debug.yaml.txt)

Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to predbat_debug_XX_XX_XX.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

mpartington, 2025-03-19:
Two more alternating plan debug files
Common sense says a freeze would be the best option to minimise later discharge 

[predbat_debugNOFREEZE.yaml.txt](https://github.com/user-attachments/files/19335175/predbat_debugNOFREEZE.yaml.txt)
[predbat_debugFREEZE.yaml.txt](https://github.com/user-attachments/files/19335176/predbat_debugFREEZE.yaml.txt)

mpartington, 2025-03-20:
[predbat_debug (1).yaml.txt](https://github.com/user-attachments/files/19361842/predbat_debug.1.yaml.txt)
[predbat_debug.yaml.txt](https://github.com/user-attachments/files/19361841/predbat_debug.yaml.txt)

2.more.examples

mpartington, 2025-03-20:
After a short period of discharge freeze, it's cancelled it and set a nearly 2 hours forced discharge at the end of day, rather than exporting the excess solar directly

[predbat_debug(1).yaml.txt](https://github.com/user-attachments/files/19363141/predbat_debug.1.yaml.txt)
[predbat_debug.yaml.txt](https://github.com/user-attachments/files/19363142/predbat_debug.yaml.txt)

mpartington, 2025-03-20:
More odd behaviour. All rates turned red and no plan to charge at cheaper rates overnight. I'm sure it will.sort out later, but looks like a bug

[predbat_debug (2).yaml.txt](https://github.com/user-attachments/files/19366635/predbat_debug.2.yaml.txt)

[image]
[image]

mpartington, 2025-03-20:
This is related to the value used for input_number.predbat_metric_min_improvement_expor

If you put a low value e.g. 2, then it plans an export freeze. However if you use a higher number (which is intented to stop marginal battery discharge - not direct export), then it removes the freezes.

Likely related to an earlier issue I raised #1997

springfall2008, 2025-03-22:
I've pushed a fix to 'main' which I will release too which decouple the min improvement export from freeze export

mpartington, 2025-03-22:
Thanks , will.givw main a go

gcoan, 2025-03-25:
mine has been pretty stable today

[image]

have you got a metric min improvement export set?  Mine is set to 5p

mpartington, 2025-03-25:
I've also got it set to 5p, however I'm running the version on main where Trefor has decoupled it, so I wonder if it needs a bit of damping adding

mpartington, 2025-03-26:
Here's a couple of debug files.  On the face of it, there appears no reason to have switched from freeze export to demand

[predbat_debug (1).yaml.txt](https://github.com/user-attachments/files/19463874/predbat_debug.1.yaml.txt)
[predbat_debug.yaml.txt](https://github.com/user-attachments/files/19463875/predbat_debug.yaml.txt)

mpartington, 2025-03-28:
Same today 

[image]

mpartington, 2025-04-08:
@springfall2008 I've updated to the latest version on main and tried to get most settings near to default, however the plan still constantly flip flops between freeze export and demand. The soc gradually increases until.it reaches 100%, so this feature doesn't seem to be working correctly (at least for me)

I have min metric improvements export set to 5 (the default 3 added unwanted early discharges)

I've captured 2 debug yamls. 1 in demand and one in freeze export

Do you have any suggestions for my config, or can anything be done to make it more stable in decision making. Plenty of solar today and low loads, so expected it to hold at 89% as originally planned.

[predbat_debug-1.yaml.txt](https://github.com/user-attachments/files/19649853/predbat_debug-1.yaml.txt)

[predbat_debug-2.yaml.txt](https://github.com/user-attachments/files/19649854/predbat_debug-2.yaml.txt)

[image]

lxalln, 2025-04-10:
I'm seeing this problem too. At the beginning of the day, the plan was to hold at 85%~ battery, and then as time approaches that, it all gets dropped and replaced with Demand followed by a dump later in the day.

mpartington, 2025-04-10:
As a work around selecting "Set Export Freeze Only" to true seems to be keeping the slots and I'm not seeing the cycling.

I'll probably set an automation to turn this switch on/off at sunrise/sunset as a temporary fix.

Meatballs1, 2025-04-11:
Yesterday PB planned mostly a full day of freeze export and seemed to hold to it. Today it isn't planning freeze export, despite several hours of forced export planned at the end of the day:

[image]

[predbat-11-04.log](https://github.com/user-attachments/files/19701736/predbat-11-04.log)

[predbat_plan-11-04.txt](https://github.com/user-attachments/files/19701740/predbat_plan-11-04.txt)

[predbat_debug-11-04.txt](https://github.com/user-attachments/files/19701741/predbat_debug-11-04.txt)

mpartington, 2025-04-11:
Debug capture. Just changed from freeze export to demand, despite significant margin to soc keep on the best10 

[image]

[predbat_debug-2.yaml.txt](https://github.com/user-attachments/files/19702138/predbat_debug-2.yaml.txt)

Meatballs1, 2025-04-11:
My plan has updated to more sane version:

[image]

mpartington, 2025-04-14:
Finally managed to catch it in both states. V8.18.1 main version for testing

[predbat_debug.yaml.txt](https://github.com/user-attachments/files/19736704/predbat_debug.yaml.txt)
[predbat_debug-3.yaml.txt](https://github.com/user-attachments/files/19736705/predbat_debug-3.yaml.txt)

chalfontchubby, 2026-08-20:
Replayed both attached debug.yaml files against current main. Comparing the two: the 15:30 slot is planned as `FrzExp` (freeze export) in the 13:55 capture, but by the 14:55 capture - 35 minutes before that slot even arrives - it's changed to plain `Demand`. That matches the reported symptom exactly. Leaving this open since it's confirmed on current code.

chalfontchubby, 2026-08-20:
Dug further into this (root-cause investigation, no fix yet). Summary for anyone picking this up:

**Confirmed on current main**, using the attached debug.yaml pair: the 15:00-16:00 slot (a flat-rate span, no price change at all across 14:30-17:00) has export candidate windows for every 30-min sub-slot, but the search leaves 900-960 (15:00-16:00) as plain `Demand` while freezing on both sides of it. Forcing freeze through that gap is measurably cheaper - confirmed via a direct `run_prediction_metric()` comparison on the fully-converged plan (~0.92 better metric, ~£1.09 better cost for this specific day).

**Ruled out**: a general-vs-freeze threshold mixup. `optimise_export()`'s per-window trial correctly hits the `this_export_limit == 99` branch and correctly applies `metric_min_improvement_export_freeze` (0.1, untouched from default), not the reporter's customised general `metric_min_improvement_export` (5.0).

**Found instead**: the live search's own per-window trial reports a *higher* (worse) metric for the freeze option than the "off"/Demand option (74.32 vs 73.29) - the opposite conclusion from a fresh `run_prediction_metric()` call on the identical window/limit configuration, which says freeze is *better* (17.84 vs 18.76). Two things stand out about this: the live search's own numbers are on a completely different scale than a fresh calculation of the same scenario (73-74 vs 17-18, not just a rounding gap), and the discrepancy is stable - re-running `calculate_plan()` two more times from the already-converged state reproduces the exact same (wrong) answer every time, so this isn't under-convergence.

**Leading hypothesis, not yet confirmed**: a stale/colliding entry in the scenario-dedup cache (`tried_list`/`scenario_hash_entry` in `optimise_charge_limit_price_threads`) returning a cached metric from an unrelated trial instead of actually evaluating this one. Would explain the scale mismatch, the sign flip, and why it never self-corrects across repeated passes. Next step is instrumenting that cache directly to check for a collision on this scenario.

Reproduction is via the two attached debug.yaml files (`--debug_file <file> --redo`) if anyone wants to pick this up independently.

chalfontchubby, 2026-08-20:
**Correcting my previous comment** - two of the things I flagged there were my own measurement error, not real signals. Apologies for the noise.

**What was wrong:**

1. The "different scale (73-74 vs 17-18)" observation was **my bug**, not a clue. `run_prediction_metric()` defaults `end_record` to the full `forecast_minutes` (2880) when you don't pass it, whereas the optimiser passes the plan's actual `end_record` (1955 here). I wasn't passing it, so I was costing a longer horizon. Re-run with the correct horizon, my numbers are 72.05 / 71.13 - same ballpark as the search's. There is no mysterious scale factor.
2. The "same question, two different answers" framing was **not like-for-like**. The live search evaluates that window against the full 97-window fine-grained candidate list; I was evaluating against the merged 10-window final list with a window spliced in. Different structures, so those numbers were never directly comparable.
3. The scenario-cache hypothesis is **ruled out**. `prediction_kernel.py` has a built-in `VALIDATE_WINDOW_CACHE` mode that raises if the identity-keyed window-bounds cache ever goes stale; enabled it for a full replay and it stayed clean throughout.

**What survives, and looks like the actual root cause:**

The improvement is real and reproducible - forcing freeze over the 15:00-15:30 gap in the final plan is better by **-0.9163 metric / -1.0925 cost**, and that delta is identical whether measured over the plan horizon or the full forecast horizon, and identical whether expressed as extending the neighbouring window or inserting a new one.

But the search is *not* making a mistake at the point it decides. Instrumenting the moment window 900-930 is evaluated, with its real neighbour context:

```
baseline (both off)    metric=73.2907
freeze 900 only        metric=74.3155
freeze 930 only        metric=74.4380
freeze BOTH            metric=75.3771
```

Leaving both off is genuinely the best of those options *at that point*, and freezing both together is worse than either alone - so this isn't a pairwise move a single-window greedy search is missing either.

The difference is **structural**. `optimise_all_windows()` makes all the export decisions against 97 fine-grained candidate windows. Everything after it in `calculate_plan()` is one-way post-processing - `prune_dead_plan_slots()`, `discard_unused_export_slots()`, `clip_export_slots()`, then `discard_unused_export_slots()` again - which collapses the list from **97 windows to 10**. Nothing re-optimises after that collapse. So a freeze that only becomes worthwhile in the collapsed structure can never be found: by the time that structure exists, the optimiser has finished.

That also explains why re-running `calculate_plan(recompute=True)` two more times reproduces the identical result - it rebuilds from scratch and retraces the same deterministic path, rather than starting from the converged structure.

Still no fix - the obvious directions (re-run an optimisation pass after the collapse, or don't discard windows that are still viable freeze candidates) both have cost/regression implications I haven't evaluated.

chalfontchubby, 2026-08-20:
Final write-up for now - stopping here with a root cause but no fix. Recording it so it's useful for diagnostics or a future attempt.

**Root cause: export freeze decisions are scored against a plan that no longer exists by the time the plan ships.**

`calculate_plan()` runs `optimise_all_windows()` once, against ~97 fine-grained export candidate windows. Everything after that is one-way post-processing with no re-optimisation:

```
optimise_all_windows()          <- all export decisions made here, 97 windows
update_target_values()
remove_intersecting_windows()
prune_dead_plan_slots()
discard_unused_export_slots()   <- drops every window with limit == 100.0, merges contiguous equal ones
clip_export_slots()             <- rewrites export limits to achievable SoC levels
discard_unused_export_slots()   <- again
                                   ends with 10 windows
```

Diffing the plan at decision-time against the final plan (matched by window *time*, not list index - the lists are different lengths, so index matching is meaningless here):

- Most of the 97 -> 10 collapse is benign merging: `(960,990)=99` + `(990,1020)=99` becomes `(960,1020)=99`, same physical behaviour.
- But `clip_export_slots` **materially rewrites the overnight export leg**: `(1560,1590): 0 -> 68`, `(1735,1740): 0 -> 85`, and `(1620,1650)=71` / `(1680,1710)=74` become active where nothing was before.
- Charge limits: **zero changed**.

So the freeze decision for the 15:00-16:00 gap was evaluated against an overnight export plan that clipping then substantially rewrote. Nothing re-scores the earlier decision afterwards, so a freeze that only becomes worthwhile against the *clipped* plan can never be found.

**Why the obvious "just don't discard the candidates" fix doesn't work on its own:** `discard_unused_export_slots()` only keeps windows where `export_limits_best[window_n] < 100.0` - it discards exactly the windows already decided as off, and it runs *after* the decision. Preserving them preserves a set of no-ops; it doesn't re-open the choice. It's likely a *prerequisite* for a real fix (a later re-optimisation pass needs candidate windows to work with, and by that point they've been merged away) rather than a fix in itself.

**Supporting evidence, for anyone picking this up:**

- The improvement is real: forcing freeze over the gap in the final plan is **-0.9163 metric / -1.0925 cost**, identical whether measured over the plan horizon or the full forecast horizon, and identical whether expressed as extending the neighbouring window or inserting a new one.
- The search is **not** making a local mistake. Instrumented at the decision point with real neighbour context: baseline (both off) 73.2907, freeze-900-only 74.3155, freeze-930-only 74.4380, freeze-both 75.3771. Off is genuinely best there, and freezing both together is worse than either alone - so it isn't a pairwise move a single-window greedy search is missing either.
- Ruled out: threshold mixup (the `this_export_limit == 99` branch correctly applies `metric_min_improvement_export_freeze`), and stale window-bounds cache (`prediction_kernel.VALIDATE_WINDOW_CACHE` stays clean across a full replay).

**Diagnostic tip that cost me a wrong conclusion earlier in this thread:** `run_prediction_metric()` defaults `end_record` to the full `forecast_minutes`, *not* the plan's `end_record`. If you compare its output against numbers the optimiser produced internally, pass `end_record` explicitly or you'll be costing a different horizon and the scales won't match.

A fix would mean preserving the fine-grained structure through clipping, re-optimising, then discarding/merging for display and execution - which puts another optimisation pass on an already hot path, so it needs proper performance scoping before anyone commits to it.

chalfontchubby, 2026-08-20:
Two corrections and an important caveat on scope, so nobody acts on this with the wrong picture.

**1. Units - the improvement is ~1.1 PENCE, not £1.09.** In my first comment I wrote "~£1.09 better cost for this specific day". That's wrong by 100x. Predbat's metric/cost are in pence (rates in this file are 27.85p import / 15.0p export, and per-slot `cost_change` values are hundredths of a penny). So the measured improvement is **-0.9163 metric / -1.0925 cost = roughly one penny for that day**. Over a year that's a few pounds at most. That materially changes the cost/benefit of fixing it - it is not a large economic bug, and anyone weighing "add an optimisation pass to the hot path" against it should be weighing it against ~1p/day, not ~£1/day.

**2. "The plan changes its mind after clipping" would be the wrong reading.** It never re-evaluates at all. The decision is made pre-clip and simply never revisited. What I demonstrated is that *if* it were re-scored post-clip it would choose differently - not that anything actually reconsiders and flips.

**3. Caveat on how much of the reported symptom this actually explains.** What I've shown solidly is that the *final shipped plan is suboptimal by its own cost model* - there's a strictly cheaper alternative it didn't pick.

What I have **not** shown as solidly is that this is the whole reason the slots *appear and then disappear between cycles*, which is the actual reported complaint. Between the two captures the forecast SoC for that slot moved (84% -> 76%), so the flip is at least partly driven by genuinely changed inputs, i.e. legitimate re-planning that happens to land somewhere suboptimal. The pre-clip/post-clip gap explains why the landing point is worse than it needs to be; it doesn't by itself prove the flip wouldn't happen without it. Someone picking this up should treat "final plan is improvable" (well-evidenced) and "this causes the cycle-to-cycle flapping" (plausible, not proven) as two separate claims.

chalfontchubby, 2026-08-20:
**Gap closed - correcting my own correction. The attribution is now nailed down.**

I previously blamed `clip_export_slots`, then walked that back after finding clipping is largely cosmetic. Isolating each side properly settles it. Holding the decision-time state and swapping in one side at a time:

```
A  decision-time state                : metric=73.2907
B  decision export + FINAL charge     : metric=98.9397   (charge-side effect: +25.6490)
C  FINAL export + decision charge     : metric=72.0471   (export-side effect: -1.2436)
D  final shipped plan                 : metric=72.0471   (total: -1.2436)
```

**C and D are identical to four decimal places.** With the final export plan in place, it makes no difference whatsoever whether the charge plan is the decision-time one or the final one. So the charge-side changes (the 21 charge windows that disappear via `remove_intersecting_windows`) contribute **exactly zero** to the outcome - my previous suspicion that they were the culprit was wrong. (B looks dramatic at +25.65, but that's an unreachable hybrid: the decision-time export plan demands far more export than the reduced final charge plan can support. It isn't a state the pipeline ever produces.)

**The entire -1.2436 sits on the export side.**

And note the sign: post-processing makes the plan **better** by 1.24, not worse. So the pipeline isn't damaging the plan - it's improving it, and then leaving a further 0.92 on the table that the optimiser would have found had it been working in the post-processed structure to begin with.

**Which export step**: not clipping. Asking for a physically impossible target simulates the same as the reachable one - `(1560,1590)` at limit 68 vs limit 0 is identical to nine decimal places, and the other clipped windows differ by only 0.03-0.05. `run_prediction` already clamps export to the achievable discharge rate, so clip's rewrite is cosmetic relabelling to make the *inverter command* match reality, not an economic change. That leaves the **structural** work - `prune_dead_plan_slots` plus the discard/merge that collapses 97 fine-grained candidate windows into 10 merged ones, moving window boundaries in the process.

**Root cause, stated properly:** all export decisions are made against ~97 fine-grained candidate windows. Export-side post-processing then restructures that into 10 merged windows - a genuinely better plan - but nothing re-optimises against the new structure, so decisions tuned for the old one are left in place. In this case that leaves ~0.92 metric (~1 penny/day) unclaimed.

**This also answers "should planning refuse to plan physically impossible things?"** - it effectively already doesn't. The simulation is rate-aware; only the *label* is briefly unreachable until clipping tidies it. So making the planner rate-aware wouldn't address this.