springfall2008/batpred #2125: SoC forecast does not reflect plan

AberDino, 2025-03-19:
I'm on Predbat v8.16.6 and noticed that the SoC forecast does not reflect the charge plan, as per screenshot below, i.e. the plan is to charge to 78% but the forecast shows the battery charging to 100%. This is at night, without any PV  generation... See:
[image]

springfall2008, 2025-03-22:
Can you please download the debug YAML file (from the Web interface dashboard) and post it here?

AberDino, 2025-03-22:
I will do, once we have a sunny day in the forecast and the overnight charge target is less than 100%.

AberDino, 2025-03-23:
Hi @springfall2008, attached the predbat_debug.yaml file, to go with this screenshot:
 
[image]

[predbat_debug.zip](https://github.com/user-attachments/files/19411463/predbat_debug.zip)

AberDino, 2025-03-26:
Hi @springfall2008, any interesting observations from the debug file?

springfall2008, 2025-03-29:
Fix on main for testing

AberDino, 2025-03-29:
Thanks @springfall2008, I'll keep an eye on it over the next few days and I'll close this issue once I can confirm it is fixed.

closed 2025-04-06 by AberDino